aws/amazon-freertos #3382: [General] AFR & ESP32: Wi-Fi/ softAP provisioning using a web-browser on the Mobile phone

dipen-1337lab, 2021-09-09:
Namaste @aggarg, @shubhamkulkarni97, @dachalco and other Developers,

### **Briefly summarize the issue being raised**
Our IoT Product and Solution offering is developed using ESP32DevKitC Board. We have provided our custom Mobile App to our client. For the Wi-Fi provisioning however, we are additionally using the [ESP SoftAP Provisioning](https://apps.apple.com/us/app/esp-softap-provisioning/id1474040630) Mobile App to achieve the one-time device setup.

Using two different Apps - one for Wi-Fi provisioning and, the other one specific to our application - isn't customer friendly. That's when the customer came up with a suggestion of Wi-Fi/ softAP provisioning over a web browser. 

Is possible to Wi-Fi/ softAP provision a device using a web-browser on the Mobile phone, instead of using the Mobile App?

### **Describe the desired outcome**
Instead of 
- using the [ESP SoftAP Provisioning](https://apps.apple.com/us/app/esp-softap-provisioning/id1474040630) Mobile App as an alternate App for Wi-Fi provisioning or,
- integrating the [ESP SoftAP Provisioning](https://apps.apple.com/us/app/esp-softap-provisioning/id1474040630) Mobile App source code with our Mobile App source code,

Is it possible to Wi-Fi provision a device by simply opening a web-page using Safari/ Chrome/ Edge/ Firefox, etc. on the mobile phone? A User would connect to the ESP32 device via the web-page instead of using an App. If yes, please suggest how this can be achieved.

### **System information**
- Hardware board: ESP32-DevKitC
- IDE used: N/A
- Operating System: Windows 10 x64 -- Version: 21H1 -- OS Build: 19043.1165
- Code version: v1.0.0-beta-35-g15068c7
- Project/Demo: custom application developed using coreHTTP, mqtt_demo and ota_demo reference source code. 

### **References**
With reference to [AFR & ESP32: Run-time configuration of Wi-Fi credentials (Wi-Fi Provisioning) for a Device](https://github.com/aws/amazon-freertos/issues/2960), we have successfully integrated the Wi-Fi provisioning logic to provision new ESP32 devices using the [ESP SoftAP Provisioning](https://apps.apple.com/us/app/esp-softap-provisioning/id1474040630) Mobile App.

### **Code/Steps to reproduce the current status**
N/A

### **Additional context**
N/A

Thanks | Regards,
Dipen

aggarg, 2021-09-09:
Seems more like an Espressif question. @shubhamkulkarni97  can you help?

Thanks.

sandipls, 2021-09-11:
@dipen-1337lab  Are you looking for Captive portal like solution for wifi provisioning? Like it should redirect the configuration page when any client device connected to your softAP?

shubhamkulkarni97, 2021-09-13:
Hi @dipen-1337lab,

There is no support for using browser to provision ESP devices. All the supported provisioning tools are listed in the [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/provisioning/provisioning.html#provisioning-tools).

However, if you want to create your own web based provisioning tool, you can check similar ESP-IDF issue at https://github.com/espressif/esp-idf/issues/4399

Thanks,
Shubham

dipen-1337lab, 2021-09-13:
Namaste @sandipls, 

> Captive portal like solution for wifi provisioning

Thank you for your response. Got to learn something new - [Captive portal](https://en.wikipedia.org/wiki/Captive_portal), so thank you for that again 👍.

> Like it should redirect the configuration page when any client device connected to your softAP?

Yes, the **captive portal** solution could be the potential implementation wherein the redirection would open a web-page and, the User be prompted to enter the set of Wi-Fi credentials (SSID and Password). The ESP32 device would then attempt to hook itself with that AP for Wi-Fi access. The further softAP provisioning steps remain the same, but over the web-page on a mobile phone.

Please do share the relevant references for this approach.

Thanks | Regards,
Dipen

dipen-1337lab, 2021-09-13:
Namaste @shubhamkulkarni97, 

> There is no support for using browser to provision ESP devices.

Thank you for the confirmation and, for sharing the relevant references. We shall explore the same.

My feedback - Please do consider adding ESP-IDF support for this alternate approach, in case it does value add. I'm sure the Team must have various thoughts about it :) . Good luck 👍 

Thanks | Regards,
Dipen

aggarg, 2021-09-13:
Hi @dipen-1337lab, I am closing this issue. If you have further questions about this topic, please use the issue Shubham mentioned above or open a new one with ESP. If you have any other question, feel free to comment here or open a new issue.

Thanks.

closed 2021-09-13 by aggarg

sandipls, 2021-09-15:
> Namaste @sandipls,
> 
> > Captive portal like solution for wifi provisioning
> 
> Thank you for your response. Got to learn something new - [Captive portal](https://en.wikipedia.org/wiki/Captive_portal), so thank you for that again +1.
> 
> > Like it should redirect the configuration page when any client device connected to your softAP?
> 
> Yes, the **captive portal** solution could be the potential implementation wherein the redirection would open a web-page and, the User be prompted to enter the set of Wi-Fi credentials (SSID and Password). The ESP32 device would then attempt to hook itself with that AP for Wi-Fi access. The further softAP provisioning steps remain the same, but over the web-page on a mobile phone.
> 
> Please do share the relevant references for this approach.
> 
> Thanks | Regards,
> Dipen

Hi Dipen, We have developed solution for SSID/Password configuration from Web, please let me know if you are interested in the same?

dipen-1337lab, 2021-09-16:
Namaste @sandipls,

We would like to understand the solution being offered. Requesting you to drop me an email (available in my profile section). I could not find your email and hence, requesting you to send an email.

Thanks | Regards
Dipen

sandipls, 2021-09-16:
Sure @dipen-1337lab, I will send you the detail soon. Thanks

danieljovan, 2022-07-07:
@sandipls Is there any link to the repo with the implementation of configuring SSID/password via web interface?

dachalco, 2022-07-07:
Ping/tag me know if you hit any insurmountable `amazon-freertos` specific road blocks while trying to do so. 

We do provide a [wifi-provisioning over SoftAP](https://github.com/aws/amazon-freertos/blob/main/vendors/espressif/boards/ports/wifi/iot_softap_wifi_provisioning.c) that's compatible with our [wifi-provisioning demo](https://github.com/aws/amazon-freertos/tree/main/demos/wifi_provisioning), but the HTTP server communicates via esp's protocomm instead of HTML. It'll at least provide some leads on implementing the SoftAP and HTTP server on esp32 w/ `amazon-freertos`