aws/amazon-freertos #3385: MQTT message not published while in Subscribe Callback Function

Prahar08modi, 2021-09-13:
**Briefly summarize the issue being raised**
Firstly, I'm using freertos in my own CMake project on ESP32 following the link https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#getting_started_espressif_cmake_project

Background Knowledge:
Following the link, I successfully created components for managing connection and reconnection of MQTT connection (using coreMQTT_Agent), Pub-Sub, and other application-specific components. I successfully tested publish and subscribe to multiple topics. A user callback function is called whenever an incoming publish message is received on the subscribed topics.

So, I want to publish an acknowledgment message from the user callback function to inform the cloud regarding the action taken. But, the MQTT message is not published when called from the callback function. However, when publish function is called outside of the callback function, it works fine. 

**My understanding and observations**:
Whenever a Publish or Subscribe command is given to coreMQTT_Agent, it first creates a command and adds that command to the global queue. MQTTAgent_CommandLoop() Function processes the commands in the queue by calling processCommand() function. Whenever I publish an MQTT message from the callback function, the command is created and added to the global queue of commands but it is never executed by MQTTAgent_CommandLoop() ie., processCommand() Function is never called. Hence, the message is not published. 

**Describe the desired outcome**
**Image showing issue**:
[image]

As seen from the above image, as the publish command was added to the queue, the code then waits for its execution and the acknowledgment from the cloud. But, it never receives an acknowledgment. 

**Image showing Ideal execution**:
[image]

Whenever sensor readings are published, it is successfully published and an acknowledgment is received as seen from the above image.

The desired outcome would be to be able to publish an MQTT message from the callback function of the incoming publish message topics.

**System information**
- Hardware board: [ esp32_wrover_kit ]
- IDE used: [ None ]
- Operating System: [ Linux ]
- Code version: my_AFR (amazon freertos used in CMake project) - 154a6aa5559e455e3b9d1f08c4e038d662f5a627
- Project/Demo: [ Custom Application ]

muneebahmed10, 2021-09-14:
Hi @Prahar08modi,

> But, the MQTT message is not published when called from the callback function

Neither the coreMQTT nor the coreMQTT Agent API should be used from within callbacks. Since you are using the coreMQTT Agent, you're using at least two threads: one for MQTT-related tasks (the one responsible for invoking [MQTTAgent_CommandLoop](https://github.com/aws/amazon-freertos/blob/65ac3f8c9ce94a976b347772585ec222d2f9b2c6/demos/coreMQTT_Agent/mqtt_agent_task.c#L856)) and another for your application's other tasks, which send commands to the agent task. Since callbacks are invoked when either a publish is received or the MQTT agent finishes processing a command, they must be executed by the same task as the MQTT agent. So, during a callback, no MQTT agent commands can be processed and no packets can be received.

When you use an MQTT Agent API, you post a command to the agent's queue. Your problem is that you are waiting for the command to be completed, which can never happen during the callback as the MQTT agent can't service the queue, it's stuck waiting in the callback. Even if you remove the wait on the completion, we wouldn't recommend using the MQTT Agent API from within the callback; if the agent's queue happens to be full, the command cannot be sent as the queue is not being processed.

If you want to send a publish in response to a received message, you should use a semaphore or task notification to notify another task to invoke the MQTT Agent API.

Prahar08modi, 2021-09-15:
Hi @muneebahmed10,

Thanks for the great explanation. I implemented a new task which is notified by the callback function for the acknowledgment. This helped me resolve the issue.

Thanks a lot for helping me resolve this issue. I'm closing the issue.

closed 2021-09-15 by Prahar08modi