aws/amazon-freertos #3387: [General] Custom OTA Job on the new release of amazon freeRTOS

yosefgouda2020, 2021-09-16:
Dear,
 
Currently, I am using version 202107 of amazon freeRTOS instead of Version 202012 so in this version there was CustomJobCallback but in version 202107 this callback was removed so how to handle this in version 202107

**Describe the desired outcome**
CustomJobCallback handling .

**System information**
- Hardware board: [ESP32 ]
- IDE used: [ ESP_IDF ]
- Operating System: [ Windows ]
- Code version: 202107 
- Project/Demo: [ ota_demo_core_mqtt\ota_demo_core_mqtt.c]

dachalco, 2021-09-20:
Hi @yosefgouda2020 

Looking into this. Will get back to you shortly.

yosefgouda2020, 2021-09-21:
Hi @dachalco  Thank you .

dachalco, 2021-09-21:
Hey @yosefgouda2020 

Including code refs to make sure we're on the same page. To verify, you want to exercise [this callback](https://github.com/aws/amazon-freertos/blob/202012.00/libraries/freertos_plus/aws/ota/src/aws_iot_ota_agent.c#L2236) that used to be set [here](https://github.com/aws/amazon-freertos/blob/202012.00/libraries/freertos_plus/aws/ota/src/aws_iot_ota_agent.c#L274) in `202012.00`?

If so, yes, the same functionality still exists in `202107.00` but was refactored somewhat. 
For `2021007.00` the app callback has abstracted more and can be set with the last parameter to [OTA_Init(...)](https://github.com/aws/ota-for-aws-iot-embedded-sdk/blob/666241d0f643b07d5146a3715b649d80f8135e0b/source/ota.c#L3063). Since it's more generic, the callback's handed events to identify source.

Considering the `202012.00` callback is exercised [following json parse failure](https://github.com/aws/amazon-freertos/blob/202012.00/libraries/freertos_plus/aws/ota/src/aws_iot_ota_agent.c#L2234), I'd say you're interested in the `OtaJobEventParseCustomJob` event provided to the `202107.00` callback.

For the demo you could modify the [generic callback](https://github.com/aws/amazon-freertos/blob/202107.00/demos/ota/ota_demo_core_mqtt/ota_demo_core_mqtt.c#L855), and handle the case for `OtaJobEventParseCustomJob` as before.

dan4thewin, 2021-09-28:
Since it looks like the question has been answered, I'll close this issue.  If this needs more attention, please open a new issue.  Thank you.

closed 2021-09-28 by dan4thewin

yosefgouda2020, 2021-09-29:
@dachalco Thank you for your answer it helps me.