aws/amazon-freertos #3389: [ERROR][NET][171350] Error -27648 while sending data. AWS FreeRTOS  V202007 GreenGrass Discovery ESP32

Yoimer, 2021-09-16:
Hello guys. Hope you're doing great.

@mingyue86010, @pvyawaha @muneebahmed10

Currently we're using the ESP32 AWS FreeRTOS GreenGrass `202007` from this repository [freertos_greengrass_202007](https://github.com/Yoimer/freertos_greengrass_202007) with the `ESP-IDF 3.3 toolchain`.

We have a dht11, bmp180 and gps attached to it.

After the BLE Over Wifi Provisioning (`Android Mobile App`), the ESP32 sends data constantly to the GreenGrass Core (Raspberry Pi 3B+) running on the same local network. That modification is here [aws_greengrass_discovery_demo.c](https://github.com/Yoimer/freertos_greengrass_202007/blob/268bfc523cd4ccf82f2ef5e0e684b2c4fa84dcfd/FreeRTOS/demos/greengrass_connectivity/aws_greengrass_discovery_demo.c#L123)

We see the expected JSON on AWS IoT Core with no issues.

Howver when testing, we decided restart the router which the ESP is connected to, and on console we noticed the `[ERROR][NET][171350] Error -27648 while sending data.`

We are sure that the ESP32 reconnects by itself to the router (we are pinging its IP address all the time, and that's how we're certain about this).

However it does not send data to AWS anymore, we have to press the physical reset button on the board.

Is there any way the ESP32 after reconnecting to the router keeps sending data to AWS?

I am attaching both logs, when the board is sensor-less and with the sensors attached. 

In case anybody wants to clone the repository, there will not by any issues for its compilation.

We have used the FreeRTOS `202007` with `ota`, `mqtt`, and `ble` in the past with no issues whatsoever. When testing the physical reset on the router, the board is capable to reconnect this router and sends data to aws with no human intervention needed on the field.

We want to achieve the same with greengrass discovery. 

Looking forward to reading your comments.

Thanks in advance.

[greengrass_discovery_with_sensors.txt](https://github.com/aws/amazon-freertos/files/7179294/greengrass_discovery_with_sensors.txt)

[greengrass_sensorless.txt](https://github.com/aws/amazon-freertos/files/7179295/greengrass_sensorless.txt)

muneebahmed10, 2021-09-17:
Hi @Yoimer,

Is this an extension of #3380? This was my earlier recommendation:
> you may want to add error handling code in case there's a disconnect or error with any of the publishes

In your infinite loop I only see [`IotMqtt_TimedPublish`](https://github.com/Yoimer/freertos_greengrass_202007/blob/268bfc523cd4ccf82f2ef5e0e684b2c4fa84dcfd/FreeRTOS/demos/greengrass_connectivity/aws_greengrass_discovery_demo.c#L130). Considering you've reset the network, MQTT needs to be reconnected as well. Even if another thread reconnects MQTT in the background, the MQTT connection handle you're referencing may not be valid any more.

You should reconnect the MQTT connection on error from either your loop or another thread. If you use another thread, you should also make sure it uses the same connection handle as the loop.

Yoimer, 2021-09-17:
@muneebahmed10 thanks a lot for replying. I will try to do what you explained and keep you posted. Hope you have a great day

Yoimer, 2021-10-10:
> Hi @Yoimer, any update on whether your issue was resolved?

Dear @yourslab Thanks for asking and taking care about this. I haven't being able to fix this yet. Currently, working on it. I will keep you posted on it.

Once again, thanks a lot.

muneebahmed10, 2021-11-21:
Hi @Yoimer,

Is there any update on this? Since this issue has been open for two months now, perhaps it would be better to close it and open a new one should the need arise for a follow up in the future?

Thanks

Yoimer, 2021-11-22:
> Hi @Yoimer,
> 
> Is there any update on this? Since this issue has been open for two months now, perhaps it would be better to close it and open a new one should the need arise for a follow up in the future?
> 
> Thanks

Hello @muneebahmed10 Thanks for your concern about this issue. I haven't been able to fix it so far. The problem remains as was expressed month ago.

aggarg, 2021-11-22:
I am closing this issue for now. Let us know if handling the suggestion from @muneebahmed10 does not work for you.

Thanks.

closed 2021-11-22 by aggarg