aws/amazon-freertos #3397: [General] OTA BLE MQTT Transport not reciving OTA updates on ESP32

flow200, 2021-10-06:
**Briefly summarize the issue being raised**
Hello everyone, I am working on "OTA_MQTT_BLE_TRANSPORT" for ESP32 and following this tutorial [https://docs.aws.amazon.com/freertos/latest/userguide/ota-updates-esp32-ble.html] for configurations on AWS side and Firmware side. Everything works fine till subscribing to the job topics for thing. but when I create a new job from aws iot manage jobs console, Esp32 do not receive any new job It just remain in the same state [waitingforjob] and not receive any block of Firmware on subscribed Topic. 

I have tried everything to resolve this from my side, I think may be I am missing something or may be this documentation is old and I need to perform any additional step. So I need a guidance on this issue that I am facing.

**Describe the desired outcome**
Identify the intended result of further discussion and eventual resolution of this issue.

**System information**
- Hardware board: ESP32
- IDE used: Visual code studio
- Operating System: Windows 
- Code version: 202107.00-31-g887697578
- Project/Demo: OTA_MQTT_BLE_TRANSPORT_DEMO_ENABLED

**References**
Providing my ESP32 Logs below:

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7796
load:0x40078000,len:13724
load:0x40080400,len:4568
0x40080400: _init at ??:?

entry 0x400806f4
I (29) boot: ESP-IDF v4.2.1-22-gf8af47e5a-dirty 2nd stage bootloader
I (29) boot: compile time 14:33:41
I (29) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00010000 00006000
I (78) boot:  1 otadata          OTA data         01 00 00016000 00002000
I (86) boot:  2 phy_init         RF data          01 01 00018000 00001000
I (93) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (101) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (108) boot:  5 storage          WiFi data        01 02 00317000 00010000
I (116) boot: End of partition table
I (120) boot: ota rollback check done
I (124) boot_comm: chip revision: 1, min. application chip revision: 0
I (131) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x25acc (154316) map
I (199) esp_image: segment 1: paddr=0x00045af4 vaddr=0x3ffbdb60 size=0x03000 ( 12288) load
I (205) esp_image: segment 2: paddr=0x00048afc vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:\Users\veralagarwal\Documents\amazon_latesfa\amazon-freertos\build/../freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1851

I (206) esp_image: segment 3: paddr=0x00048f08 vaddr=0x40080404 size=0x07110 ( 28944) load
I (228) esp_image: segment 4: paddr=0x00050020 vaddr=0x400d0020 size=0x88f98 (561048) map
0x400d0020: _stext at ??:?

I (442) esp_image: segment 5: paddr=0x000d8fc0 vaddr=0x40087514 size=0x0f2e8 ( 62184) load
0x40087514: r_assert_param at ??:?

I (481) boot: Loaded app from partition at offset 0x20000
I (481) boot: Disabling RNG early entropy source...
I (482) cpu_start: Pro cpu up.
I (485) cpu_start: Application information:
I (490) cpu_start: Project name:     amazon-freertos
I (496) cpu_start: App version:      1
I (500) cpu_start: Compile time:     Aug 19 2021 20:19:33
I (506) cpu_start: ELF file SHA256:  bdef962b78107f58...
I (512) cpu_start: ESP-IDF:          v4.2.1-22-gf8af47e5a-dirty
I (519) cpu_start: Single core mode
[... 339 lines cut ...]
2021-10-06 10:48:02.375 8233-8911/software.amazon.freertos.demo D/FRD: GattService: 00001800-0000-1000-8000-00805f9b34fb
2021-10-06 10:48:02.375 8233-8911/software.amazon.freertos.demo D/FRD:  |-characteristics: 00002a00-0000-1000-8000-00805f9b34fb
2021-10-06 10:48:02.377 8233-8911/software.amazon.freertos.demo D/FRD:  |-characteristics: 00002a01-0000-1000-8000-00805f9b34fb
2021-10-06 10:48:02.377 8233-8911/software.amazon.freertos.demo D/FRD: GattService: 00001801-0000-1000-8000-00805f9b34fb
2021-10-06 10:48:02.377 8233-8911/software.amazon.freertos.demo D/FRD:  |-characteristics: 00002a05-0000-1000-8000-00805f9b34fb
2021-10-06 10:48:02.378 8233-8911/software.amazon.freertos.demo D/FRD: GattService: 8a7f1168-48af-4efb-83b5-e679f932ff00
2021-10-06 10:48:02.378 8233-8911/software.amazon.freertos.demo D/FRD:  |-characteristics: DEVICE_VERSION
2021-10-06 10:48:02.378 8233-8911/software.amazon.freertos.demo D/FRD:  |-characteristics: IOT_ENDPOINT
2021-10-06 10:48:02.378 8233-8911/software.amazon.freertos.demo D/FRD:  |-characteristics: DEVICE_MTU
2021-10-06 10:48:02.378 8233-8911/software.amazon.freertos.demo D/FRD:  |-characteristics: DEVICE_PLATFORM
2021-10-06 10:48:02.378 8233-8911/software.amazon.freertos.demo D/FRD:  |-characteristics: DEVICE_ID
2021-10-06 10:48:02.378 8233-8911/software.amazon.freertos.demo D/FRD: GattService: a9d7166a-d72e-40a9-a002-48044cc30000
2021-10-06 10:48:02.378 8233-8911/software.amazon.freertos.demo D/FRD:  |-characteristics: MQTT_CONTROL
2021-10-06 10:48:02.378 8233-8911/software.amazon.freertos.demo D/FRD:  |-characteristics: MQTT_TX
2021-10-06 10:48:02.378 8233-8911/software.amazon.freertos.demo D/FRD:  |-characteristics: MQTT_RX
2021-10-06 10:48:02.379 8233-8911/software.amazon.freertos.demo D/FRD:  |-characteristics: MQTT_TXLARGE
2021-10-06 10:48:02.379 8233-8911/software.amazon.freertos.demo D/FRD:  |-characteristics: MQTT_RXLARGE
2021-10-06 10:48:02.379 8233-8911/software.amazon.freertos.demo D/FRD: Getting ble software version on device.
2021-10-06 10:48:02.379 8233-8911/software.amazon.freertos.demo D/FRD: Ble operation is in progress. mqtt queue: 0 network queue: 1
2021-10-06 10:48:02.379 8233-8911/software.amazon.freertos.demo D/FRD: Processing BLE command: READ_CHARACTERISTIC remaining mqtt queue 0, network queue 0

**Additional context**
Request you to help me out with this issue ASAP.

Thank you

ravibhagavandas, 2021-11-07:
Hello @flow200,
The error posted above seems to be benign and not directly related to the issue you are reporting. 
```
2021-10-06 10:47:23.410 8233-8908/software.amazon.freertos.demo W/FRD: There's no such service found with uuid: a9d7166a-d72e-40a9-a002-48044cc30100
2021-10-06 10:47:23.412 8233-9480/software.amazon.freertos.demo E/FRD: Ble command failed to be sent OR timeout after 3000ms
```

The characteristic `a9d7166a-d72e-40a9-a002-48044cc30100` is that of wifi provisioning GATT service and its reporting that service not found since most probably wifi provisioning is not enabled on your ESP32 device.

However looking at the following error:
```
2021-10-06 10:47:47.029 8233-9480/software.amazon.freertos.demo D/FRD: <-<-<- Writing to characteristic: MQTT_RX with data: a3616905617301617709
2021-10-06 10:47:47.030 8233-9480/software.amazon.freertos.demo E/FRD: Failed to write characteristic.
2021-10-06 10:47:47.031 8233-9480/software.amazon.freertos.demo E/FRD: Ble command failed to be sent OR timeout after 3000ms
2021-10-06 10:47:47.031 8233-9480/software.amazon.freertos.demo D/FRD: There's no ble command in the queue.
2021-10-06 10:47:49.166 8233-8908/software.amazon.freertos.demo D/FRD: onCharacteristicWrite for: MQTT_RX; status: 133; value: a3616905617301617709
2021-10-06 10:47:49.167 8233-8908/software.amazon.freertos.demo D/FRD: There's no ble command in the queue.
2021-10-06 10:47:49.169 8233-8911/software.amazon.freertos.demo D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=17 device=F0:08:D1:C8:30:76
2021-10-06 10:47:49.194 8233-8911/software.amazon.freertos.demo I/FRD: BLE connection state changed: 8; new state: BLE_DISCONNECTED
```
It seems the BLE peripheral device (ESP32) gets disconnected at the time of writing back an  SUBACK for a topic on the MQTT_RX characteristic. But the same logs on the device side does not show the disconnect. So are they write pairs of logs for the issue ? If not can you post the right pair of peripheral and mobile phone logs from startup for the issue?

flow200, 2021-11-08:
Hi @ravibhagavandas, Firstly Thanks for replying to this issue,
I want to tell you that I have already posted logs from my ESP32 (Bluetooth Peripheral Device) as well as from my mobile's logs in my first post/comment, you can check logs from there. Those logs are collected from startup at the same time from both devices.

let me know if you need anything else. 
Thanks

ravibhagavandas, 2021-11-26:
Hi @flow200 

```
2021-10-06 10:47:47.029 8233-9480/software.amazon.freertos.demo D/FRD: <-<-<- Writing to characteristic: MQTT_RX with data: a3616905617301617709
2021-10-06 10:47:47.030 8233-9480/software.amazon.freertos.demo E/FRD: Failed to write characteristic.
2021-10-06 10:47:47.031 8233-9480/software.amazon.freertos.demo E/FRD: Ble command failed to be sent OR timeout after 3000ms
2021-10-06 10:47:47.031 8233-9480/software.amazon.freertos.demo D/FRD: There's no ble command in the queue.
2021-10-06 10:47:49.166 8233-8908/software.amazon.freertos.demo D/FRD: onCharacteristicWrite for: MQTT_RX; status: 133; value: a3616905617301617709
2021-10-06 10:47:49.167 8233-8908/software.amazon.freertos.demo D/FRD: There's no ble command in the queue.
2021-10-06 10:47:49.169 8233-8911/software.amazon.freertos.demo D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=17 device=F0:08:D1:C8:30:76
2021-10-06 10:47:49.194 8233-8911/software.amazon.freertos.demo I/FRD: BLE connection state changed: 8; new state: BLE_DISCONNECTED
2021-10-06 10:47:49.195 8233-8911/software.amazon.freertos.demo I/FRD: Disconnected from GATT server due to error ot peripheral initiated disconnect.
```
From the logs on android above , the time out for writing to BLE characteristic, is because of a peripheral initiated disconnect. This could happen if the BLE connection supervision timeout expired or the ESP32 device rebooted. If there is a BLE disconnection then you should see the [error log](https://github.com/aws/amazon-freertos/blob/main/demos/network_manager/aws_iot_network_manager.c#L491) from ESP32 device. But I couldn't find the same log from your logs above.

ravibhagavandas, 2021-12-06:
Hi @flow200 

Checking back on this issue, are you still seeing this behavior? If so, could you reproduce this and send the error log as mentioned above?