aws/amazon-freertos #3403: Fleet Provisioning by Claim

BasemTarek, 2021-10-14:
Can I do fleet provisioning by claim, as described in the [aws documentation](https://docs.aws.amazon.com/iot/latest/developerguide/provision-wo-cert.html#claim-based), using Amazon FreeRTOS..?
Or I should use this [AWS IoT Fleet Provisioning Library](https://github.com/aws/fleet-provisioning-for-aws-iot-embedded-sdk)
If I should use [AWS IoT Fleet Provisioning Library](https://github.com/aws/fleet-provisioning-for-aws-iot-embedded-sdk), how can I build this library along with AFR libraries..?
Or I should do something else, what is the best practice..?

I am using Espressif ESP32-DevKitC

Thanks in advance.

muneebahmed10, 2021-10-20:
Hi @BasemTarek,

The Fleet Provisioning API is just a set of MQTT topics that you can find [here](https://docs.aws.amazon.com/iot/latest/developerguide/fleet-provision-api.html). You could subscribe and publish to these topics without using the Fleet Provisioning library, but the library was written to make obtaining the necessary topics easier.

If you are using a CMake-based build system, the AWS IoT Fleet Provisioning Library contains a [CMake file](https://github.com/aws/Fleet-Provisioning-for-AWS-IoT-embedded-sdk/blob/main/fleetprovisioningFilePaths.cmake) listing its source and header files,  which can be included in your `CMakeLists.txt` or another `.cmake` file. This will allow you to use the variables `FLEET_PROVISIONING_SOURCES` and `FLEET_PROVISIONING_INCLUDE_PUBLIC_DIRS` to set the module's sources and include directories. You can find examples of how it is done for libraries in this repository [here](https://github.com/aws/amazon-freertos/tree/main/libraries), but you should note that not all of this information is necessary for compilation, and is only there for metadata generation for the FreeRTOS Online Configuration Wizard.

If not using CMake, you just need to make sure that the library's [include directory](https://github.com/aws/Fleet-Provisioning-for-AWS-IoT-embedded-sdk/tree/main/source/include) is added to your project's include path, and that its [source file](https://github.com/aws/Fleet-Provisioning-for-AWS-IoT-embedded-sdk/blob/main/source/fleet_provisioning.c) is added to the list of project sources, however your project's include path and sources are configured.

ncohen17, 2021-11-08:
Hi @BasemTarek 

Did you get the fleet provisioning to work on the AWS FreeRTOS? 

If so, which process did you follow. 

Thanks

BasemTarek, 2021-11-09:
Hi @ncohen17 

No, it did not..
But I'm trying to follow the process used in this [demo](https://github.com/aws/aws-iot-device-sdk-embedded-C/blob/main/demos/fleet_provisioning/fleet_provisioning_with_csr/fleet_provisioning_with_csr_demo.c)
I noticed the steps of the demo and I want to refactor it to work using the FreeRTOS libraries and the [Fleet-Provisioning-for-AWS-IoT-embedded-sdk](https://github.com/aws/Fleet-Provisioning-for-AWS-IoT-embedded-sdk/tree/main)

While trying to understand the fleet provisioning process I can not understand how the permanent certificate and private key will be stored on the device's storage for future usage.

Thank you.

BasemTarek, 2021-11-16:
Hi..!