aws/amazon-freertos #3418: Intregate custom components and use in coreQMTT Agent demo.

Git-shouvik, 2021-11-01:
**Briefly summarize the issue being raised**
I want to intregate a custom component that will be used by an aws demo. I have added the component at (vendors/espressif/esp-idf/components), created the CMakelists.txt. The [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html) i used as reference.
 
(vendors/espressif/esp-idf/components) CMakelists.txt  Created.

`if(NOT DEFINED ENV{IDF_PATH})
  message(FATAL_ERROR "The IDF_PATH environment variable must point to the location of the ESP-IDF.")
endif()

set(include_dirs
    tensorflow/lite
    tensorflow/lite/c
    tensorflow/lite/core/api
    tensorflow/lite/kernels
    tensorflow/lite/kernels/internal
    tensorflow/lite/micro
    tensorflow/lite/micro/benchmarks
    tensorflow/lite/micro/kernels
    tensorflow/lite/kernels/internal 
	  tensorflow/lite/kernels/internal/reference 
  	tensorflow/lite/kernels/internal/reference/integer_ops 
	  tensorflow/lite/kernels/internal/optimized 
    tensorflow/lite/micro/memory_planner
    tensorflow/lite/micro/testing
    tensorflow/lite/schema
    tensorflow/core
    third_party/flatbuffers/include
	  third_party/gemmlowp/fixedpoint
	  third_party/gemmlowp/internal
	  third_party/kissfft
	  third_party/kissfft/tools
	  third_party/ruy
    )

set(srcs 
      "tensorflow/lite/micro/simple_memory_allocator.cc" 
      "tensorflow/lite/micro/all_ops_resolver.cc" 
      "tensorflow/lite/micro/memory_helpers.cc" 
      "tensorflow/lite/micro/recording_micro_allocator.cc" 
      "tensorflow/lite/micro/micro_error_reporter.cc" 
      "tensorflow/lite/micro/micro_time.cc" 
      "tensorflow/lite/micro/recording_simple_memory_allocator.cc" 
      "tensorflow/lite/micro/micro_string.cc" 
      "tensorflow/lite/micro/micro_profiler.cc" 
      "tensorflow/lite/micro/micro_utils.cc" 
      "tensorflow/lite/micro/micro_optional_debug_tools.cc" 
      "tensorflow/lite/micro/debug_log.cc" 
      "tensorflow/lite/micro/test_helpers.cc" 
      "tensorflow/lite/micro/micro_interpreter.cc" 
      "tensorflow/lite/micro/micro_allocator.cc" 
      "tensorflow/lite/micro/benchmarks/keyword_scrambled_model_data.cc" 
      "tensorflow/lite/micro/kernels/strided_slice.cc" 
      "tensorflow/lite/micro/kernels/prelu.cc" 
      "tensorflow/lite/micro/kernels/dequantize.cc" 
      "tensorflow/lite/micro/kernels/ethosu.cc" 
      "tensorflow/lite/micro/kernels/l2norm.cc" 
      "tensorflow/lite/micro/kernels/resize_nearest_neighbor.cc" 
      "tensorflow/lite/micro/kernels/tanh.cc" 
      "tensorflow/lite/micro/kernels/activations.cc" 
      "tensorflow/lite/micro/kernels/logistic.cc" 
      "tensorflow/lite/micro/kernels/arg_min_max.cc" 
      "tensorflow/lite/micro/kernels/pad.cc" 
      "tensorflow/lite/micro/kernels/reduce.cc" 
      "tensorflow/lite/micro/kernels/floor.cc" 
      "tensorflow/lite/micro/kernels/conv.cc" 
      "tensorflow/lite/micro/kernels/unpack.cc" 
      "tensorflow/lite/micro/kernels/fully_connected.cc" 
      "tensorflow/lite/micro/kernels/add.cc" 
      "tensorflow/lite/micro/kernels/split.cc" 
      "tensorflow/lite/micro/kernels/softmax.cc" 
      "tensorflow/lite/micro/kernels/circular_buffer.cc" 
      "tensorflow/lite/micro/kernels/svdf.cc" 
      "tensorflow/lite/micro/kernels/sub.cc" 
      "tensorflow/lite/micro/kernels/concatenation.cc" 
      "tensorflow/lite/micro/kernels/round.cc" 
      "tensorflow/lite/micro/kernels/ceil.cc" 
      "tensorflow/lite/micro/kernels/elementwise.cc" 
      "tensorflow/lite/micro/kernels/mul.cc" 
      "tensorflow/lite/micro/kernels/maximum_minimum.cc" 
      "tensorflow/lite/micro/kernels/pooling.cc" 
      "tensorflow/lite/micro/kernels/neg.cc" 
      "tensorflow/lite/micro/kernels/pack.cc" 
      "tensorflow/lite/micro/kernels/logical.cc" 
      "tensorflow/lite/micro/kernels/quantize.cc" 
      "tensorflow/lite/micro/kernels/comparisons.cc" 
      "tensorflow/lite/micro/kernels/reshape.cc" 
      "tensorflow/lite/micro/kernels/depthwise_conv.cc" 
      "tensorflow/lite/micro/memory_planner/linear_memory_planner.cc" 
      "tensorflow/lite/micro/memory_planner/greedy_memory_planner.cc" 
      "tensorflow/lite/micro/testing/test_conv_model.cc" 
      "tensorflow/lite/c/common.c" 
      "tensorflow/lite/core/api/error_reporter.cc" 
      "tensorflow/lite/core/api/flatbuffer_conversions.cc" 
      "tensorflow/lite/core/api/op_resolver.cc" 
      "tensorflow/lite/core/api/tensor_utils.cc" 
      "tensorflow/lite/kernels/internal/quantization_util.cc" 
      "tensorflow/lite/kernels/kernel_util.cc" 
      "tensorflow/lite/micro/testing/test_utils.cc")

idf_component_register(SRCS "${srcs}"
                    INCLUDE_DIRS "${include_dirs}"
                    # LDFRAGMENTS linker.lf
                    PRIV_REQUIRES driver)

# Reduce the level of paranoia to be able to compile TF sources
target_compile_options(${COMPONENT_LIB} PRIVATE
  -Wno-maybe-uninitialized
  -Wno-missing-field-initializers
    )

target_compile_options(${COMPONENT_LIB} PRIVATE -DTF_LITE_STATIC_MEMORY -DNDEBUG -O3 -Wno-nonnull -Wno-nonnull -Wno-nonnull -Wno-nonnull -Wno-nonnull -Wno-nonnull)
target_compile_options(${COMPONENT_LIB} PRIVATE $<$<COMPILE_LANGUAGE:CXX>: -std=c++11 -DTF_LITE_STATIC_MEMORY -DNDEBUG -O3 -Wno-return-type -Wno-strict-aliasing -Wno-return-type -Wno-strict-aliasing -Wno-return-type -Wno-strict-aliasing -Wno-return-type -Wno-strict-aliasing -Wno-return-type -Wno-strict-aliasing -Wno-ignored-qualifiers -Wno-return-type -Wno-strict-aliasing -Wno-ignored-qualifiers >)
target_link_libraries(${COMPONENT_LIB} PRIVATE -lm)
`

It fails in building the custom component.

**Describe the desired outcome**
The component should get build. it fails when it builds the custom component.

**System information**
- Hardware board: [ ESP-32 DevKitC ]
- IDE used: [ Ubuntu Termial ]
- Operating System: [ Linux ]
- Code version: (202107.00-3-g19c657cf0)
- Project/Demo: [ mqtt_agent]

**References**
CoreMQTT_Agent demo CMakelists.txt 

`# FreeRTOS coreMQTT demos
afr_demo_module(core_mqtt_agent)

# Set the default demo that is enabled in the downloaded package.
afr_set_demo_metadata(ID "CORE_MQTT_AGENT_DEMO")
afr_set_demo_metadata(DESCRIPTION "Examples that demonstrate the MQTT Agent library")
afr_set_demo_metadata(DISPLAY_NAME "coreMQTT Agent Demos")

# Add the CMakeLists.txt file of module to metadata list.
afr_module_cmake_files(${AFR_CURRENT_MODULE} 
    ${CMAKE_CURRENT_LIST_DIR}/CMakeLists.txt
)

afr_module_sources(
    ${AFR_CURRENT_MODULE}
    INTERFACE
        "${CMAKE_CURRENT_LIST_DIR}/mqtt_agent_task.c"
        "${CMAKE_CURRENT_LIST_DIR}/simple_sub_pub_demo.c"
        "${CMAKE_CURRENT_LIST_DIR}/subscription_manager.c"
        "${CMAKE_CURRENT_LIST_DIR}/subscription_manager.h"
        
        "${CMAKE_CURRENT_LIST_DIR}/ota_demo_core_mqtt.c"
        # "${CMAKE_CURRENT_LIST_DIR}/sntpclienttask.c"
        "${CMAKE_CURRENT_LIST_DIR}/main_functions.cc"
        "${CMAKE_CURRENT_LIST_DIR}/constants.cc"
        "${CMAKE_CURRENT_LIST_DIR}/model.cc"
        "${CMAKE_CURRENT_LIST_DIR}/output_handler.cc"
        "${CMAKE_CURRENT_LIST_DIR}/model.cc"

        # Add the header file added to the target's metadata so that it
        # is available in code downloaded from the FreeRTOS console.
        ${AFR_DEMOS_DIR}/include/aws_iot_metrics.h
)

afr_module_include_dirs(
    ${AFR_CURRENT_MODULE}
    INTERFACE
       "${AFR_DEMOS_DIR}/include"
       ${CMAKE_CURRENT_LIST_DIR}
)

afr_module_dependencies(
    ${AFR_CURRENT_MODULE}
    INTERFACE
        AFR::core_mqtt
        AFR::core_mqtt_agent
        AFR::mqtt_agent_interface
        # AFR::ota
        AFR::backoff_algorithm
        AFR::transport_interface_secure_sockets
        AFR::secure_sockets
        AFR::pkcs11_helpers
        # Add dependency on the core_mqtt_agent_demo_dependencies metadata module
        # so that FreeRTOS console shows this demo ONLY when the core MQTT Agent
        # (or another library depending on coreMQTT Agent) is selected on the console.
        AFR::core_mqtt_agent_demo_dependencies
        # AFR::ota_demo_dependencies
)
`

Can anybody reference me where is my failing to build the custom component for coreMQTT agent demo.

yanjos-dev, 2021-11-01:
Hello @Git-shouvik thank you for the detailed information. Can you please share a log of the failure?

Git-shouvik, 2021-11-02:
[CMakeOutput.log](https://github.com/aws/amazon-freertos/files/7463410/CMakeOutput.log)

ravibhagavandas, 2021-11-06:
Hello,

From the output log above, it does not indicate any errors in the cmake output. Is this log complete? What command did you use for cmake?

Also it seems you are trying to build your project from amazon freertos cmake tree instead of ESP-IDF cmake tree. So instead of placing the extra components directly in the esp idf repository, you can place them under [vendors/espressif/boards/components](https://github.com/aws/amazon-freertos/tree/main/vendors/espressif/boards/components) folder. You could create a cmakelist.txt for the component similar to one existing such as [this](https://github.com/aws/amazon-freertos/blob/main/vendors/espressif/boards/components/mbedtls/CMakeLists.txt). Finally you can build the component within the  target CMakefile using `idf_build_component` function as shown [here](https://github.com/aws/amazon-freertos/blob/main/vendors/espressif/boards/esp32/CMakeLists.txt#L631)

Git-shouvik, 2021-11-06:
Okay, But what is the difference between the components in esp-idf directory and in boards directory.

What would be solution when someone is trying to add a custom component to use in coreMQTT_Agent demo ?

Git-shouvik, 2021-11-12:
Hi, thanks for example of tensorflow.cmake

So, Here is the snapshot of the error.
[image]

Looking at the error i think it misses the definition of the function when it compiles.
I have referenced the header files using ../../ formats, according needed.
Can you tell me what should i do overcome the issue.

ravibhagavandas, 2021-11-13:
Hi,
It looks like the header file path for the declaration is not getting set correctly? Could you elaborate on the how did you specify the header file paths?
In the specific example I provided, I have used CMAKE_CURRENT_LIST_DIR ( `set(TENSOR_FLOW_DIR ${CMAKE_CURRENT_LIST_DIR}/tensorflow_src)`) so that paths are set correctly irrespective from where the cmake is included.

Git-shouvik, 2021-11-13:
libraries/tensorflow_src/tensorflow
  lite
  core
  thrid_party

the place where all the header files reside.

AniruddhaKanhere, 2021-11-24:
Hello @Git-shouvik, did you add these files to the CMake list? Similar to the example shared by @ravibhagavandas?

By doing something like:
```
set(include_dirs
<LIST ALL INCLUDE DIRECTORIES HERE ON SEPARATE LINES>
)

afr_module_include_dirs(
    ${AFR_CURRENT_MODULE}
    PUBLIC
    ${include_dirs}
)
```

Let me know if this helps.

Git-shouvik, 2021-11-25:
Hi, I have included and named the file as the sample example given by @ravibhagavandas.
After that when i build the project, I get the error shown on screenshot.

AniruddhaKanhere, 2021-11-26:
Hello,
Do you have this code in a public repository so that we can also test that out in our end?

AniruddhaKanhere, 2021-11-26:
Also, can you update your code to the latest version of tensor flow lite? The main branch seems ahead of the version you are using seeing from the code in the error you mentioned above.

Thanks

Git-shouvik, 2021-11-26:
Hi,
Here the files I am using and the .cmake.
https://github.com/Git-shouvik/RTOS_TFLITE.git

AniruddhaKanhere, 2021-11-26:
Hello @Git-shouvik, in the cmake file [here](https://github.com/Git-shouvik/RTOS_TFLITE/blob/master/tensor_flow.cmake#L3) I see that you have set: `set(TENSOR_FLOW_DIR ${CMAKE_CURRENT_LIST_DIR}/tensorflow_src)`. However, in your case, there is no directory named `tensorflow_src`. You should modify that line to: `set(TENSOR_FLOW_DIR ${CMAKE_CURRENT_LIST_DIR})`.

Git-shouvik, 2021-11-26:
Hi, 
Let me tell you the directory structure.
amazon-freertos/libraries
 inside of libraries: 
      tensor_flow.cmake
      tensorflow_src/
          tensorflow
          third_party

I think so, I able to clarify you the structure.

AniruddhaKanhere, 2021-11-26:
I see. So what is inside `tensorflow_src` and in `tensorflow`?
I am asking because in this repository - https://github.com/Git-shouvik/RTOS_TFLITE/tree/master I do not see a folder called `tensorflow_src`.

Thanks

Git-shouvik, 2021-11-27:
> Hello, Do you have this code in a public repository so that we can also test that out in our end? It seems like an error due to non proper inclusion of header file(s).
> 
> As far as I can tell, `reference_integer_ops` is a namespace and that needs a header file which defines it as such.

I just shared the contents inside tensorflow_src in the repository. Those are tensorflow and thridparty.

AniruddhaKanhere, 2021-11-28:
l see. Now I understand your directory structure. Thanks for the explanation and thanks for bearing with me.

So, is this the structure?
```
|- libraries
    |- tensorflow_src
        |- tensorflow
        |- thirdparty
        |- tensor_flow.cmake
```

If so, then the file tensor_flow.cmake is misplaced. It should be put one level up - like so.
```
|- libraries
    |- tensor_flow.cmake
    |- tensorflow_src
        |- tensorflow
        |- thirdparty
```

If not so, then can you copy your cmake output in a file and attach it to your comment? Thanks.

AniruddhaKanhere, 2021-11-28:
Hello @Git-shouvik, thanks for that explanation. And thank you for bearing with me.

If I clone your repository as is in my `libraries/tensorflow_src/` folder and include the `libraries/tensorflow_src/tensor_flow.cmake` to `libraries/CMakelists.txt`, it complains that it cannot find several files since there is not directory named `libraries/tensorflow_src/tensorflow_src/`. To fix that, I modified [this line](https://github.com/Git-shouvik/RTOS_TFLITE/blob/master/tensor_flow.cmake#L3) to look like this: 
`set( TENSOR_FLOW_DIR ${CMAKE_CURRENT_LIST_DIR} )`.

Now, CMake is happy.

But, when I build that (using `cmake --build build`), I am getting errors different from the ones you are getting. Your version of tensor flow is not the same as the version in the Espressif repository [here](https://github.com/espressif/tensorflow). Is that intentional? There seem to be a lot less files in your repository. I have uploaded my changes [here](https://github.com/AniruddhaKanhere/amazon-freertos/tree/TF_LITE). Let me know if that looks right.

Git-shouvik, 2021-11-28:
I followed this [tutorial](https://medium.com/@dmytro.korablyov/first-steps-with-esp32-and-tensorflow-lite-for-microcontrollers-c2d8e238accf) on intregating TFLITE. So thats why i have only limited files.

The above tutorial worked successfully with esp-idf, but i could not able to intregate with amazon-freertos demo.

> Hi @Git-shouvik
> 
> Another (possibly better) solution to integrate a new component could be to create a new AFR module target. A module can be either a library or a demo or a test. To define tensorflow as a library target you can add the corresponding source code under `libraries/` folder and create a cmake file defining the module its sources and dependencies. Please take a look at the example I have created for tensor flow here: [ravibhagavandas@7e6df69](https://github.com/ravibhagavandas/amazon-freertos/commit/7e6df69655139081e4bb1832a1e49dd65d3bc7e5)
> 
> Hope this helps.

@ravibhagavandas  suggested me to try the method, but still i face the error.
There is yet no solution i found of not able to build.

I hope this helps you and me to come a conclusion of the issue.
**Can you share the errors you get.**

AniruddhaKanhere, 2021-11-29:
Hello @Git-shouvik,

I tried what @ravibhagavandas has suggested and it works quite well. I did not run the project but it compiled successfully without any error.

Here is what I did.
1. Clone amazon freertos repository (`git clone --recurse-submodules https://github.com/aws/amazon-freertos.git ./amazonFreeRTOS`).
2. Add the tensorflow submodule from espressif to the libraries folder of the repo. (`git submodule add https://github.com/espressif/tensorflow.git ./libraries/tensorflow_src`). Note: call this command from the `amazonFreeRTOS` folder that was created in step 1 above.
3. Then, to install some thirdparty sources I followed this guide: https://www.tensorflow.org/lite/guide/build_arm using the `Bazel` tool - you'll need to install it see details [here](https://docs.bazel.build/versions/4.2.1/install-ubuntu.html). Note: Run the Bazel tool from `./libraries/tensorflow_src/` directory.
4. I had to copy some files into required location after running the above Bazel tool.
5. Run cmake from the `amazonFreeRTOS` directory like so: `cmake  -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S. -Bbuild -GNinja`
6. Build the project: `cmake --build build/`
7. Done!

Let me know if that process works for you. With incomplete sources it will be very difficult to figure out the error. It will be best alternative to clone the whole tensor-flow repo from espressif.

Git-shouvik, 2021-12-03:
Hi,
**For bazel tool to build:** bazel build --config=elinux_aarch64 -c opt //tensorflow/lite/c:libtensorflowlite_c.so

**Generate Build Files:** cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B ./YOUR_BUILD_DIRECTORY -DAFR_ENABLE_ALL_MODULES=1 -DAFR_ENABLE_TESTS=0 -GNinja

**Build the Project:**    ninja -C ./YOUR_BUILD_DIRECTORY -j8

And errors i enconter
[tflite.txt](https://github.com/aws/amazon-freertos/files/7652390/tflite.txt)

AniruddhaKanhere, 2021-12-06:
Hello @Git-shouvik, as I mentioned in the step 4 above, I had to copy some files from one location to where it is required.

Can you try to find `flatbuffers.h` and copy the flatbuffers folder in `../../../libraries/tensorflow_src/tensorflow/lite/schema`? After that, the error should go away. You will see more errors like this one. Just copy the required folders/files at the required location.

Let me know if that doesn't work.

Thanks.

Git-shouvik, 2021-12-09:
Hi,
I works as you  said to follow the process. Thanks a lot for your support.

when I create the tensorflow_lite model and use in one of the demos of amazon-freertos. The extra files I add in the demo(coreMQTT_Agent) each file has path("tensorflow/lite/kernels/internal/compatibility.h") to the header, but it still it fails to build, does the [tensor_flow.cmake](https://github.com/Git-shouvik/RTOS_TFLITE/blob/master/tensor_flow.cmake) created in amazon-freertos/libraries does not solve the issue of "No such file or directory".  Though the file asked by TF_lite files is present.

Then how can i make that possible, the path to be visible when TF_lite files are used in demo(coreMQTT)

closed 2021-12-09 by Git-shouvik

AniruddhaKanhere, 2021-12-09:
Hello @Git-shouvik, I am glad that this worked out for you!

Feel free to reopen this issue or a new one if you ever need anymore clarification/support.

Regards,
Aniruddha