aws/amazon-freertos #3419: [General] Greengrass v2 with Local Shadow and client devices

smartnimbus, 2021-11-02:
HI All
I have the Greengrass v2 working with freertos following these steps https://docs.aws.amazon.com/greengrass/v2/developerguide/client-devices-tutorial.html
However, has anyone manage to use the Local Shadow with client devices?
https://docs.aws.amazon.com/greengrass/v2/developerguide/interact-with-shadows.html

ravibhagavandas, 2021-11-06:
Hi @smartnimbus 
You need to have some way of mapping the mqtt messages to the local shadow component running on Greengrass. You could have these options:
1. Greengrass provide IPC APIs for interacting with local shadow. You could deploy a moquitee broker on the device as per the docs [here](https://docs.aws.amazon.com/greengrass/v2/developerguide/mqtt-broker-moquette-component.html) and then map the messages received to local shadow APIS.
2. Also from green grass docs, it seems the IPC APIs can also be performed using local pub/sub topics, so you could deploy an mqtt bridge and then bridge the device shadow topic from mqtt topic to local pubsub topic as mentioned [here](https://docs.aws.amazon.com/greengrass/v2/developerguide/mqtt-bridge-component.html).

For additional details, you can post your queries in greengrass github repository [here](https://github.com/aws-greengrass/aws-greengrass-nucleus).

smartnimbus, 2021-11-09:
Thank you ravibhagavandas

closed 2021-11-09 by ravibhagavandas