aws/amazon-freertos #3421: Fleet Provisioning by Claim

BasemTarek, 2021-11-07:
Hi,
I asked a question in this issue #3403 but no one answered it

After doing device provisioning using the MQTT API, how the persisted private keys and certificates are stored on the esp32 flash memory (NVS) ..?

I found this [core_pkcs11_pal.c](https://github.com/aws/amazon-freertos/blob/1da0cb644601f80b5e55301d347fcd2d1c726a03/vendors/espressif/boards/ports/pkcs11/core_pkcs11_pal.c) but I do not know how can I use it..?

Thank you.

_Originally posted by @BasemTarek in https://github.com/aws/amazon-freertos/issues/3403#issuecomment-957786069_

ravibhagavandas, 2021-11-08:
Hi @BasemTarek 

To key pair can be generated or a certificate key pair can be provisioned using the PKCS11 APIs rather than directly calling the PKCS11 PAL APIs. The PAL APIs are the underlying porting implementations for mbedtls PKCS11 implementation. You can refer to the demo [here](https://github.com/aws/aws-iot-device-sdk-embedded-C/blob/main/demos/fleet_provisioning/fleet_provisioning_with_csr/fleet_provisioning_with_csr_demo.c) to see how a key pair is generated using PKCS11 and the corresponding CSR is registered using the fleet provisioning library.

BasemTarek, 2021-11-09:
Thank you @ravibhagavandas,

Generally I do not understand anything about PKCS11 or mbedtls, I am sorry for my stupidity.
I can understand C programming and RTOS.

But..

I think I understand the [demo](https://github.com/aws/aws-iot-device-sdk-embedded-C/blob/main/demos/fleet_provisioning/fleet_provisioning_with_csr/fleet_provisioning_with_csr_demo.c), I understand the general flow of the demo to on-board devices at scale, as described in this whitepaper [Fleet Provisioning by Claim](https://d1.awsstatic.com/whitepapers/device-manufacturing-provisioning.pdf).

> **Device Logic**
Devices must implement the logic necessary to publish and subscribe to Fleet
Provisioning MQTT topics, accept the permanent credentials, and write the credentials
to secure storage. 

So my question now is how the device (in my case it is ESP-32) will write the unique private key and certificate to its permanent storage for future usage when trying to connect again and again to AWS cloud..?
How can I do that using AFR..?

Thanks.

BasemTarek, 2021-11-16:
Hi..!

tianmc1, 2021-11-16:
Hi @BasemTarek ,

Sorry about the delayed response. In the fleet provisioning [demo](https://github.com/aws/aws-iot-device-sdk-embedded-C/blob/main/demos/fleet_provisioning/fleet_provisioning_with_csr/fleet_provisioning_with_csr_demo.c), it makes use of corePKCS11 library, which stores the certificate in non-volatile memory. Note that corePKCS11 library is a mock of PKCS11 implemented in software with mbedTLS. The purpose of the corePKCS11 library is to provide a non hardware specific PKCS11 implementation that allows for rapid prototyping and development. For actual production use cases, corePKCS11 will need to be replaced by a PKCS11 implementation provided by the crypto-processor vendor. You can read more about corePKCS11 [here](https://github.com/FreeRTOS/corePKCS11/blob/main/README.md).

BasemTarek, 2021-11-19:
Hi @tianmc1 

Thank you for your response.

I can not understand how the corePKCS11 library is used to store the certificate in the NVM.

> For actual production use cases, corePKCS11 will need to be replaced by a PKCS11 implementation provided by the **crypto-processor** vendor.

Must I use a crypto-processor to do fleet provisioning, or I can do  fleet provisioning simply using esp32..?

According to the following:
> [The purpose of the PAL is to allow our software-based reference implementation of PKCS #11 to access onboard flash for storage of persisted private keys and certificates.](https://github.com/aws/amazon-freertos/blob/30e14afb403d78732388461f14f1d4195e72e4ad/libraries/abstractions/pkcs11/ReadMe.md#when-to-implement-a-pal)

I understand that I can use the [corePKCS11 library](https://github.com/FreeRTOS/corePKCS11) to generate the certificate key pair, and use [core_pkcs11_pal.c](https://github.com/aws/amazon-freertos/blob/1da0cb644601f80b5e55301d347fcd2d1c726a03/vendors/espressif/boards/ports/pkcs11/core_pkcs11_pal.c) to store the unique private key and certificate to the NVM of the esp32.

If that is not true, how can I write the unique private key and certificate to the ESP-32's permanent storage..?

aggarg, 2021-11-22:
> If that is not true, how can I write the unique private key and certificate to the ESP-32's permanent storage..?

To be clear, are you asking how can you write to non-volatile storage? If so, you can use [nvs flash](https://github.com/espressif/esp-idf/tree/master/components/nvs_flash/include) APIs from ESP. The `core_pkcs11_pal.c` you mentioned above uses the same. 

Thanks.

BasemTarek, 2021-11-22:
Thank you @aggarg

OK that is fine, but is this the best practice..?

According to @tianmc1 

> In the fleet provisioning demo, it makes use of corePKCS11 library, which stores the certificate in non-volatile memory.

How the [fleet provisioning demo](https://github.com/aws/aws-iot-device-sdk-embedded-C/blob/main/demos/fleet_provisioning/fleet_provisioning_with_csr/fleet_provisioning_with_csr_demo.c) uses corePKCS11 library to store the certificate in non-volatile memory, can you guide me please..?

Also
Can I do the fleet provisioning using esp32 only without using a crypto-processor..?

tianmc1, 2021-11-22:
Hi @BasemTarek ,

In the fleet provisioning demo, it calls a helper function [loadCertificate()](https://github.com/aws/aws-iot-device-sdk-embedded-C/blob/299c98b8e7eb256741bbe51341060e6410f8cc6e/demos/fleet_provisioning/fleet_provisioning_with_csr/fleet_provisioning_with_csr_demo.c#L613), which is defined [here](https://github.com/aws/aws-iot-device-sdk-embedded-C/blob/299c98b8e7eb256741bbe51341060e6410f8cc6e/demos/fleet_provisioning/fleet_provisioning_with_csr/pkcs11_operations.c#L1162). loadCertificate() calls another helper function [provisionCertificate()](https://github.com/aws/aws-iot-device-sdk-embedded-C/blob/299c98b8e7eb256741bbe51341060e6410f8cc6e/demos/fleet_provisioning/fleet_provisioning_with_csr/pkcs11_operations.c#L680) which gets [function pointers](https://github.com/aws/aws-iot-device-sdk-embedded-C/blob/299c98b8e7eb256741bbe51341060e6410f8cc6e/demos/fleet_provisioning/fleet_provisioning_with_csr/pkcs11_operations.c#L725) for PKCS11 and calls PKCS11 functions to [store the certificate](https://github.com/aws/aws-iot-device-sdk-embedded-C/blob/299c98b8e7eb256741bbe51341060e6410f8cc6e/demos/fleet_provisioning/fleet_provisioning_with_csr/pkcs11_operations.c#L772).

aggarg, 2021-11-22:
> OK this is fine, but is this the best practice to write the private key and certificate to the ESP-32's NVM..?

This is the way if you want to write to NVS flash without PKCS11. Otherwise, you can use PKCS11 as @tianmc1 mentioned. 

> Can I do the fleet provisioning using esp32 only without using a crypto-processor..?

Fleet provisioning does not require crypto-processor, so yes you can.

Thanks.

BasemTarek, 2021-11-22:
Now I got it, thank you very much.

closed 2021-11-22 by BasemTarek