aws/amazon-freertos #3422: [General] Loading device certificates from external flash

laszlopataki, 2021-11-07:
**Briefly summarize the issue being raised**
Currently, the certificates are embedded inside the app binary however, for production purposes, I want to separate them from binaries(OTA is easier, production is easier). 
Currently, these files are embedded into Network Manager which does not accept input context. 
The certificate is embedded into the NetworkManager. https://github.com/aws/amazon-freertos/issues/3408

What is the best practice to use NetworkManager in a custom application that consumes AFR? 
Should `/freertos/demos/network_manager/aws_iot_netowork_manager.c` moved to project root dir and modified there to load vertificated from NVS (my_project/network_manager)? 
Maybe improve NetworkManager to be able to accept a context?

Years ago it was possible to give a SPIFFS path as the keyCLIENT_CERTIFICATE_PEM, and ESP32 loaded it from there. 
Would this be the solution? 

**Describe the desired outcome**
I am able to load certificated from external flash, so that I can separate my application binary from the embedded certificate. 

**System information**

Hardware board: [ esp32]
IDE used: [ esp-idf]
Code version: 202107 latest release
Project/Demo: [ custom CMAKE application consuming AFR ]

ravibhagavandas, 2021-11-09:
Hi @laszlopataki 
To achieve this, you can take a look at the reference implementation of key provisioning in developer mode [here](https://github.com/aws/amazon-freertos/tree/main/demos/dev_mode_key_provisioning)
The README shows different options for key provisioning. The first option (which is recommended only for prototype/lab testing) shows how to take a keypair configured in `aws_clientcredential_keys.h` and import this into the PKCS11 slot which is backed by underlying flash or a secure element. API [vDevModeKeyProvisioning](https://github.com/aws/amazon-freertos/blob/main/demos/dev_mode_key_provisioning/include/aws_dev_mode_key_provisioning.h#L76) can be called to perform this. Once provisioned the API can be removed from the code and the TLS connection always reads the certificate/key pair from designated slot backed by the flash location. You can leverage on this example to develop provisioning through a serial port or other means based on your hardware capabilities.
The code in network manager does not use the `keyCLIENT_CERTIFICATE_PEM`  and `keyCLIENT_PRIVATE_KEY_PEM` credentials and can be ignored.

ravibhagavandas, 2021-11-15:
We will close this issue due to inactivity. Please open a new issue if you have any questions.

closed 2021-11-15 by ravibhagavandas