aws/amazon-freertos #3427: Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed).

SarahHamed, 2021-11-17:
Hello Team,

I've tried to run the MQTT and Greengrass demos, and both worked well.

But when trying to write my own application using the demo's functions I've got errors.

My project structure is as the following:
- amazon-freertos
- freertos-conigs
- src
  - main.c
- CMakeLists.txt 

In the main, I've tried 4 scenarios. 
1- Called the MQTT'S DEMO function -RunCoreMqttMutualAuthDemo()- directly in the main.
2- Created a task to run the MQTT'S DEMO function.
3- Called the Greengrass DEMO function directly in the main -discoverGreengrassCore().
4- Created a task to run the Greengrass DEMO function.

The first case was the only one to succeed. The others usually fail after connecting to the wifi.

 **The logs for each attempt.** 
[greengrass-direct-call-in-main.txt](https://github.com/aws/amazon-freertos/files/7554347/greengrass-direct-call-in-main.txt)
[greengrass-task.txt](https://github.com/aws/amazon-freertos/files/7554352/greengrass-task.txt)
[greengrass-task2.txt](https://github.com/aws/amazon-freertos/files/7554353/greengrass-task2.txt)
[mqtt-fn-direct-call-in-main.txt](https://github.com/aws/amazon-freertos/files/7554356/mqtt-fn-direct-call-in-main.txt)
[greengrass-mqtt-direct-call-in-main.txt](https://github.com/aws/amazon-freertos/files/7554348/greengrass-mqtt-direct-call-in-main.txt)
[mqtt-task.txt](https://github.com/aws/amazon-freertos/files/7554357/mqtt-task.txt)
**I always get the error "Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)." in all the cases.**

**- For the greengrass task attempt in the logs it fails here:**

```
[Greengrass] [aws_greengrass_discovery.c:165] [INFO] [Greengrass_Discovery_Demo] GG

Re-enable cpu cache.
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed).

```
**- Then when I tried to run it without a task it kept resetting until it ended as the following:**

```
14 3779 [main] [aws_greengrass_discovery.c:165] [INFO] [Greengrass_Discovery_Demo] GG

15 3780 [main] Unexpected error (4294967290) from dns_gethostbyname_addrtype() while resolving (xxxxxxxxxxxxxxx-ats.iot.eu-west-1.amazonaws.com)!
16 3781 [main] ERROR! Failed to resolve host address: ServerHost=xxxxxxxxxxxxxxx-ats.iot.eu-west-1.amazonaws.com
17 3782 [main] LwIP connect fail -1 118

18 3782 [main] ERROR! SOCKETS_Connect call failed: ServerAddress=0, Port=64288, ReturnCode=-1

19 3783 [main] About to close socket.

20 3783 [main] Socket closed.

21 3784 [main] Stack high watermark for discovery helper task: 1388.

22 3784 [main] JSON request could not connect to end point

23 3785 [main] [aws_greengrass_discovery.c:177] [INFO] [Greengrass_Discovery_Demo] GGD_JSONRequestStart

24 3786 [main] [aws_greengrass_discovery_demo.c:615] [ERROR] [Greengrass_Discovery_Demo] Auto-connect: Failed to retrieve Greengrass address and certificate.

```
**- When I use the greengrass without a task, in the logs it fails here:**

```
[main] [aws_greengrass_discovery.c:165] [INFO] [Greengrass_Discovery_Demo] GG

E (9909) PKCS11: failed nvs get file size 4354 0

abort() was called at PC 0x40080fe0 on core 0

```
**Here is the app_main() function.**
[the-main-fn.txt](https://github.com/aws/amazon-freertos/files/7554490/the-main-fn.txt)

Please let me know what could be causing all of this, and if there is a better way to follow to use these functions in my main.

**System information**
- Hardware board: ESP32
- Operating System: [ Windows ]

aggarg, 2021-11-18:
From a first glance, the main function looks okay. Can you share your code, which I can use to repro it? 

Also, can you not directly edit the main function in https://github.com/aws/amazon-freertos/blob/main/vendors/espressif/boards/esp32/aws_demos/application_code/main.c#L127? Alternatively, you can use esp-idf and our standalone libraries to write your application. 

Thanks.

SarahHamed, 2021-11-21:
Thank you for your response.

Well, we haven't edited the main function here directly:
https://github.com/aws/amazon-freertos/blob/main/vendors/espressif/boards/esp32/aws_demos/application_code/main.c#L127
We created another main.c file in an external src directory and used CMake to compile and flash. Following the instructions here to create our own application.  https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#getting_started_espressif_cmake_project 
Here is how my project's structure looks like. 
- amazon-freertos
- freertos-conigs
- src
    - main.c
- CMakeLists.txt

We didn't edit or use the demo's main. We tried to imitate it and created another external one, trying to create our own app. If there is another way using esp-idf. Please guide me on how to do that.

aggarg, 2021-11-21:
> If there is another way using esp-idf. Please guide me on how to do that.

You can consume esp-idf and our libraries in your application. Here is one layout to achieve that:

```
/
├─ esp-idf (submodule)/
├─ libs/
│  ├─ coreMQTT (submodule)/
│  ├─ Device-Shadow-for-AWS-IoT-embedded-sdk (submodule)/
│  ├─ Device-Defender-for-AWS-IoT-embedded-sdk (submodule)/
│  ├─ ota-for-aws-iot-embedded-sdk (submodule)/
│  ├─ ...
├─ app/
│  ├─ application_src1.c
│  ├─ application_src2.c

```

BasemTarek, 2021-11-23:
As we are using greengrass we are not able to use the AWS IoT Device SDK for Embedded C.

So we need to use AFR.

The issue is that this error `Guru Meditation Error: Core 0 panic'ed (Cache disabled but cached memory region accessed` appears when trying to run our application as described [above](https://github.com/aws/amazon-freertos/issues/3427#issue-1056063964).

**To reproduce**
Steps to reproduce the behavior:

1. Flash and monitor

**Code to reproduce the bug**
[main.zip](https://github.com/aws/amazon-freertos/files/7586773/main.zip)

**Additional context**
Amazon's demo working fine.

Thank you.

AniruddhaKanhere, 2021-11-24:
Hello @BasemTarek,

I see that you are using `vTaskDelay` in `app_main`. I am not sure that it is correct to do so since the scheduler is not running. This should not cause the issue that you are facing - however, it would be better if you remove the delay calls.

SarahHamed, 2021-11-25:
Removing all the calls for **vTaskDelay** in the main maybe have enhanced it just a little. But it still resets with the same **Guru** error. Here are the logs.
[gg-without-delays-in-main2.txt](https://github.com/aws/amazon-freertos/files/7605522/gg-without-delays-in-main2.txt)

**[main]** gets printed only when we call **vStartGreenGrassDiscoveryTask()** directly in the main without creating a task.

Here are some of the backtraces. I don't know if it can help.
[gg-backtraces.txt](https://github.com/aws/amazon-freertos/files/7605526/gg-backtraces.txt)

Thank you.

aggarg, 2021-11-26:
> [main] gets printed only when we call vStartGreenGrassDiscoveryTask() directly in the main without creating a task.

Thank you for sharing backtraces. I see that the crash is observed during TLS handshake. One possible reason can be stack overflow. Can you try increasing the size of the stack which calls this function? 

Also, enable malloc failed hook, if you have not already - https://www.freertos.org/a00016.html

Thanks.

AniruddhaKanhere, 2021-11-26:
Also @SarahHamed, I take back my statement about 'one should not use delays in app_main'. Because the `app_main` function is called from a task after the scheduler is started as seen [here](https://github.com/espressif/esp-idf/blob/master/components/freertos/FreeRTOS-Kernel/portable/port_common.c#L127).
Apologies.

SarahHamed, 2021-11-28:
@AniruddhaKanhere No worries. Thank you for your response.

I tried what @aggarg suggested and did the following:

**1- Increased the stack size used in xTaskCreate to be multiplied by 16 instead of 12. In the file "aws_demo_config.h":**
`#define democonfigGREENGRASS_DISCOVERY_TASK_STACK_SIZE               ( configMINIMAL_STACK_SIZE * 16 ) `

> xTaskCreate( cal,
>                      "Greengrass",
>                       democonfigGREENGRASS_DISCOVERY_TASK_STACK_SIZE,
>                       NULL,
>                       democonfigDEMO_PRIORITY,
>                       NULL ) != pdPASS )

**2- Added to the file "amazon-freertos\vendors\espressif\boards\esp32\aws_demos\config_files\FreeRTOSConfig.h"** 

```
#define configUSE_MALLOC_FAILED_HOOK      1
#define configCHECK_FOR_STACK_OVERFLOW     1
```

**3- Added these to the main.c**
```
extern void vApplicationMallocFailedHook( void );
extern void vApplicationStackOverflowHook( TaskHandle_t xTask,  char * pcTaskName );
```

As they were defined here:
- https://github.com/aws/amazon-freertos/blob/fdcedfa1e973797da0c0d0d7211d8ecc9102d5ad/demos/demo_runner/iot_demo_freertos.c#L406
- https://github.com/aws/amazon-freertos/blob/fdcedfa1e973797da0c0d0d7211d8ecc9102d5ad/demos/demo_runner/iot_demo_freertos.c#L430

But unfortunately, I couldn't find in the logs any of these errors:

- "ERROR: stack overflow in task ".
- "ERROR: Malloc failed to allocate memory"

4- Enabled **INCLUDE_uxTaskGetStackHighWaterMark** in **aws_greengrass_discovery_demo.c** 

[gg-logs.txt](https://github.com/aws/amazon-freertos/files/7613601/gg-logs.txt)

Thank you.

laukik-hase, 2021-11-29:
Hello @SarahHamed @BasemTarek,

Could you share the entire project (maybe host it on GitHub) that gives you the error? It would help us understand the issue better.

aggarg, 2021-11-30:
Looking at the logs, it seems like a WatchDog reset. What are all the tasks in the system. Is it possible that the Idle task is starved? 

Additionally, can you share the complete code in a fork or repo so that I can give it a try? 

Thanks.

SarahHamed, 2021-12-01:
Here it is https://github.com/SarahHamed/amazon-freertos-proj 

Thank you.

laukik-hase, 2021-12-02:
Hello @SarahHamed,

Could you check out this patch for your project - [gme_cache_accessed.zip](https://github.com/aws/amazon-freertos/files/7639041/gme_cache_accessed.zip)?

Also, there are some issues with the way you have used the demo in your `main.c`. Could you follow the template for executing demos from an external project as given [here](https://github.com/shubhamkulkarni97/amazon-freertos-examples)?

SarahHamed, 2021-12-07:
Hello @laukik-hase ,

Thank you for your help and the patch file. It has been useful.

I've applied it locally. Here are the updated files after applying it:
[CMakeLists.txt](https://github.com/aws/amazon-freertos/files/7671636/CMakeLists.txt)
[main.txt](https://github.com/aws/amazon-freertos/files/7671664/main.txt)

1. In the main file. Why `_initialize() `was called before `vDevModeKeyProvisioning()`. Does the order matter?
2. If I should follow the template [here](https://github.com/shubhamkulkarni97/amazon-freertos-examples) to not get the guru error. I am not sure how to deal with a task function that never returns. The function `_threadRoutineWrapper()` returns and the task gets deleted.

My question. What if we don't want the task function to return and to loop forever, as we were trying to implement in the main using while(1){} . How can we do that without facing the guru error?

Thank you.

aggarg, 2021-12-08:
> What if we don't want the task function to return and to loop forever, as we were trying to implement in the main using while(1){} . How can we do that without facing the guru error?

If your task keeps looping and never gives up CPU and it has higher priority than Idle task, then Idle task won't get a chance to run. As a result, Idle task won' t be able to kick WatchDog and you will get a WatchDog reset. Why do you want to loop endlessly? Your task should block when there is nothing useful to do instead of looping endlessly. Try adding a `vTaskDelay` after each iteration at least.

pvyawaha, 2021-12-16:
Hello @SarahHamed ,

Do you still need help on this topic or this issue can be closed?

SarahHamed, 2021-12-17:
Hello @pvyawaha & @aggarg,

We downloaded another version of amazon freertos from the console which should be more stable than cloning the repo. 
We haven't tried the same main that we were using. Since we know now that this wasn't the best way. So we are now trying to use the template that you've suggested.

Thank you all for your help. I believe we can close it.

aggarg, 2021-12-17:
Thank you. Feel free to open another issue if you need any help.

closed 2021-12-17 by aggarg