aws/amazon-freertos #3431: Fleet Provisioning by claim.

buneshwarEc, 2021-11-22:
Hi ,

I'm new to iot and aws.
After going through the process of aws fleet provisioning.
[https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_esp32wroom-32se.html#setup-esp32wroom-32se-idf42-download-freertos](url)

AND

[https://github.com/aws/amazon-freertos/blob/main/vendors/microchip/secure_elements/app/example_trust_chain_tool/README.md](url)

In the above readme, 
I'm stuck at the following part.

[image]

Questions -
1 : what's the point of going to the freertos code if there's nothing to do.
2 : How to build the demo application and run the code in the 2nd step.

End Goal(Any help if possible) -
Able to update certificates on esp32 dynamically by using provisioning by claim.

aggarg, 2021-11-24:
Would you try following the instructions mentioned here: https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html

Thanks.

buneshwarEc, 2021-11-25:
> Would you try following the instructions mentioned here: https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html
> 
> Thanks.

Hi ,
Thanks for that i tried it and it worked.
For the fleet provisioning part i'm using the python sdk , so my question is ,
1 : Where shall i put the python sdk for it to run as when i try idf.py build in sdk it says idf.py command not found.

aggarg, 2021-11-25:
> For the fleet provisioning part i'm using the python sdk 

Which python SDK? Do you want to run python on ESP32?

If you want to run Fleet Provisioning on ESP32, then you can refer to this example - https://github.com/aws/aws-iot-device-sdk-embedded-C/tree/main/demos/fleet_provisioning/fleet_provisioning_with_csr

buneshwarEc, 2021-11-26:
> Which python SDK? Do you want to run python on ESP32?
>
> If you want to run Fleet Provisioning on ESP32, then you can refer to this example - [...]

Hi ,
Thanks for that ,
I went through the Readme and built the demo , There wasn't a mention on how to flash that demo onto the device.
Can you guide me through that?

aggarg, 2021-11-26:
If you are using ESP32, you need to familiarize yourself with building and flashing any software on ESP32. You can [follow this guide](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html) for that as I mentioned before. 

Once you understand that, you need to add fleet provisioning. You can use the demos in [this repo](https://github.com/aws/amazon-freertos) as a starting point and use the above mentioned fleet provisioning demo as a reference. 

Thanks.

jasonpcarroll, 2021-12-06:
@buneshwarEc It seems this issue has been resolved. If this issue has not been resolved, feel free to re-open.

closed 2021-12-06 by jasonpcarroll