aws/amazon-freertos #3433: [General] How to figure out wether the device is connected to the broker ?

Devyami-Apcon, 2021-11-23:
**Briefly summarize the issue being raised**
I am trying to implement a feature where my app knows that my device is connected to the mqtt broker or not. According to my research MQTT LWT messages is what i need to implement using device shadows.  Under the heading "Detecting a device is connected" in this [https://docs.aws.amazon.com/iot/latest/developerguide/device-shadow-comms-app.html#thing-connection](url) it is described how to implement it. However, how to trigger the aws broker to send a "disconnected" status when a sudden error occurs, say power failure etc. In brief how does point number 5 in the link provided works? 

Any help would be appreciated. Thanks

**System information**
- Hardware board: [esp32 ]
- IDE used: [ esp-idf ]
- Operating System: [ MacOS ]
- Code version: (run ``git describe --tags`` to find it)

aggarg, 2021-11-24:
You specify the LWT topic and message during connect - https://github.com/FreeRTOS/coreMQTT/blob/main/source/include/core_mqtt.h#L417

Your app should subscribe to the same LWT topic. When your device disconnects, the AWS IoT will publish the same message that you specified during connect on the LWT topic.

Thanks.

Devyami-Apcon, 2021-11-27:
Thank you @aggarg for the answer, it cleared out most of the doubts and it is working as expected. 

However, I have a further extended doubt. For LWT message, reserved topics can not be used and thus i have to use a rule to republish the message to my shadow. The rule I am using is universal for all my devices by using certain sql functions. However the current rule is not working as expected because of policy mismatch, what exact policy shall I attach to my current rule ? 

Here is how my rule look like.
[image]

Here is how my role's policy look like.
[image]

aggarg, 2021-11-29:
You cannot use substitution template expressions in your policy document. You need to update your policy to use the topic `$aws/things/*/shadow/name/settings/update`. 

Regarding the use of `*` instead of `+`, the following is from this document: https://docs.aws.amazon.com/iot/latest/developerguide/pub-sub-policy.html 
```
The MQTT wildcards + and # are treated as literal strings in an AWS IoT Core policy for MQTT clients. To specify wildcard characters in topic names and topic filters in AWS IoT Core policies for MQTT clients, you must use *.
```

Thanks.

jasonpcarroll, 2021-12-06:
@Devyami-Apcon Has this issue been resolved?

Devyami-Apcon, 2021-12-08:
@jasonpcarroll @aggarg 

Yes thank you for the support.

closed 2021-12-10 by paulbartell