aws/amazon-freertos #3434: Developer-mode key provisioning

BasemTarek, 2021-11-24:
Hi

In the [Developer-mode key provisioning](https://docs.aws.amazon.com/freertos/latest/userguide/dev-mode-key-provisioning.html), how can I import the certificate into the device..?

> **Certificate Import**
> _**With the certificate issued, the next step is to import it into your device.**_ You will also need to import your Certificate Authority (CA) certificate, since it is required in order for first-time authentication to AWS IoT to succeed when using JITP. In the `aws_clientcredential_keys.h` file in your project, set the keyCLIENT_CERTIFICATE_PEM macro to be the contents of deviceCert.pem and set the keyJITR_DEVICE_CERTIFICATE_AUTHORITY_PEM macro to be the contents of rootCA.pem.

How can I insert the edited file `aws_clientcredential_keys.h` back to the device, this requires me to re-build the project and re-flash the device, which will lead to losing the generated public & private keys from the device..?

Is this demo require me to use a dedicated crypto-processor (like `ATECC608A chip`) to be accomplished..?

**System information**
I am using Espressif ESP32-DevKitC

aggarg, 2021-11-24:
You do not need a crypto-processor to write the key to the non-volatile storage. You can keep a partition in the flash and write the credentials to it using the esp nvs APIs.

AniruddhaKanhere, 2021-11-27:
Hello @BasemTarek, did that answer your question?

BasemTarek, 2021-11-27:
Hi @aggarg 
Thank you for your help, I really appreciate it.
I do not ask about the need for the crypto-processor to write the key to the non-volatile storage, 

How can I write to the esp32 after editing the `aws_clientcredential_keys.h` file without losing the generated public & private keys from the device..?

> You can keep a partition in the flash and write the credentials to it using the esp nvs APIs.

How can I do this..?

Thank you.

AniruddhaKanhere, 2021-11-30:
@BasemTarek, you may find these links useful as @aggarg mentioned above.

How to create flash partition: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html
Flash partition APIs: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spi_flash.html#flash-partition-apis
NVS APIs: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html

There are some links to examples on those pages as well.

BasemTarek, 2021-11-30:
@aggarg 

> Are you asking that once you have generated a key pair on the ESP32 device at run time, how to store that and use it for further TLS connections?

Yes this is my question, but check the following paragraph from [here](https://docs.aws.amazon.com/freertos/latest/userguide/dev-mode-key-provisioning.html)

> **Certificate Import**
> **With the certificate issued, the next step is to import it into your device**. You will also need to import your Certificate Authority (CA) certificate, since it is required [...]

I understand that I should import the new certificate into the device, also disable the temporary key generation setting `#define keyprovisioningFORCE_GENERATE_NEW_KEY_PAIR 0`
So, I need to use `aws_clientcredential_keys.h ` and import it to the ESP-32.

Are you mean that re-flashing the device after doing the mentioned edits will not affect the generated keys stored in the ESP-32's NVS..?

Thank you @AniruddhaKanhere, I'll check these links.

BasemTarek, 2021-12-02:
Hi,

Update:

I found that re-flashing the device did not affect the generated keys stored in the ESP-32's NVS.
Because the generated public key, printed on the terminal and did not changed when I re-flashed the esp32.

But the connection to AWS IoT Core failed, using the generated private key and the registered certificate..!

aggarg, 2021-12-03:
> But the connection to AWS IoT Core failed, using the generated private key and the registered certificate..!

At what step did the connection fail? Can you check the policy attached to the certificate?

jasonpcarroll, 2021-12-06:
@BasemTarek Did you resolve your issue?

BasemTarek, 2021-12-07:
@jasonpcarroll Not yet, I am sorry for my late response, @aggarg I'll share my logs within few hours.
My policy is very generic for testing:
```
{
  "Version": "2012-10-17",
  "Statement": [
    {
      "Effect": "Allow",
      "Action": [
        "iot:Publish",
        "iot:Subscribe",
        "iot:Receive"
      ],
      "Resource": [
        "*"
      ]
    }
  ]
}
```

aggarg, 2021-12-07:
You are missing "iot:Connect" in your policy actions. For **TESTING ONLY**, can you try using iot:*:
```
{
  "Version": "2012-10-17",
  "Statement": [
    {
      "Effect": "Allow",
      "Action": [
        "iot:*"
      ],
      "Resource": [
        "*"
      ]
    }
  ]
}
```

BasemTarek, 2021-12-07:
Thank you @aggarg, thank you very much.

The demo worked as expected.

**Abstract:**

- The first reason for the issue was that I erased the ESP-32's flash memory when re-flashing it. 
- & I used an incorrect policy.

As I said [here](https://github.com/aws/amazon-freertos/issues/3434#issuecomment-984566421) I should re-flash the device without erasing it first.

BasemTarek, 2021-12-07:
> In this case, the device will not be signing the certificate request (that is, the Certificate Service Request or **CSR**) because **the X.509 encoding logic required for creating and signing a CSR has been excluded from the FreeRTOS demo projects to reduce ROM size**. Instead, for lab testing purposes, create a private key on your workstation and use it to sign the CSR.

Is it possible to create a CSR using Amazon FreeRTOS, which library should I use..?

aggarg, 2021-12-07:
esp-idf has mbedTLS which you should be able to use to create CSR.

BasemTarek, 2021-12-08:
OK that's fine, thank you.

closed 2021-12-08 by BasemTarek