aws/amazon-freertos #3436: [General] Example of Custom Projects Nordic

vlockhead, 2021-11-26:
**Briefly summarize the issue being raised**
I am trying to compile my own application outside of the amazon freertos source tree with using a CMakeLists.txt

**Describe the desired outcome**
cmake ends with a success.

**System information**
- Hardware board: nordic.nrf52840-dk
- IDE used:not a the moment
- Operating System: [ Linux ]
- Code version: 202107.00-46-g6b5f95024
- Project/Demo: [ Own Project ]
- If your project is a custom application, please add the relevant code snippet(s) in the section titled `"Code/Steps to reproduce the current status"`.

**Code/Steps to reproduce the current status**
1. Clone the amazon-freertos repository
2. Create a CMakeLists.txt with the following code snippet

```cpp
cmake_minimum_required(VERSION 3.13)

project(active_gps)

add_executable(active_gps sources/main.c)

# Tell IDF build to link against this target
set(IDF_PROJECT_EXECUTABLE active_gps)

include_directories(my_app PRIVATE "amazon-freertos/demos/include")
include_directories(my_app PRIVATE "amazon-freertos/demos/network_manager")
include_directories(active_gps PRIVATE "amazon-freertos/vendors/nordic/nRF5_SDK_15.2.0")

# AFR_BOARD tells which board we need to target.
set(AFR_BOARD nordic.nrf52840-dk CACHE INTERNAL "")

add_subdirectory(amazon-freertos)

target_link_libraries(active_gps PRIVATE AFR::ble)
target_link_libraries(active_gps PRIVATE AFR::ble_hal)
target_link_libraries(active_gps PRIVATE AFR::platform)
target_link_libraries(active_gps PRIVATE AFR::posix)
target_link_libraries(active_gps PRIVATE 3rdparty::pkcs11)
```
3.Run cmake
```cpp
cmake -DCMAKE_TOOLCHAIN_FILE=$amazon-freertos/tools/cmake/toolchains/arm-segger.cmake -DAFR_TOOLCHAIN_PATH=/usr/share/segger_embedded_studio_for_arm_5.50a/gcc/arm-none-eabi/bin -S . -B build

cmake --build build --clean-first
```

**Additional context**
I am getting the following error when running cmake:

AniruddhaKanhere, 2021-12-02:
Hello @vlockhead, apologies for the late response. I was busy figuring out an easy way to achieve what you want without much change to the amazon-freertos repository.

Instead of your CMakeLists.txt, you should use the one I have attached.
[CMakeLists.txt](https://github.com/aws/amazon-freertos/files/7637616/CMakeLists.txt)

It builds on your version with some additions. Except this, you'd need to do the below steps to get the code to compile.

1. Modify the Nordic CMakeLists.txt and add the line `set(exe_target my_gps)` [here](https://github.com/aws/amazon-freertos/blob/e0ed68b8dc870809c887b8b4432951aad825cab7/vendors/nordic/boards/nrf52840-dk/CMakeLists.txt#L49). Note that I am going to raise a PR shortly so that if `exe_target` is already defined, it won't be set again.
2. Use this command to build CMake: `cmake -DCMAKE_TOOLCHAIN_FILE=amazon-freertos/tools/cmake/toolchains/arm-segger.cmake -DVENDOR=nordic -DBOARD=nrf52840-dk -DAFR_ENABLE_TESTS=0 -DAFR_ENABLE_DEMOS=0 -DAFR_TOOLCHAIN_PATH=/<YOUR_PATH_HERE>/arm_segger_embedded_studio_v560_linux_x64_nordic/gcc/arm-none-eabi/bin -S . -B build`

In point 2., we explicitly disable demos and tests so that amazon-freertos is built as a library instead of making it as an executable.
 
Let me know if this method works for you.

Thanks,
Aniruddha

vlockhead, 2021-12-02:
Hello, 

1. I have updated amazon-freertos repository to the latest main branch.
2. I have copied a paste the cmake file that you have linked
2. I have implemented step one of your solution
3. Here is my bash file to compile

```
# Create the makefile
build_path=build2
arm_segger_cmake_path=amazon-freertos/tools/cmake/toolchains/arm-segger.cmake
arm_segger_compiler_path=/opt/arm_segger_embedded_studio_v568_linux_x64_nordic/gcc/arm-none-eabi/bin

cmake -DCMAKE_TOOLCHAIN_FILE=$arm_segger_cmake_path -DVENDOR=nordic -DBOARD=nrf52840-dk -DAFR_ENABLE_TESTS=0 -DAFR_ENABLE_DEMOS=0 -DAFR_TOOLCHAIN_PATH=$arm_segger_compiler_path -S . -B $build_path

# Build 
cmake --build $build_path --clean-first
```

4. Here is the output [amazonFreertosOuput.txt](https://github.com/aws/amazon-freertos/files/7641979/amazonFreertosOuput.txt)

I dont know If I need the specific v560 of the nordic compiler?

I must say I am a bit worried about trying to create a custom project with amazon freertos. Is there a documentation that I missed that would have explained what you just answered me? If not that means that I will need to ask question at every bump? (You have answered pretty fast I must say)

I also got caught by a bug with the cmake version. Here is the link to the bug https://githubmemory.com/repo/aws/amazon-freertos/issues/3220.

Thank you,

Vincent

AniruddhaKanhere, 2021-12-02:
Hello Vincent,

I am not sure about the exact version number issues. I just downloaded the first one that I saw (Not a very scientific method, I know 🙂). 
I have disabled bootloader as well by commenting [this](https://github.com/aws/amazon-freertos/blob/e0ed68b8dc870809c887b8b4432951aad825cab7/vendors/nordic/boards/nrf52840-dk/CMakeLists.txt#L479) line. Also my `main.c` had a main function which didn't do much - it was a placeholder for testing - maybe that's the issue on my end?

You wrote:
> I must say I am a bit worried about trying to create a custom project with amazon freertos. Is there a documentation that I missed that would have explained what you just answered me?

I would advise against using `amazon-freertos` to create a custom project. The documentation is lacking a bit in that domain and doing that is 'uncharted waters'. It was mostly designed to be used as an application rather than a library in my opinion.

I would instead advise you to use our [libraries](https://www.freertos.org/libraries/categories.html) directly by submoduling them in your project. That would be easier and cleaner. We currently do not have an example for using all the libraries at once directly, but we are working on that. I expect that to be simple - if you have any questions you can raise an issue here about that or in [FreeRTOS Forums](https://forums.freertos.org).

AniruddhaKanhere, 2021-12-03:
Hello,

I might have not made myself clear. I am NOT asking you to add amazon-freertos as a submodule to your project.
What I am saying is just submodule all the components you need.

For example, if you need the [FreeRTOS-Plus-TCP library](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP), just add that into your project as a submodule.
If you need [coreMQTT](https://github.com/FreeRTOS/coreMQTT) then submodule it into your project.
And whe you need [PKCS11](https://github.com/amazon-freertos/pkcs11/tree/9304ef6842d08d881ccd1daa8d3a428f3a9671ee) just submodule that into your project.

And keep your source files as they are.

Let me know if that makes sense. As far as using your main is concerned, I shall give it a try.

Thanks,
Aniruddha

vlockhead, 2021-12-06:
Hello,

I see what you mean, I will try to do that. I think it will be the easiest way to build a custom project.

Thank you,

Vincent

pvyawaha, 2021-12-17:
Hello @vlockhead ,

Did that helped or you have more questions regarding this topic?

vlockhead, 2021-12-20:
Hello @pvyawaha,

It did answer my questions, I having no more questions.

 I am just a bit sad that the integration of this project as a custom project is not easily feasible.

closed 2021-12-20 by vlockhead