aws/amazon-freertos #3438: [General] <replace with your title>

BlairArrowes, 2021-12-08:
**Briefly summarize the issue being raised**
panic IllegalInstruction is occurring during vDevModeKeyProvisioning function call in an esp32s2 custom project.
A delay function (delay_us) in the esp-idf esp_flash_write function is calling xTaskGetIdleTaskHandle.
I suspect that because Icache is disabled during the write operation, the cpu cannot access the function in flash, which causes the panic.
As I don't currently have a JTAG adapter, I cannot confirm that this is the case.

My question is *why* does it occur in this custom project and not in the demo?
Is there some configuration that's done by the demos to change how delay_us works?

**Describe the desired outcome**
Successful key provisioning without a panic occurring.
Eventually key provisioning will be performed with ESP32S2 built-in security hardware, somehow.

**System information**
- Hardware board: esp32s2-saola1
- IDE used: vscode
- Operating System: Windows
- Code version: 20210526_Archive-139-gb7151fe03
- Project/Demo: custom application

**References**
main function:
```
/*
 * FreeRTOS V1.4.7
 * Copyright (C) 2020 Amazon.com, Inc. or its affiliates.  All Rights Reserved.
 */

#include "iot_config.h"

/* FreeRTOS includes. */

#include "FreeRTOS.h"
#include "task.h"

/* AWS System includes. */
#include "iot_system_init.h"
#include "iot_logging_task.h"
#include "aws_clientcredential.h"
#include "aws_dev_mode_key_provisioning.h"

#include "nvs_flash.h"

#include "driver/uart.h"
#include "driver/gpio.h"

#include "iot_uart.h"

#include "esp_system.h"

//#include "core_mqtt_agent.h"

/* Demonstrate use of component foo. */
#include <foo.h>

/* Logging Task Defines. */
#define mainLOGGING_MESSAGE_QUEUE_LENGTH (32)
#define mainLOGGING_TASK_STACK_SIZE (configMINIMAL_STACK_SIZE * 4)
#define mainDEVICE_NICK_NAME "Espressif_Demo"

/* Static arrays for FreeRTOS+TCP stack initialization for Ethernet network connections
 * are use are below. If you are using an Ethernet connection on your MCU device it is
 * recommended to use the FreeRTOS+TCP stack. The default values are defined in
 * FreeRTOSConfig.h. */

/**
 * @brief Initializes the board.
 */
static void prvMiscInitialization(void);

[... 324 lines cut ...]

0x4009825b: nvs_set_blob at C:\Users\Blair\ESP_AWS\build/../freertos/vendors/espressif/esp-idf/components/nvs_flash/src/nvs_api.cpp:441

0x4008d634: PKCS11_PAL_SaveObject at C:\Users\Blair\ESP_AWS\build/../freertos/vendors/espressif/boards/ports/pkcs11/core_pkcs11_pal.c:204

0x4008d89b: PKCS11_PAL_DestroyObject at C:\Users\Blair\ESP_AWS\build/../freertos/vendors/espressif/boards/ports/pkcs11/core_pkcs11_pal.c:477

0x4008c4f2: C_DestroyObject at C:\Users\Blair\ESP_AWS\build/../freertos/libraries/abstractions/pkcs11/corePKCS11/source/portable/mbedtls/core_pkcs11_mbedtls.c:2580

0x4008ddda: xDestroyProvidedObjects at C:\Users\Blair\ESP_AWS\build/../freertos/demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c:765

0x4008df15: xDestroyDefaultCryptoObjects at C:\Users\Blair\ESP_AWS\build/../freertos/demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c:815

0x4008df56: xProvisionDevice at C:\Users\Blair\ESP_AWS\build/../freertos/demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c:1099

0x4008e256: vAlternateKeyProvisioning at C:\Users\Blair\ESP_AWS\build/../freertos/demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c:1309

0x4008e285: vDevModeKeyProvisioning at C:\Users\Blair\ESP_AWS\build/../freertos/demos/dev_mode_key_provisioning/src/aws_dev_mode_key_provisioning.c:1369

0x400830fc: app_main at C:\Users\Blair\ESP_AWS\build/../main/main.c:111

0x400965b4: main_task at C:\Users\Blair\ESP_AWS\build/../freertos/vendors/espressif/esp-idf/components/esp32s2/cpu_start.c:466
```
**Additional context**
Add any remaining information not mentioned in the previous responses.

BlairArrowes, 2021-12-09:
Just got JTAG debugging going, it doesn't appear to be caused solely by delay_us.
All crashes so far have been caused between spiflash_start(chip) and spiflash_end(chip) in the esp_flash_read function.
curiously, those functions are used to stop and start the rtos/interrupts/cache.
It makes sense that an interrupt call would cause a panic in this area, but I'm still mystified as to why the systick timer is still generating interrupts.
the correct esp_intr_noniram_disable() function in the esp-idf/components/esp32s2/intr_alloc.c file is being called, everything *should* just work.

BlairArrowes, 2021-12-10:
Two threads running main would do it (threads 2 and 6). Not sure how this is possible or if it's just the debugger freaking out.
[image]

paulbartell, 2021-12-10:
@BlairArrowes One potential source of this issue could be a stack overflow. Have you tried increasing the stack size of the task accessing the nvs filesystem?

BlairArrowes, 2021-12-13:
@paulbartell I just tested increasing the "main" task from 4k to 16k stack size, along with increasing the "TmrSvc" from 2k to 3k. No change. Though I am now getting some DoubleExceptionVector errors which is interesting.

paulbartell, 2021-12-15:
Generally this type of error occurs along with some type of memory corruption. Do you have any other tasks running on the system?

pvyawaha, 2021-12-20:
@BlairArrowes ,

Do you still need support on this topic or ticket can be closed ?