aws/amazon-freertos #3440: Separate sendMessageToGGC() from discoverGreengrassCore()

SarahHamed, 2021-12-15:
Hello Team,

I am trying to create a task to publish messages only. So In the demo file [aws_greengrass_discovery_demo.c](https://github.com/aws/amazon-freertos/blob/main/demos/greengrass_connectivity/aws_greengrass_discovery_demo.c#L518 ). When I try to separate the fn `sendMessageToGGC() `and create a task in `vStartGreenGrassDiscoveryTask()`. I get the error: **Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.** It happens when the function `prvCreateMQTTConnectionWithBroker()` gets called.

But when I comment the task creation for `sendMessageToGGC()`.  `prvCreateMQTTConnectionWithBroker()` works without any problems and I get the message: **"MQTT connection established with the broker.".**

**Here are the changes:**

1. Declared `MQTTContext_t xMQTTContext = { 0 };` as a global variable. So we don't pass any parameters to `discoverGreengrassCore()`
2. Separated sendMessageToGGC() from discoverGreengrassCore();
 ```
int vStartGreenGrassDiscoveryTask( bool awsIotMqttMode,
                                   const char * pIdentifier,
                                   void * pNetworkServerInfo,
                                   void * pNetworkCredentialInfo,
                                   const IotNetworkInterface_t * pNetworkInterface )
{
    /* Exit status of this program. */
    int status = EXIT_SUCCESS;

    /* Unused parameters */
    ( void ) awsIotMqttMode;
    ( void ) pIdentifier;
    ( void ) pNetworkServerInfo;
    ( void ) pNetworkCredentialInfo;

    status = discoverGreengrassCore();
   xTaskCreate( sendMessageToGGC,
		        "sendMsg",
		        democonfigDEMO_STACKSIZE,
		        NULL,
		        democonfigDEMO_PRIORITY,
		        NULL );

    return status;
}
```

Would you please help me understand why this error happens.
I also get the same error when I try to call the function sendMessageToGGC() directly but using LogInfo() before it.

Here is the whole file.
[aws_greengrass_discovery_demo.txt](https://github.com/aws/amazon-freertos/files/7718889/aws_greengrass_discovery_demo.txt)

paulbartell, 2021-12-15:
@SarahHamed: Most likely, democonfigDEMO_STACKSIZE is too small for the function that is being called. 

The best way to determine the required stack size is by algorithmically or manually validating the maximum stack size necessary for any combination of function calls. 

GCC has the "-fstack-usage" option which will calculate the required minimum stack size for a given function during compilation. You can then parse this information along with the function callgraph to determine worst case stack usage ( avstack.pl is a common script for doing this ).

A quick alternative is to use set the stack much higher than necessary and then use the uxTaskGetStackHighWaterMark function to check stack usage for a given task optimize the size form from there. The code fragment below could be a good starting point:
```
        /* Report on space efficiency of this demo task. */
        #if ( INCLUDE_uxTaskGetStackHighWaterMark == 1 )
            {
                configPRINTF( ( "Heap low watermark: %u. Stack high watermark: %u.\r\n",
                                xPortGetMinimumEverFreeHeapSize(),
                                uxTaskGetStackHighWaterMark( NULL ) ) );
            }
        #endif
```

pvyawaha, 2021-12-20:
Hello @SarahHamed ,

Did that resolved your issue ?

SarahHamed, 2021-12-21:
Hello @pvyawaha & @paulbartell,

No, Increasing the stack's size didn't solve it. I want to point out that I also get the same error when I call the function without creating a task for it. 
I get the error when I call `LogInfo()` before `sendMessageToGGC()`. But when I don't use `LogInfo()` before I call the function. It works.
But it never works when I create a task, even after I increased the stack size.

```

int vStartGreenGrassDiscoveryTask( bool awsIotMqttMode,
                                   const char * pIdentifier,
                                   void * pNetworkServerInfo,
                                   void * pNetworkCredentialInfo,
                                   const IotNetworkInterface_t * pNetworkInterface )
{
    /* Exit status of this program. */
    int status = EXIT_SUCCESS;

    /* Unused parameters */
    ( void ) awsIotMqttMode;
    ( void ) pIdentifier;
    ( void ) pNetworkServerInfo;
    ( void ) pNetworkCredentialInfo;

    status = discoverGreengrassCore();
    LogInfo( ( "starting sendMessageToGGC()" ) );
    sendMessageToGGC();

    return status;
}

```
Thank you.

archigup, 2022-01-05:
Hi, apologies for the late reply. Do you still have the issue?
If so, could you let us know the platform you are running it on? That would help with reproducing the issue.

SarahHamed, 2022-01-06:
Yes, we still have it.

We could produce it in Ubuntu. Our hardware is ESP32.
We downloaded freertos from the console.  Our project structure is similar to this [one](https://github.com/shubhamkulkarni97/amazon-freertos-examples/blob/feature/idf_v4.2/src/main.c) with the same main.c file.
We built the files using cmake as mentioned [here](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#getting_started_espressif_cmake_project)

Thanks.

SarahHamed, 2022-01-17:
Hello all, 

Any updates here? 

Thanks.

dachalco, 2022-01-26:
Hi @SarahHamed 

Could you provide the full output log from your device? I'd like to reproduce the issue so I can debug locally.

SarahHamed, 2022-01-26:
[Link to code](https://github.com/SarahHamed/amazon-freertos/blob/main/demos/greengrass_connectivity/aws_greengrass_discovery_demo.c)
[gg-log.txt](https://github.com/aws/amazon-freertos/files/7945993/gg-log.txt)

Thank you.

aggarg, 2022-02-10:
Closing this one as the related issue https://github.com/aws/amazon-freertos/issues/3450 is resolved. Feel free to re-open or create a new one if needed.