aws/amazon-freertos #3444: Add support for ESP32-S2 USB CDC to use serial console

findmyname666, 2021-12-30:
I cannot flash esp32-s2 without failure. Here is the log from **idf.py -p /dev/ttyACM0 flash**:
```
Writing at 0x000d5b31... (97 %)
Writing at 0x000d6562... (98 %)
Writing at 0x000d6f4e... (98 %)
Writing at 0x000d7a42... (99 %)
Writing at 0x000d86d5... (99 %)
Writing at 0x000d944b... (100 %)
Wrote 760864 bytes (494876 compressed) at 0x00020000 in 7.4 seconds (effective 825.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
ERROR: ESP32-S2 chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this error, set --after option to 'no_reset'.
CMake Error at run_cmd.cmake:14 (message):
  esptool.py failed
Call Stack (most recent call first):
  run_esptool.cmake:21 (include)

FAILED: CMakeFiles/flash 
cd /home/findmyname/Documents/iot/aws/amazon-freertos/vendors/espressif/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH="/home/findmyname/Documents/iot/aws/amazon-freertos/vendors/espressif/esp-idf" -D ESPTOOLPY="/home/findmyname/.espressif_aws/python_env/idf4.2_py3.9_env/bin/python /home/findmyname/Documents/iot/aws/amazon-freertos/vendors/espressif/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32s2" -D ESPTOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="/home/findmyname/Documents/iot/aws/amazon-freertos/build" -P /home/findmyname/Documents/iot/aws/amazon-freertos/vendors/espressif/esp-idf/components/esptool_py/run_esptool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

After manual reset the "test" app seems to run.
So far I'm unable to use **monitor** command.

This can be solved by adding support for USB Communication Device Class (CDC) to implement the serial console. Here is the [ESP documentation](https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32s2/api-guides/usb-console.html) with more details.
However this seems to be documented only for `v4.3` so I'm not sure if that works for `v4.2`.
I actually used it in `v4.3` and it works!

P.S.: I'm not really expert either in electronics and / or esp-idf build system so I cannot help or provided more technical details. 

Thx

alfred2g, 2022-01-04:
Thanks for reporting, we will be looking at this shortly

dachalco, 2022-01-11:
Hi @findmyname666 

Exactly which esp32-s2 board are you using (eg. esp32-s2-saola, esp32-s2-devkitc, etc.)?

findmyname666, 2022-01-18:
Hi @dachalco @alfred2g ,

Thanks for feedback.

I'm using [these Olimex boards (ESP32-S2-DevKit-Lipo-USB)](https://www.olimex.com/Products/IoT/ESP32-S2/ESP32-S2-DevKit-Lipo-USB/open-source-hardware) which is compatible with **esp32-s2-saola-1**.

BR

dachalco, 2022-01-18:
Hi @findmyname666 

USB support should already be provided by esp-idf. We should be updating our IDF version to `v4.3` soon, afte which these errors won't pop up. In the meantime, it appears you're at least able to flash the application. As for monitoring, instead of using `idf.py monitor` you can use another usb serial monitoring program such as `teraterm`, `serial`, `minicom`, etc. to monitor the device UART output. If alternative serial monitoring does not work, please re-open the ticket.

closed 2022-01-18 by dachalco