aws/amazon-freertos #3445: [BUG] MQTTNoMemory on MQTT_Agent when reconnect to WIFI ( 2nd execute on prvConnectAndCreateDemoTasks()) on ESP32

eelaix, 2022-01-01:
**Describe the bug**
        while ( 1 ) 
        {
            ulTaskNumber++;  
            prvConnectAndCreateDemoTasks( NULL );
        }

in prvConnectAndCreateDemoTasks function, two tasks (prvMQTTAgentTask, prvSimpleSubscribePublishTask) will be created,
when error happen the two tasks will be deleted ( vTaskDelete(NULL)),  and prvConnectAndCreateDemoTasks will be executed again.

when I reboot(restart) wifi device, error happened and prvConnectAndCreateDemoTasks function re executed on expected, but in the 2nd execution, ERROR happens at :
prvSubscribeToTopic()
    xCommandAdded = MQTTAgent_Subscribe( &xGlobalMqttAgentContext,   -->>>>>>>>>ERROR happened here!!!!!
                                         &xSubscribeArgs,
                                         &xCommandParams );
    if( xCommandAdded == MQTTSuccess )
    {
        /* Wait for acks to the subscribe message - this is optional but done here
         * so the code below can check the notification sent by the callback matches
         * the ulNextSubscribeMessageID value set in the context above. */
        xCommandAcknowledged = prvWaitForCommandAcknowledgment( &ulNotifiedValue );
    }
    else
    {
        LogError( ( "Failed to enqueue subscribe command. %s", MQTT_Status_strerror( xCommandAdded ) ) );
    }

THE ERROR SHOWS:
No command structure available.
Failed to enqueue subscribe command. MQTTNoMemory

====================PARAMS tries for:================
#ifndef MQTT_AGENT_NETWORK_BUFFER_SIZE
    #define MQTT_AGENT_NETWORK_BUFFER_SIZE    ( 10000 ).   ====> defaults is 5000
#endif
democonfigSIMPLE_SUB_PUB_TASK_STACK_SIZE      SETTING to :::   configMINIMAL_STACK_SIZE * 6

WHAT THE MATTER?  anybody can help me? THX.

**System information**
- Hardware board: [ ESP32S2 ]
- IDE used: [ VSCode ]
- Operating System: [ MacOS ]
- Code version: (202107.00-48-g30f4b7257)
- Project/Demo: [ coreMQTT_Agent demo ]
- If your project is a custom application, please add the relevant code snippet(s) in the section titled `"Code to reproduce bug"`.

**Expected behavior**
esp32 reconnect to wifi in 2nd execution , MQTTNoMemory ERROR happens

**Screenshots or console output**

**Steps to reproduce bug**

**Code to reproduce bug**

```cpp
```

**Additional context**

tianmc1, 2022-01-05:
Hi @eelaix ,

Thank you for reaching out. Can you point me to which code base your project is based on? Is it https://github.com/FreeRTOS/coreMQTT-Agent-Demos/tree/main/source?

eelaix, 2022-01-06:
@tianmc1  yes, just this code.
but I have made some changes based on it.
prvMQTTAgentTask used to subscribe(receive) message from IoT server and 
vStartSimpleSubscribePublishTask (prvSimpleSubscribePublishTask) used to publish device shadow (send) to server.
I think I have resolved this problem in some ways about:
1.  when prvMQTTAgentTask fails ("Could not reconnect to MQTT broker") exit  prvSimpleSubscribePublishTask right now
2.  make ulTaskNumber global to make different task name of prvSimpleSubscribePublishTask
3. change the task PRIORITY carefully, for example:
`
        /* Create an instance of the MQTT agent task. Give it higher priority than the
         * subscribe-publish tasks so that the agent's command queue will not become full,
         * as those tasks need to send commands to the queue. */
`
4. change the task stack size carefully

tianmc1, 2022-01-07:
Here's my analysis of where the MQTTNoMemory error comes from. [MQTTAgent_Subscribe](https://github.com/FreeRTOS/coreMQTT-Agent/blob/main/source/core_mqtt_agent.c#L1130) calls a [createAndAddCommand](https://github.com/FreeRTOS/coreMQTT-Agent/blob/main/source/core_mqtt_agent.c#L702) helper function. This function tries to get a command struct from the [pool](https://github.com/FreeRTOS/coreMQTT-Agent-Demos/blob/main/lib/FreeRTOS/mqtt-agent-interface/freertos_command_pool.c#L54). It returns a MQTTNoMemory error when it cannot get a new struct from the pool. I'm guessing between deleting and recreating the prvMQTTAgentTask, some initializations are not done properly, which causes the issues. Notably, some context variables are static. They are automatically initialized to 0 on start up, but when the task is deleted and recreated, they may not be 0. 

Is there a reason you prefer to delete the task when network is disconnected, instead of letting the task handle reconnect?

eelaix, 2022-01-08:
@tianmc1 yes, you are right, I had check the static variables and try to make them global.
I think another important thing is keep the two tasks synchronized, when network is not ready, do not send message to server again is the best way.

eelaix, 2022-01-08:
@tianmc1 I had read and test the code in 
demos/ota/ota_demo_core_mqtt/ota_demo_core_mqtt.c
when "Could not reconnect to MQTT broker" appear, the hardware rebooted by configASSERT

closed 2022-01-09 by eelaix