aws/amazon-freertos #3449: Possibility of configuring OTA Tasks to only run at startup and terminate these tasks if no update is currently available.

CaptainMJx, 2022-01-18:
I'm using a memory limited device (256kb RAM) and it doesn't seem possible to run a Shadow and OTA at the same time with this limitation. I don't have enough heap even after various memory optimizations around stacks etc.. As a workaround I am wondering if there is a way to configure the OTA Tasks so that all OTA related tasks terminate if no update is currently available. Following this, trigger my actual application tasks to start, like Shadow and other user tasks,

If that is possible it should be an easy job to allow the shadow to handle reboot command, which would trigger the OTA job when required. This basically provides the same functionality as running the OTA permanently but at a much lower memory requirement.

Thanks,
Marcus

archigup, 2022-01-18:
Hi, you can terminate the OTA tasks by calling OTA_Shutdown and then waiting for OTA_GetState to return OtaAgentStateStopped. After that you can start your application tasks.

CaptainMJx, 2022-01-19:
Hi, that looks very promising to me, just one follow up. Before I can call OTA_Shutdown I would need to make sure that it has completed handshake with cloud and that there is no update currently available, or alternatively that the OTA failed somehow.

I would not want to shut it down before it had a chance to check for updates. Is there a way to verify this, I'm guessing OTA_GetState is the way, but not sure if there is a suitable state(s) for this purpose, are you able to provide some guidance?

pvyawaha, 2022-01-26:
Hello @CaptainMJx ,

We are working on a PR for adding an application callback event when no more jobs are queued. https://github.com/aws/ota-for-aws-iot-embedded-sdk/pull/416

ActoryOu, 2022-05-24:
Hello @CaptainMJx ,
PR [aws/ota-for-aws-iot-embedded-sdk#427](https://github.com/aws/ota-for-aws-iot-embedded-sdk/pull/427) has been merged to OTA repo. Will merge back to amazon-freertos later.

ActoryOu, 2022-06-13:
Fixed in #3495 .

closed 2022-06-13 by ActoryOu