aws/amazon-freertos #3450: Guru Error when calling sendMessageToGGC()

SarahHamed, 2022-01-24:
Hello,

This is a follow-up to the issue #3440. 

I get the following error **Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.** When I call the function `sendMessageToGGC();` in the greengrass Demo.

I call this function outside the function ` discoverGreengrassCore()`. I provided the code in the aforementioned issue.

Please let me know if you can help with this.

aggarg, 2022-01-25:
Would you please do the 2 things:
1. Please share the complete log.
2. Would you please share your code which shows the problem so that we can also try to repro. May be you can put it in your fork and share a link?

Thanks.

SarahHamed, 2022-01-26:
[Link to code](https://github.com/SarahHamed/amazon-freertos/blob/main/demos/greengrass_connectivity/aws_greengrass_discovery_demo.c)
[gg-log.txt](https://github.com/aws/amazon-freertos/files/7945976/gg-log.txt)

Thank you.

aggarg, 2022-01-29:
I ran your code and I am able to run it successfullly:
```
6 451 [iot_thread] [aws_greengrass_discovery_demo.c:537] [INFO] [Greengrass_Discovery_Demo] Attempting automated selection of Greengrass device

E (5817) PKCS11: failed nvs get file size 4354 0
7 852 [iot_thread] JSON file retrieval completed

8 852 [iot_thread] About to close socket.

9 852 [iot_thread] Socket closed.

10 853 [iot_thread] Stack high watermark for discovery helper task: 5736.

13 853 [iot_thread] port 8883

E (9397) PKCS11: failed nvs get file size 4354 0
14 1281 [iot_thread] About to close socket.

15 1282 [iot_thread] Socket closed.

16 1282 [iot_thread] Stack high watermark for discovery helper task: 3048.

17 1282 [iot_thread] [aws_greengrass_discovery_demo.c:547] [INFO] [Greengrass_Discovery_Demo] Greengrass device discovered.

18 1282 [iot_thread] [aws_greengrass_discovery_demo.c:439] [INFO] [Greengrass_Discovery_Demo] Creating a TLS connection to *******************.

E (13657) PKCS11: failed nvs get file size 4354 0
19 1721 [iot_thread] [core_mqtt.c:886] [INFO] [MQTT] Packet received. ReceivedBytes=2.

20 1721 [iot_thread] [core_mqtt_serializer.c:970] [INFO] [MQTT] CONNACK session present bit not set.

21 1722 [iot_thread] [core_mqtt_serializer.c:912] [INFO] [MQTT] Connection accepted.

22 1722 [iot_thread] [core_mqtt.c:1563] [INFO] [MQTT] Received MQTT CONNACK successfully from broker.

23 1722 [iot_thread] [core_mqtt.c:1829] [INFO] [MQTT] MQTT connection established with the broker.

24 1722 [iot_thread] [aws_greengrass_discovery_demo.c:477] [INFO] [Greengrass_Discovery_Demo] An MQTT connection is established with ***************.

25 1722 [iot_thread] Heap low watermark: 84480. Stack high watermark: 3048.

26 1722 [iot_thread] [aws_greengrass_discovery_demo.c:602] [INFO] [Greengrass_Discovery_Demo] starting sendMessageToGGC()

27 1723 [iot_thread] [aws_greengrass_discovery_demo.c:517] [INFO] [Greengrass_Discovery_Demo] MQTT PUBLISH message sent successfully to the broker: Topic=freertos/demos/ggd

28 1873 [iot_thread] [aws_greengrass_discovery_demo.c:517] [INFO] [Greengrass_Discovery_Demo] MQTT PUBLISH message sent successfully to the broker: Topic=freertos/demos/ggd

29 2023 [iot_thread] [aws_greengrass_discovery_demo.c:517] [INFO] [Greengrass_Discovery_Demo] MQTT PUBLISH message sent successfully to the broker: Topic=freertos/demos/ggd

30 2173 [iot_thread] [INFO ][DEMO][21730] memory_metrics::freertos_heap::before::bytes::217896

31 2173 [iot_thread] [INFO ][DEMO][21730] memory_metrics::freertos_heap::after::bytes::84480

32 2173 [iot_thread] [INFO ][DEMO][21730] memory_metrics::demo_task_stack::before::bytes::8920

33 2173 [iot_thread] [INFO ][DEMO][21730] memory_metrics::demo_task_stack::after::bytes::3048

I (23397) wifi:state: run -> init (0)
I (23397) wifi:pm stop, total sleep time: 14783471 us / 18898568 us

I (23397) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (23397) WIFI: WIFI_EVENT_STA_DISCONNECTED: 63
34 2273 [iot_thread] [INFO ][DEMO][22730] Demo completed successfully.

E (23407) wifi_init: Wi-Fi not stop
I (23417) WIFI: WIFI_EVENT_STA_STOP
I (23417) wifi:flush txq
I (23417) wifi:stop sw txq
I (23427) wifi:lmac stop hw txq
I (23427) wifi:Deinit lldesc rx mblock:10
GAP procedure initiated: stop advertising.
GAP procedure initiated: stop advertising.
35 2278 [iot_thread] [INFO ][INIT][22780] SDK cleanup done.

36 2278 [iot_thread] [INFO ][DEMO][22780] -------DEMO FINISHED-------

```

I used Ubuntu OS and ESP32 DevKitC platform. What is the difference in your platform? Can you try with a fresh git clone?

aggarg, 2022-01-29:
Here is my command history:
```sh
gh repo clone git@github.com:SarahHamed/amazon-freertos.git
cd amazon-freertos/
git submodule update --init --recursive
vendors/espressif/esp-idf/install.sh
. vendors/espressif/esp-idf/export.sh
idf.py -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 build
idf.py flash monitor
idf.py monitor
```

aggarg, 2022-01-29:
Also, please apply this patch with additional logs and try again: [additional_log.zip](https://github.com/aws/amazon-freertos/files/7962843/additional_log.zip)

SarahHamed, 2022-02-03:
My apologies for the delay. Thank you for your help.

I tried to clone the fork and got no errors either. I get the error when I
work with the version I downloaded from the console. Attached below.

It's strange because previously I got the guru cache disabled error when I
cloned the repo from here <https://github.com/aws/amazon-freertos> directly
or when I download it from here
<https://github.com/aws/amazon-freertos/releases/tag/202107.00>

I applied the patch file but these two lines weren't printed:
ggdconfigPRINT( "GG IP: %s\n", pxHostAddressData->pcHostAddress );
ggdconfigPRINT( "GG Port: %u\n", pxHostAddressData->usPort );

 Connect_to_AWS_Greengrass_-_ESP32-DevKitC.zip
<https://drive.google.com/file/d/1l3PevzPRzeOzbBTC9ATFZaOIokxfEYBi/view?usp=drive_web>

Compressed 1077280 bytes to 676514...
Writing at 0x00020000... (2 %)
Writing at 0x00024000... (4 %)
Writing at 0x00028000... (7 %)
Writing at 0x0002c000... (9 %)
Writing at 0x00030000... (11 %)
Writing at 0x00034000... (14 %)
Writing at 0x00038000... (16 %)
Writing at 0x0003c000... (19 %)
Writing at 0x00040000... (21 %)
Writing at 0x00044000... (23 %)
Writing at 0x00048000... (26 %)
Writing at 0x0004c000... (28 %)
Writing at 0x00050000... (30 %)
Writing at 0x00054000... (33 %)
Writing at 0x00058000... (35 %)
Writing at 0x0005c000... (38 %)
Writing at 0x00060000... (40 %)
Writing at 0x00064000... (42 %)
Writing at 0x00068000... (45 %)
Writing at 0x0006c000... (47 %)
Writing at 0x00070000... (50 %)
Writing at 0x00074000... (52 %)
Writing at 0x00078000... (54 %)
Writing at 0x0007c000... (57 %)
Writing at 0x00080000... (59 %)
Writing at 0x00084000... (61 %)
Writing at 0x00088000... (64 %)
Writing at 0x0008c000... (66 %)
Writing at 0x00090000... (69 %)
Writing at 0x00094000... (71 %)
Writing at 0x00098000... (73 %)
Writing at 0x0009c000... (76 %)
Writing at 0x000a0000... (78 %)
Writing at 0x000a4000... (80 %)
Writing at 0x000a8000... (83 %)
Writing at 0x000ac000... (85 %)
Writing at 0x000b0000... (88 %)
Writing at 0x000b4000... (90 %)
Writing at 0x000b8000... (92 %)
Writing at 0x000bc000... (95 %)
Writing at 0x000c0000... (97 %)
Writing at 0x000c4000... (100 %)
Wrote 1077280 bytes (676514 compressed) at 0x00020000 in 15.5 seconds (effective 555.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done
***@***.***:~/Desktop/send_msg/Connect_to_AWS_Greengrass_-_ESP32-DevKitC/FreeRTOS$ idf.py monitor
Executing action: monitor
Choosing default port b'/dev/ttyUSB0' (use '-p PORT' option to set a specific serial port)
Running idf_monitor in directory /home/sarah/Desktop/send_msg/Connect_to_AWS_Greengrass_-_ESP32-DevKitC/FreeRTOS
[... 406 lines cut ...]

I (589) boot: Loaded app from partition at offset 0x20000
I (589) boot: Disabling RNG early entropy source...
I (590) cpu_start: Pro cpu up.
I (593) cpu_start: Application information:
I (598) cpu_start: Project name:     amazon-freertos
I (604) cpu_start: App version:      1
I (608) cpu_start: Compile time:     Feb  3 2022 14:59:43
I (614) cpu_start: ELF file SHA256:  0024499154854a8c...
I (620) cpu_start: ESP-IDF:          HEAD-HASH-NOTFOUND
I (626) cpu_start: Single core mode
I (631) heap_init: Initializing. RAM available for dynamic allocation:
I (638) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
I (644) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (650) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (656) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (662) heap_init: At 3FFCAF58 len 000150A8 (84 KiB): DRAM
I (668) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (675) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (681) heap_init: At 4009DE2C len 000021D4 (8 KiB): IRAM
I (687) cpu_start: Pro cpu start user code
I (705) spi_flash: detected chip: generic
I (706) spi_flash: flash io: dio
I (706) cpu_start: StartE (718) uart: uart_param_config(649): rx flow thresh error
0 1 [main] Initializing lwIP TCP stack

aggarg, 2022-02-03:
> I tried to clone the fork and got no errors either.

Great!

>I applied the patch file but these two lines weren't printed:
ggdconfigPRINT( "GG IP: %s\n", pxHostAddressData->pcHostAddress );
ggdconfigPRINT( "GG Port: %u\n", pxHostAddressData->usPort );

This is strange. While I try your bundle, can you add a log to dump the retrieved JSON file: https://github.com/aws/amazon-freertos/blob/main/libraries/freertos_plus/aws/greengrass/src/aws_greengrass_discovery.c#L460

```
ggdconfigPRINT( "JSON: %s\n", pcBuffer );
```

Thanks.

aggarg, 2022-02-04:
I do not have access to the Google Drive link you shared but I tried the `202107.00` tag and it works. Here is my command history:

```
git clone git@github.com:aws/amazon-freertos.git
cd amazon-freertos/
git checkout tags/202107.00
git submodule update --init --recursive 
vendors/espressif/esp-idf/install.sh
. vendors/espressif/esp-idf/export.sh
idf.py -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 build
idf.py flash monitor
```

Here is the output:
```
I (5102) WIFI: WIFI_EVENT_STA_CONNECTED
I (5172) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (5732) esp_netif_handlers: sta ip: 192.168.154.193, mask: 255.255.255.0, gw: 192.168.154.34
I (5732) WIFI: SYSTEM_EVENT_STA_GOT_IP
5 500 [sys_evt] [INFO ][DEMO][5000] Connected to WiFi access point, ip address: 255.255.255.0.

6 500 [iot_thread] [INFO ][DEMO][5000] Successfully initialized the demo. Network type for the demo: 1

7 500 [iot_thread] [aws_greengrass_discovery_demo.c:541] [INFO] [Greengrass_Discovery_Demo] Attempting automated selection of Greengrass device

8 503 [NetworkManager] [INFO ][DEMO][5030] Invoked callback for network state change event.

9 505 [NetworkManager] [INFO ][DEMO][5050] Invoked callback for network state change event.

GAP procedure initiated: stop advertising.
GAP procedure initiated: stop advertising.
E (6612) PKCS11: failed nvs get file size 4354 0
10 916 [iot_thread] JSON file retrieval completed

11 916 [iot_thread] About to close socket.

12 917 [iot_thread] Socket closed.

13 917 [iot_thread] Stack high watermark for discovery helper task: 5576.

15 917 [iot_thread] GG IP: **************

16 917 [iot_thread] GG Port: 8883

E (10262) PKCS11: failed nvs get file size 4354 0
17 1387 [iot_thread] About to close socket.

18 1388 [iot_thread] Socket closed.

19 1388 [iot_thread] Stack high watermark for discovery helper task: 3080.

20 1388 [iot_thread] [aws_greengrass_discovery_demo.c:551] [INFO] [Greengrass_Discovery_Demo] Greengrass device discovered.

21 1388 [iot_thread] [aws_greengrass_discovery_demo.c:440] [INFO] [Greengrass_Discovery_Demo] Creating a TLS connection to ******************.

E (14872) PKCS11: failed nvs get file size 4354 0
22 1858 [iot_thread] [core_mqtt.c:886] [INFO] [MQTT] Packet received. ReceivedBytes=2.

23 1858 [iot_thread] [core_mqtt_serializer.c:970] [INFO] [MQTT] CONNACK session present bit not set.

24 1858 [iot_thread] [core_mqtt_serializer.c:912] [INFO] [MQTT] Connection accepted.

25 1858 [iot_thread] [core_mqtt.c:1563] [INFO] [MQTT] Received MQTT CONNACK successfully from broker.

26 1858 [iot_thread] [core_mqtt.c:1829] [INFO] [MQTT] MQTT connection established with the broker.

27 1858 [iot_thread] [aws_greengrass_discovery_demo.c:478] [INFO] [Greengrass_Discovery_Demo] An MQTT connection is established with ******************.

28 1859 [iot_thread] [aws_greengrass_discovery_demo.c:520] [INFO] [Greengrass_Discovery_Demo] MQTT PUBLISH message sent successfully to the broker: Topic=freertos/demos/ggd

29 2009 [iot_thread] [aws_greengrass_discovery_demo.c:520] [INFO] [Greengrass_Discovery_Demo] MQTT PUBLISH message sent successfully to the broker: Topic=freertos/demos/ggd

30 2159 [iot_thread] [aws_greengrass_discovery_demo.c:520] [INFO] [Greengrass_Discovery_Demo] MQTT PUBLISH message sent successfully to the broker: Topic=freertos/demos/ggd

31 2309 [iot_thread] [aws_greengrass_discovery_demo.c:575] [INFO] [Greengrass_Discovery_Demo] Disconnecting from broker.

32 2309 [iot_thread] [core_mqtt.c:2149] [INFO] [MQTT] Disconnected from the broker.

33 2310 [iot_thread] [aws_greengrass_discovery_demo.c:597] [INFO] [Greengrass_Discovery_Demo] Disconnected from the broker.

34 2310 [iot_thread] Heap low watermark: 116896. Stack high watermark: 3080.

35 2310 [iot_thread] [INFO ][DEMO][23100] memory_metrics::freertos_heap::before::bytes::217836

36 2310 [iot_thread] [INFO ][DEMO][23100] memory_metrics::freertos_heap::after::bytes::116896

37 2310 [iot_thread] [INFO ][DEMO][23100] memory_metrics::demo_task_stack::before::bytes::8920

38 2310 [iot_thread] [INFO ][DEMO][23100] memory_metrics::demo_task_stack::after::bytes::3080

I (24832) wifi:state: run -> init (0)
I (24832) wifi:pm stop, total sleep time: 15697466 us / 19734676 us

I (24832) wifi:new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
I (24832) WIFI: WIFI_EVENT_STA_DISCONNECTED: 63
39 2410 [iot_thread] [INFO ][DEMO][24100] Demo completed successfully.

E (24842) wifi_init: Wi-Fi not stop
I (24852) WIFI: WIFI_EVENT_STA_STOP
I (24852) wifi:flush txq
I (24852) wifi:stop sw txq
I (24862) wifi:lmac stop hw txq
I (24862) wifi:Deinit lldesc rx mblock:10
40 2414 [iot_thread] [INFO ][INIT][24140] SDK cleanup done.

41 2414 [iot_thread] [INFO ][DEMO][24140] -------DEMO FINISHED-------
```

Can you also try the same commands?

Thanks.

aggarg, 2022-02-04:
>I applied the patch file but these two lines weren't printed:
ggdconfigPRINT( "GG IP: %s\n", pxHostAddressData->pcHostAddress );
ggdconfigPRINT( "GG Port: %u\n", pxHostAddressData->usPort );

A possibility is that what are you editing and what you are compiling are different. A quick way to confirm is to put some non-compilable code and verify that the compilation fails.

SarahHamed, 2022-02-06:
I tried the same commands and printed the additional logs.
[gg-log3.txt](https://github.com/aws/amazon-freertos/files/8010337/gg-log3.txt)

My apologies, I adjusted the access to the link.
[amazon-freertos](https://drive.google.com/file/d/1l3PevzPRzeOzbBTC9ATFZaOIokxfEYBi/view?usp=sharing)

Thank you.

SarahHamed, 2022-02-06:
Note, I tried the same commands on the project I attached its link but not the repo "github.com:aws/amazon-freertos.git"
```
. vendors/espressif/esp-idf/export.sh
idf.py -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 build
idf.py flash monitor
```

aggarg, 2022-02-07:
> but not the repo "github.com:aws/amazon-freertos.git".

Would you please try that.

SarahHamed, 2022-02-08:
I tried to run this command but I don't have a public ssh key so it failed.
- git clone git@github.com:aws/amazon-freertos.git

I used the HTTPS method instead.
**All the commands that I used in the terminal:**
```

git init
git clone git@github.com:aws/amazon-freertos.git (failed)
git clone https://github.com/aws/amazon-freertos.git --recurse-submodules
cd amazon-freertos/
git checkout tags/202107.00
git submodule update --init --recursive
vendors/espressif/esp-idf/install.sh
. vendors/espressif/esp-idf/export.sh
```
I switched the file "aws_greengrass_discovery_demo.c" to the new one that I am trying, then I built it.
```
idf.py -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 build
idf.py flash  monitor
```
Unfortunately, I got the guru error again.
[log-gg.txt](https://github.com/aws/amazon-freertos/files/8027948/log-gg.txt)

aggarg, 2022-02-10:
I am finally able to repro the issue. The root cause is that the address of `xNetworkContext` variable is stored in the MQTT context which in-turn stores the address of `secureSocketsTransportParams`. Both of these variables are local to the function `discoverGreengrassCore` and therefore, go out-of-scope as soon as the function returns. Later when these are accessed in `MQTT_Publish` call in `sendMessageToGGC`, these result in invalid memory access leading to `LoadStoreError` panic. The solution is simple - declare `secureSocketsTransportParams` and `xNetworkContext` as file global just as you have declared `xMQTTContext`.

Thanks.

SarahHamed, 2022-02-10:
Thank you so much for your help. I tried it and it's working.

I am gonna close it now.

Thank you.

closed 2022-02-10 by SarahHamed