aws/amazon-freertos #3451: [General] OtaJobEventActivate not getting triggered when OTA job created using aws cli

kiran-g, 2022-01-26:
**Briefly summarize the issue being raised**
I am able to initiate OTA update via console. But OTA is stuck when initiating job via cli. I could see that 'OtaJobEventActivate' is not triggered

**Describe the desired outcome**
I want my OTA create job to be able to successfully complete an OTA job

**System information**
- Hardware board: ESP-32
- IDE used: Gedit
- Operating System: linux
- Code version: cant find as I had downloaded a specific version arvhive from github
- Project/Demo: ota demo
.

**Code/Steps to reproduce the current status**
```cpp
echo '[
  {
    "fileName": $fwimagefilename,                
    "fileType": 1,
    "fileLocation": {
      "stream": {
        "streamId": $streamid,                         
        "fileId":123
      }                        
    },
    "codeSigning": {
      "awsSignerJobId": $signingjobid     
    }
  }
]' |  jq -n  --arg  fwimagefilename $FIRMWAREIMAGENAME --arg  streamid $OTASTREAMID \
     --arg signingjobid  $SIGNJOBID -f /dev/stdin > $CREATE_JOB_CLI_INPUT_JSON_FILE

aws iot create-ota-update --ota-update-id $OTAJOBNAME  --target-selection SNAPSHOT \
 --description "a cli ota update" --files file://$CREATE_JOB_CLI_INPUT_JSON_FILE  \
 --targets arn:aws:iot:$AWS_REGION:$AWS_ACCOUNT_ID:thing/$THINGNAME \
 --role-arn arn:aws:iam::$AWS_ACCOUNT_ID:role/$OTAROLE
```

pvyawaha, 2022-01-26:
Hello @kiran-g ,

The activate event is for activating the new firmware and in case of ESP32 it is rebooting the device so that bootloader will boot new image. When you are creating the job using CLI you are using file type ID of non zero and by default 0 is for self-firmware unless you have changed the OTA config for that. If you can try using 0 for filetype ID you should receive the activate event. This is done by default when you create a job using console. 

Please try that and let me know how it goes.

prasadvya, 2022-02-03:
Hello @kiran-g ,

Did that resolved your issue?

kiran-g, 2022-02-04:
Yes. This issue is resolved. Thanks a lot for the help. You may please
close the ticket.

closed 2022-03-11 by pvyawaha