aws/amazon-freertos #3453: [General] <Green Grass Discovery Demo>

jitendrazala, 2022-02-12:
Hello Folks,

I have been trying to get green grass discovery demo working on ESP32-WROVER hardware but couldn't not get any resource of green grass device v2. 

**Hardwares Used:**
1. Raspberry Pi3 [RPi]
2. ESP32-WROVER

**Steps I followed to setup GGC on RPi:**
1. Setup Raspberry Pi with Raspbian buster and installed Java
2. Created a user on AWS IAM console with policies i.e.

**_IAMFullAccess
AmazonS3FullAccess
AWSGreengrassFullAccess
AWSIoTFullAccess_**

3. Downloaded AWS Secret Key and AWS Access Key of user
4. Exported AWS Secret Key and AWS Access Key in RPi
5. Setup green grass core device in RPi [https://docs.aws.amazon.com/greengrass/v2/developerguide/getting-started.html]
6. Downloaded green grass installer, run it and successfully installed Nucleus as system service in RPi
7. Checked and verified by command : **_sudo /greengrass/v2/bin/greengrass-cli component list_** 
and it was showing **_aws.greengrass.cli_** component's status as **_Running_**
8.  Then added a new component for publish and subscribe to AWS IoT core by creating recipes and artifacts. Component was deployed and successfully running. I could publish and subscribe to AWS IoT core.

**Steps I followed to setup GGC Discovery Demo on ESP32:**
1. Setup python and python-pip on windows PC
2. Downloaded AWS CLI and installed on windows PC
3. Configured AWS user configuration using command : **_aws configure_** and configured AWS Access Key, AWS Secret Key, Region and default data format [json].
4. Cloned amazon-freertos 
**_git clone https://github.com/aws/amazon-freertos.git --recurse-submodules
cd amazon-freertos
git fetch --all –tags_**
5. Installed ESP-IDF dependencies : amazon-freertos/vendors/espressif/esp-idf & run:
**_install.bat_**
6. Setup ESP-IDF environment: amazon-freertos/vendors/espressif/esp-idf & run:
**_export.bat_**
7. Setup AWS configuration i.e. Things name, Wi-Fi SSID and Password in **_configure.json_** from amazon-freertos/tools/aws_config_quick_start path
8. Run : **_python SetupAWS.py setup_** to setup thing and AWS credentials
9. Run:  **_python SetupAWS.py update_creds_** to update AWS credentials
10. Replaced **_CONFIG_CORE_MQTT_MUTUAL_AUTH_DEMO_ENABLED_** with the **_CONFIG_GREENGRASS_DISCOVERY_DEMO_ENABLED_** in vendors/espressif/boards/esp32/aws_demos/config_files/aws_demo_config.h 
11. In amazon-freertos root directory: built demo using below command
_**idf.py_ -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 build**
12 .Flashed and monitored firmware using **_idf.py -p COM3 flash monitor_** 

But not getting green grass device information 😕 Don't know what was the mistake !
Attaching logs of ESP32 monitoring.

Can anyone point me about the possible issue/s and probably solution !?

Thanks.
Jitendra

```
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory c:\amazon-freertos
Executing "C:\Users\Jitendra\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe C:\amazon-freertos\vendors\espressif\esp-idf\tools/idf_monitor.py -p COM5 -b 115200 --toolchain-prefix xtensa-esp32-elf- c:\amazon-freertos\build\aws_demos -m 'C:\Users\Jitendra\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe' 'C:\amazon-freertos\vendors\espressif\esp-idf\tools\idf.py' '-p' 'COM5'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM5 instead...
C:\amazon-freertos\vendors\espressif\esp-idf\tools/idf_monitor.py:474: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if StrictVersion(serial.VERSION) < StrictVersion('3.3.0'):
--- idf_monitor on \\.\COM5 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
[... 129 lines cut ...]
14 773 [iot_thread] [INFO ][DEMO][7730] memory_metrics::freertos_heap::after::bytes::81876

15 774 [iot_thread] [INFO ][DEMO][7730] memory_metrics::demo_task_stack::before::bytes::11992

16 774 [iot_thread] [INFO ][DEMO][7740] memory_metrics::demo_task_stack::after::bytes::7112

I (9406) wifi:state: run -> init (0)
I (9406) wifi:pm stop, total sleep time: 3744775 us / 6192357 us

I (9406) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (9406) WIFI: WIFI_EVENT_STA_DISCONNECTED: 63
17 874 [iot_thread] [ERROR][DEMO][8740] Error running demo.

E (9416) wifi_init: Wi-Fi not stop
I (9426) WIFI: WIFI_EVENT_STA_STOP
I (9426) wifi:flush txq
I (9426) wifi:stop sw txq
I (9436) wifi:lmac stop hw txq
I (9436) wifi:Deinit lldesc rx mblock:10
GAP procedure initiated: stop advertising.
GAP procedure initiated: stop advertising.
18 879 [iot_thread] [INFO ][INIT][8790] SDK cleanup done.

19 879 [iot_thread] [INFO ][DEMO][8790] -------DEMO FINISHED-------
```

jitendrazala, 2022-02-16:
That would be really helpful.
Thanks.

archigup, 2022-02-16:
Hi, did you enable cloud discovery for the Greengrass core? If not, see [docs to enable it here](https://docs.aws.amazon.com/greengrass/v2/developerguide/client-devices-tutorial.html?icmpid=docs_greengrassv2_help_panel_core_device_cloud_discovery).

Let me know if this does not solve your issue.

jitendrazala, 2022-02-17:
Hi,

I already enabled cloud discovery for the greengrass core.
Though I did some debugging and came to know that device getting greengrass core information but somehow it was not able to connect to that.
It always failed in **_GGD_SecureConnect_Connect_** function in **_aws_helper_secure_connect.c_** file specifically in **_SOCKETS_Connect_** in  GGD_SecureConnect_Connect function.

```
`BaseType_t GGD_SecureConnect_Connect( const GGD_HostAddressData_t * pxHostAddressData,
                                      Socket_t * pxSocket,
                                      uint32_t ulReceiveTimeOut,
                                      uint32_t ulSendTimeOut )
{
    const TickType_t xReceiveTimeOut = pdMS_TO_TICKS( ulReceiveTimeOut );
    const TickType_t xSendTimeOut = pdMS_TO_TICKS( ulSendTimeOut );
    SocketsSockaddr_t xServerAddress;
    size_t xURLLength;
    BaseType_t xIsIPAddress;
    BaseType_t xStatus;
    int32_t returnCode;

    configASSERT( pxHostAddressData != NULL );
    configASSERT( pxSocket != NULL );

    /* Calculate the length of the supplied URL. */
    xURLLength = strlen( pxHostAddressData->pcHostAddress );

    /* Ensure that the length of the specified URL is
     * within the permitted limits. */
    if( xURLLength <= ( size_t ) securesocketsMAX_DNS_NAME_LENGTH )
    {
        printf("1111");
        /* Create the socket. */
        *pxSocket = SOCKETS_Socket( SOCKETS_AF_INET,
                                    SOCKETS_SOCK_STREAM,
                                    SOCKETS_IPPROTO_TCP );

        if( *pxSocket == SOCKETS_INVALID_SOCKET )
        {
            xStatus = pdFAIL;
            printf("2222");
        }
        else
        {
            xStatus = pdPASS;
            printf("3333");
        }

        if( xStatus == pdPASS )
        {
            printf("4444");    
            if( prvIsIPaddress( pxHostAddressData->pcHostAddress ) == ( uint32_t ) 0 )
            {
                xIsIPAddress = pdFALSE;
                printf("5555");
            }
            else
            {
                xIsIPAddress = pdTRUE;
                printf("6666");
            }

            xServerAddress.ucLength = sizeof( SocketsSockaddr_t );
            xServerAddress.usPort = SOCKETS_htons( pxHostAddressData->usPort );
            xServerAddress.ulAddress =
                SOCKETS_GetHostByName( pxHostAddressData->pcHostAddress );

            if( xServerAddress.ulAddress == 0u )
            {
                ggdconfigPRINT( "ERROR! Failed to resolve host address: ServerHost=%.*s",
                                xURLLength, pxHostAddressData->pcHostAddress );
[... 67 lines cut ...]
                printf("1010");
                returnCode = SOCKETS_Connect( *pxSocket,
                                              &xServerAddress,
                                              ( uint32_t ) sizeof( xServerAddress ) );

                 /***It always failed here***/
                if( returnCode != SOCKETS_ERROR_NONE )
                {
                    ggdconfigPRINT( "ERROR! SOCKETS_Connect call failed: ServerAddress=%lu, Port=%u, ReturnCode=%d\r\n",
                                    xServerAddress.ulAddress, xServerAddress.usPort, returnCode );
                    GGD_SecureConnect_Disconnect( pxSocket );
                    xStatus = pdFAIL;
                }
            }
        }
    }
    else
    {
        ggdconfigPRINT( "Malformed URL\r\n" );
        xStatus = pdFAIL;
    }

    return xStatus;
}`
```

archigup, 2022-02-18:
Hmm, from the logs posted initial post, it looks like its getting past that and failing in `GGD_GetIPandCertificateFromJSON` after not finding the certificate key in `prvGGDGetCertificate`. Is that still the case? Would you be able to provide the JSON file retrieved by the application if so?

jitendrazala, 2022-02-18:
Yes. Sure.
Please find attachment.
[image]

archigup, 2022-02-19:
Looking at that screenshot, the SOCKETS_Connect call is getting the correct ip/port from the json document, so the issue is likely with the Greengrass certificate. What is the error if the section with the `/* Override TLS trust store with server certificate. */` comment is commented out? If that is done and the SOCKETS_Connect return code is one of the TLS ones then we can narrow the issue down to the certificate.

aggarg, 2022-03-14:
Are you still facing this issue?