aws/amazon-freertos #3454: [General] How to use ksz8041 eth driver in 202107.00 release?

laszlopataki, 2022-02-15:
**Briefly summarize the issue being raised**
We are planning to use the ksz8041 ETH driver. 
The driver works well in `esp-idf:release/v4.3`
The issue is that this driver is present from IDF 4.3 alongside further changes in `esp_rom`, `esp_eth` and `hal` components. 
AFR integrates a special type of dev framework called  [esp-afr-sdk](https://github.com/espressif/esp-afr-sdk) rather than the [ESP-IDF](https://github.com/espressif/esp-idf). 
The latest esp-afr-sdk is based on esp-idf 4.2 and I can not see any active development.

**Describe the desired outcome**
Optimally I would like to just update the IDF version instead of using another repo that is based on IDF. 

I can see the following options to solve the issue: 
- Porting ksz8041 driver with the relevant changes in `hal`, `esp_eth` and `esp_rom` to `esp-afr-sdk:relase/v4.2`
- Porting `esp-idf:relase/v4.3` to the `AFR:/release`

Please share the roadmap of integrating new versions of IDF into AFR? 
Can you suggest the best way to utilize ksz8041 driver? 
Can you share more information about the difference between `esp-idf` and `esp-afr-sdk`?
Would you recommend porting a new version of esp-idf to AFR and if yes what are the highlights of doing it?

Thank you

ravibhagavandas, 2022-02-16:
Hi @laszlopataki 

Unfortunately, we don't have a specific timeline as to when `esp-afr-sdk:relase/v4.3` will be released into AFR. I would suggest the best way to unblock yourself, will be to port your ethernet driver to work on `esp-afr-sdk:release/v4.2`. That way the changes should be minimal and isolated. If possible, I would prefer to first make it working on esp-idf release/4.2 and then port the changes to esp-afr-sdk/release-4.2 so that there are less surprises due to version changes.
ESP IDF and esp-afr-sdk differ in the way they are built as esp-afr-sdk is built into AFR source tree. Porting entire `espidf release-4.3` will involve lot more changes and lot more testing of changes.

laszlopataki, 2022-02-16:
Hi @ravibhagavandas, 

Thank you for your answer. It is really helpful. 
You are right about the short-term solution. Better to port the eth driver to `esp-afr-sdk:release/v4.2`. 
If you do not have a timeline, can you share if there is a plan for this at all, or do we need to think about in the future how to update the esp-idf ourselves? 
I am trying to map how much work does it take to port new esp-idf to afr. Do you know any major points that could help this estimation? Files to be changed, steps to be taken, other considerations?

n9wxu, 2022-02-17:
We do not have a timeline but ESP IDF updates are always in our plans.

closed 2022-02-18 by n9wxu