aws/amazon-freertos #3459: Support for ESP32-S3?

marcusufg, 2022-02-20:
Version release/v4.4 of Espressif esp-idf (https://github.com/espressif/esp-idf/tree/release/v4.4) already supports ESP32-S3.
[image]
My new truck tracker system is intended to use it. I can upload very basic espressif's demos do it, but I can't do it for my project. Does AFR support for this new microcontroller in near future? Is there any timeline? Currently only v4.2 is supported on AFR.
[image]

n9wxu, 2022-02-22:
At present, there is not a schedule for updating the IDF of AFR beyond 4.2.  Your request will help set the priorities of this project.  Thank you.

marcusufg, 2022-02-22:
Thanks, Joseph. We, from Simemap, believe this will help also a lot of 
other programmers who are migrating to this new ESP32-S3 processor. I'll 
keep looking forward to the updates.

closed 2022-02-22 by n9wxu

thomasklaush, 2022-09-08:
Since esp-afr-sdk is now on v4.4, is there a timeframe to integrate it?

mahavirj, 2022-09-12:
@thomasklaush We are working on integrating ESP-IDF v4.4 support. Please expect the draft PR in next few days. CC @laukik-hase

thomasklaush, 2022-10-04:
Hi @mahavirj 
any news on that?

mahavirj, 2022-10-05:
@thomasklaush Apologies for the delay here. We are in the final phase of the testing on ESP32-S3. We shall raise the PR by early next week.