aws/amazon-freertos #3461: [BUG] OTA Agent and IoT Core Job status synchronisation issue

laszlopataki, 2022-02-23:
**Describe the bug**
When you start a new OTA Job for a device the `otaconfigOTA_UPDATE_STATUS_FREQUENCY ` tells how frequently the device should report a status. 
The condition below resusts that the 1st packet will not report status. 
```if( ( received % otaconfigOTA_UPDATE_STATUS_FREQUENCY ) == 0U )```

This causes the misunderstanding of "why the OTA did not start" and desynchronization between the actual status of the OTA and what AWS IoT Core tells us. 
The problem is magnified if the IoT device has a narrow bandwidth and the packet transfer is slow. In this case, you can see Queued status for minutes while the OTA is actually running. 

Subtitute 1 from `otaconfigOTA_UPDATE_STATUS_FREQUENCY` , or extending the condition with `received == 1` can solve the issue.

**System information**
- Hardware board: ESP32
- IDE used: VSCode/Eclipse
- Operating System:  Windows 
- Code version: 20210526_Archive-139-gb7151fe03
- Project/Demo: OTA_MQTT_UPDATE_DEMO_ENABLED

**Expected behavior**
The OTA Agent sends a status update to IoT Core at the first packet and after at every `otaconfigOTA_UPDATE_STATUS_FREQUENCY`.

pvyawaha, 2022-05-20:
Hello @laszlopataki ,

The requested change is now implemented and merged in mainline.

https://github.com/aws/ota-for-aws-iot-embedded-sdk/pull/426

Please let me know if you have any more questions on this or this issue can be closed.

ActoryOu, 2022-06-13:
Fixed in #3495.

closed 2022-06-13 by ActoryOu