aws/amazon-freertos #3462: [General] FreeRTOS Device Software fails to build with EPS-IDF

CloudedThings, 2022-02-28:
I am trying to use the Device software downloaded from IoT Core to flash it on my ESP32 with ESP-IDF tools. I've used the ESP tools with examples provided with it but when I'm trying to use the package from IoT Core I'm receiving CMake Error. I've followed the [Get started with ESP-IDF v4.2](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_esp32wroom-32se.html#setup-esp32wroom-32se-idf42) carefully and as I said the toolchain works with other projects.

This is the result of `idf.py build` command run in the project folder:

```
Running cmake in directory c:\esp32\esp32freertosaws\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 c:\esp32\esp32freertosaws"...
-- The C compiler identification is unknown
-- The CXX compiler identification is GNU 10.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: C:/cygwin64/bin/cc
-- Check for working C compiler: C:/cygwin64/bin/cc - broken
CMake Error at C:/Espressif/tools/cmake/3.20.3/share/cmake-3.20/Modules/CMakeTestCCompiler.cmake:66 (message):
  The C compiler

    "C:/cygwin64/bin/cc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: C:/ESP32/ESP32freertosaws/build/CMakeFiles/CMakeTmp

    Run Build Command(s):C:/Espressif/tools/ninja/1.10.2/ninja.exe cmTC_7b540 &&
    CreateProcess failed. Command attempted:
    "C:\cygwin64\bin\cc    -o CMakeFiles\cmTC_7b540.dir\testCCompiler.c.obj -c testCCompiler.c"
    ninja: fatal: CreateProcess: Access is denied.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:8 (project)

```
So my question is how come ninja responds with `Access is denied` to cygwin64 even though that toolchain works with other projects?

**System information**
- Hardware board: ESP32-WROOM-32E
- IDE used: VSCode
- Operating System: Windows 
- Code version: ESP-IDF 4.4
- Project/Demo: greengrass_connectivity

tianmc1, 2022-03-01:
In the error logs this line `-- The C compiler identification is unknown` doesn't look quite right. Also I wonder if the compiler should be espressif's xtensa-esp32-elf-gcc instead of cygwin64. Can you check you IDF is properly installed and configured? You can try running `idf.py build` in ESP's example projects to see if they build successfully.

aggarg, 2022-03-14:
Are you still facing this issue?

closed 2022-03-16 by aggarg