aws/amazon-freertos #3464: [General] Demo defaults to single core mode

genie-ltran, 2022-03-09:
I'm using the MQTT demo which works as expected.  First, I'm using the amazon-freertos repo as a submodule as outlined in "https://github.com/aws/amazon-freertos/issues/2978".  This works.  I added threads to do a low latency decoding of proprietary signals on pins.  My problem is that I'm attempting to pin this thread to the app_cpu for this thread instead of the core_cpu.  Except, that when I use the xTaskCreatePinnedToCore(), the thread itself is reporting that it's still running in the pro_cpu.  I figured out through a lot of searching online that I had to override the directive "CONFIG_FREERTOS_UNICORE" which I've tried with:

cmake -S . -B build-directory -DIDF_SDKCONFIG_DEFAULTS=<path_to_your_sdkconfig_defaults> -DCMAKE_TOOLCHAIN_FILE=freertos/tools/cmake/toolchains/xtensa-esp32.cmake -GNinja

I did this no result.  It keeps reverting the code that forces the directive in the sdkconfig.h file
to "#define CONFIG_FREERTOS_UNICORE 1".

No IDE.
Windows command line.
IDF version: v3.3 - "git checkout -f tags/202012.01"
project:  mqtt_demo_mutual_auth 

Can someone explain to me how to force dual core mode and where the setting for this feature is?

Thank you.

aggarg, 2022-03-14:
When you use `amazon-freertos`, the FreeRTOS-Kernel used is the official version which does not have SMP support in main yet. Can you write your application such that it consumes `esp-idf` and our libraries? This way you will be able to use SMP features available in esp-idf kernel and will able to independently update idf and libraries. You can take a look at this repo for an example - https://github.com/FreeRTOS/Labs-Project-Espressif-Demos.

aggarg, 2022-03-15:
Did you take a look at the repo I mentioned above? What else would you need for your application?

genie-ltran, 2022-03-15:
Sorry, I didn't fully grasp what you were trying to convey before.  I understand now.  Let me see how much effort this will take.  I'm trying to stick with amazon-freertos, because it's requires much less work on my part to get it running.  I didn't realize how out of sync it was with Espressif's source.

closed 2022-03-15 by genie-ltran