aws/amazon-freertos #3469: help in compile aws_demos in Espressif IDE

WilliamFrasson, 2022-03-15:
I am running into several problems while trying to compile aws_demos from Espressif IDE for esp32 target.
can I ask for a support in the correct steps to setup the project into Espressif IDE and compile it? 
Espressif IDE Version is 2.4.2

thanks in advaces

aggarg, 2022-03-16:
Can you try to use this demo and share if you are able to compile - https://github.com/FreeRTOS/Labs-Project-Espressif-Demos

WilliamFrasson, 2022-03-19:
nope

WilliamFrasson, 2022-03-20:
I found something useful here in issues https://github.com/aws/amazon-freertos/issues/832  and https://github.com/aws/amazon-freertos/pull/887

aggarg, 2022-03-21:
Glad that you found a solution.

closed 2022-03-21 by aggarg

WilliamFrasson, 2022-03-29:
sorry but i'm again not able to create a espressif IDE  project for the AWS demo . 
Someone can help me?

aggarg, 2022-03-31:
Did you try the link I mentioned before?

WilliamFrasson, 2022-03-31:
I see that the are instructions for windows Power Shell , while I need instructions for compile, build and flash the  aws_demo project in Espressif IDE.
So I didn't try it

aggarg, 2022-04-01:
Do not pay attention to the Power Shell instructions as those are specific to the AWS setup needed for delta OTA demo. Try and see if you are able to the build this demo is the Eclipse IDE. Adding the MQTT demo then would be the next step.

WilliamFrasson, 2022-04-03:
Hi Gaurav
i had tried it as you described and i was able to build up and run the code on the esp32-wroom-devkit
i follow these steps:
0. create a folder and then clone the code 
	git clone https://github.com/FreeRTOS/Labs-Project-Espressif-Demos.git --recurse-submodules
1. import the "Labs-Project-Espressif-Demos" in espressif IDF as "Existing IDF project"  (imported from folder:  ..\Labs-Project-Espressif-Demos\demos\delta_ota), without copy files and folder
2. check "install tool in IDE menu "Espressif->ESP-IDF Tool Manager ->Install Tools 
		--> i select my current version of esp-idf (: ESP-IDF v4.4-dirt ) --> from "C:\Espressif\frameworks\esp-idf"
		
3. launch "build" by clicking the Project->build  
	all compile correclty

4.	flash on the device and code is running fine

-----------------------------------------
personally i still not well understand how aws is used instead of standard freertos from espressif.  
but going to my original issues,. how can going up to  the aws_demo ?  I have to inclued MQTT demo? 
this demo? ...\amazon-freertos\demos\coreMQTT

(thank for the help :))

aggarg, 2022-04-04:
> i had tried it as you described and i was able to build up and run the code on the esp32-wroom-devkit

That is great.

> personally i still not well understand how aws is used instead of standard freertos from espressif. 

This demo uses FreeRTOS included in ESP-IDF.

> how can going up to the aws_demo ? I have to inclued MQTT demo?

This repo currently does not have MQTT demo. You can add it and use this one as a reference - https://github.com/aws/aws-iot-device-sdk-embedded-C/tree/main/demos/mqtt/mqtt_demo_mutual_auth. Would that work for you? 

> here below when i try to compile from aws_demo from the Espressif IDE

Assuming that you are talking about aws/amazon-freertos repo, this is not tested with IDE and only with command line.