aws/amazon-freertos #3476: build [BUG] Fail to build

larryxia-ddl, 2022-03-31:
I got FreeRTOS from AWS console. I have esp-idf-tools-setup-offline-4.2.3 installed on Windows 10. I can make the build files by using this command 
cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -G Ninja -S . -B ..\build\

But when I try to build the project, I got a lot of error.  If I use idf 4.4. the build also failed, but it is failed in other c file. Please advise how to fix this issue.  

cmake --build ..\build\

286/1634] Building C object CMakeFiles/afr_ble_hal.dir/vendors/espressif/boards/ports/ble/iot_ble_hal_common_gap.c.obj
FAILED: CMakeFiles/afr_ble_hal.dir/vendors/espressif/boards/ports/ble/iot_ble_hal_common_gap.c.obj
C:\usr\esp\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DAFR_ESP_LWIP -DESP_PLATFORM -DIDF_VER=\"v4.2.3\" -DMBEDTLS_CONFIG_FILE=\"C:/iot/FreeRTOS/vendors/espressif/esp-idf/components/mbedtls/port/include/mbedtls/esp_config.h\" -D_GNU_SOURCE -IC:/iot/FreeRTOS/libraries/abstractions/ble_hal/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/bt/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/bt/common/osi/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/bt/host/nimble/nimble/porting/nimble/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/bt/host/nimble/port/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/bt/host/nimble/nimble/nimble/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/bt/host/nimble/nimble/nimble/host/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/bt/host/nimble/nimble/nimble/host/services/ans/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/bt/host/nimble/nimble/nimble/host/services/bas/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/bt/host/nimble/nimble/nimble/host/services/gap/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/bt/host/nimble/nimble/nimble/host/services/gatt/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/bt/host/nimble/nimble/nimble/host/services/ias/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/bt/host/nimble/nimble/nimble/host/services/ipss/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/bt/host/nimble/nimble/nimble/host/services/lls/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/bt/host/nimble/nimble/nimble/host/services/tps/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/bt/host/nimble/nimble/nimble/host/util/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/bt/host/nimble/nimble/nimble/host/store/ram/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/bt/host/nimble/nimble/nimble/host/store/config/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/bt/host/nimble/nimble/porting/npl/freertos/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/bt/host/nimble/esp-hci/include -Iconfig -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/newlib/platform_include -IC:/iot/FreeRTOS/vendors/espressif/boards/components/freertos/include -IC:/iot/FreeRTOS/freertos_kernel/include -IC:/iot/FreeRTOS/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/include -IC:/iot/FreeRTOS/vendors/espressif/boards/esp32/aws_demos/config_files -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/heap/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/log/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/lwip/include/apps -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/lwip/include/apps/sntp -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/lwip/lwip/src/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/lwip/port/esp32/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/lwip/port/esp32/include/arch -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/lwip/port/esp32/tcp_isn -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/soc/src/esp32/. -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/soc/src/esp32/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/soc/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/esp_rom/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/esp_rom/esp32 -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/esp_common/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/esp_system/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/xtensa/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/xtensa/esp32/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/esp32/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/driver/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/driver/esp32/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/esp_ringbuf/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/efuse/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/efuse/esp32/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/espcoredump/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/esp_timer/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/esp_ipc/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/soc/soc/esp32/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/soc/soc/esp32/../include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/soc/soc/esp32/private_include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/vfs/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/esp_wifi/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/esp_wifi/esp32/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/esp_event/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/esp_netif/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/esp_eth/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/tcpip_adapter/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/app_trace/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/nvs_flash/include -IC:/usr/esp/frameworks/esp-idf-v4.2.3/components/spi_flash/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/mbedtls/port/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/xtensa/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/soc/soc/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/soc/src/esp32/include -IC:/iot/FreeRTOS/libraries/3rdparty/pkcs11 -IC:/iot/FreeRTOS/libraries/3rdparty/mbedtls/include -IC:/iot/FreeRTOS/libraries/3rdparty/mbedtls_config -IC:/iot/FreeRTOS/libraries/3rdparty/mbedtls/include/mbedtls -IC:/iot/FreeRTOS/libraries/3rdparty/mbedtls_utils -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/lwip/include/apps -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/lwip/include/apps/sntp -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/lwip/lwip/src/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/lwip/port/esp32/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/lwip/port/esp32/include/arch -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/lwip/lwip/src/include/compat -IC:/iot/FreeRTOS/libraries/ble/include -IC:/iot/FreeRTOS/libraries/logging/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/app_update/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/bootloader_support/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/esp32/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/esp_event/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/efuse/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/driver/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/heap/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/log/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/nvs_flash/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/pthread/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/newlib/platform_include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/soc/soc/esp32/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/soc/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/spi_flash/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/vfs/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/esp_ringbuf/include -IC:/iot/FreeRTOS/vendors/espressif/boards/esp32/../components/freertos/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/xtensa/esp32/include/xtensa -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/xtensa/esp32/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/esp_rom/include -IC:/iot/FreeRTOS/vendors/espressif/esp-idf/components/esp_timer/include [... cut ...]
                                                                 ^~
C:/iot/FreeRTOS/vendors/espressif/boards/ports/ble/nimble/iot_ble_hal_common_gap.c:284:31: error: 'struct ble_gap_event' has no member named 'pairing_req'
                 else if( event->pairing_req.max_enc_key_size < BLE_SM_PAIR_KEY_SZ_MAX )
                               ^~
[291/1634] Building C object CMakeFiles/afr_ble_hal.dir/vendors/espressif/boards/ports/ble/iot_ble_hal_gap.c.obj
ninja: build stopped: subcommand failed.

jasonpcarroll, 2022-03-31:
Hi @larryxia-ddl ,

Could you let me know what version/commit ID of AFR you are using?

Best,

Jason Carroll

jasonpcarroll, 2022-03-31:
Oh I see you said you got it from AWS Console. Do you mean the Online Configuration Wizard?

larryxia-ddl, 2022-03-31:
Yes, I got from AWS wizard. 

Release 202107-00  ESP32-DevKitc

larryxia-ddl, 2022-04-01:
I updated the ESP IDF to the latest version, 4.4. still can not build. but the error is different. 

C:\iot\FreeRTOS-1>cmake --build ..\build\
[0/2] Re-checking globbed directories...
[10/1717] Building C object CMakeFiles/afr_wifi.dir/vendors/espressif/esp-idf/components/mbedtls/port/esp32/bignum.c.obj
FAILED: CMakeFiles/afr_wifi.dir/vendors/espressif/esp-idf/components/mbedtls/port/esp32/bignum.c.obj
C:\usr\esp\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DAFR_ESP_LWIP -DESP_PLATFORM -DIDF_VER=\"v4.4\" -DMBEDTLS_CONFIG_FILE=\"C:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/mbedtls/port/include/mbedtls/esp_config.h\" -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -IC:/iot/FreeRTOS-1/libraries/abstractions/wifi/include -IC:/iot/FreeRTOS-1/libraries/c_sdk/standard/common/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/mbedtls/port/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/xtensa/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/soc/src/esp32/include -IC:/iot/FreeRTOS-1/libraries/3rdparty/pkcs11 -IC:/iot/FreeRTOS-1/libraries/3rdparty/mbedtls_config -IC:/iot/FreeRTOS-1/libraries/3rdparty/mbedtls/include/mbedtls -IC:/iot/FreeRTOS-1/libraries/3rdparty/mbedtls_utils -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/lwip/lwip/src/include/compat -IC:/iot/FreeRTOS-1/freertos_kernel/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/app_update/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/bootloader_support/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/esp32/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/esp_event/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/efuse/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/driver/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/heap/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/log/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/nvs_flash/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/pthread/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/newlib/platform_include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/soc/soc/esp32/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/soc/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/spi_flash/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/vfs/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/esp_ringbuf/include -IC:/iot/FreeRTOS-1/vendors/espressif/boards/esp32/../components/freertos/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/xtensa/esp32/include/xtensa -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/xtensa/esp32/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/esp_rom/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/esp_timer/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/esp_common/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/esp_system/include -IC:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/tcpip_adapter/include -IC:/iot/FreeRTOS-1/vendors/espressif/boards/esp32/aws_demos/config_files/default_pkcs11_config -IC:/iot/FreeRTOS-1/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/include -IC:/iot/FreeRTOS-1/demos/include -IC:/iot/FreeRTOS-1/vendors/espressif/boards/esp32/aws_demos/config_files -Iconfig -isystem C:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/lwip/include/apps -isystem C:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/lwip/include/apps/sntp -isystem C:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/lwip/lwip/src/include -isystem C:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/lwip/port/esp32/include -isystem C:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/lwip/port/esp32/include/arch -isystem C:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/protocomm/include/common -isystem C:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/protocomm/include/security -isystem C:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/protocomm/include/transports -isystem C:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/wifi_provisioning/include -isystem C:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/esp_wifi/include -isystem C:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/esp_netif/include -isystem C:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/esp_eth/include -isystem C:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/soc/soc/include -isystem C:/iot/FreeRTOS-1/libraries/coreHTTP/source/dependency/3rdparty/http_parser -isystem C:/iot/FreeRTOS-1/libraries/3rdparty/mbedtls/include -isystem C:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/protobuf-c/protobuf-c -isystem C:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/protocomm/proto-c -isystem C:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/protocomm/src/common -isystem C:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/wifi_provisioning/proto-c -isystem C:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/esp_http_server/include -mlongcalls -Wno-maybe-uninitialized -std=gnu99 -Wno-old-style-declaration -Wno-stringop-overflow -Wno-bool-operation -Wno-sizeof-pointer-div -Wno-nonnull -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprec [... cut ...]
C:/iot/FreeRTOS-1/vendors/espressif/esp-idf/components/mbedtls/port/esp32/bignum.c:62:5: error: implicit declaration of function '_lock_release'; did you mean '__sync_lock_release'? [-Werror=implicit-function-declaration]
     _lock_release(&mpi_lock);
     ^~~~~~~~~~~~~
     __sync_lock_release
cc1.exe: some warnings being treated as errors
[15/1717] Building C object CMakeFiles/afr_wifi.dir/vendors/espressif/esp-idf/components/mbedtls/port/esp32/sha.c.obj
ninja: build stopped: subcommand failed.

jasonpcarroll, 2022-04-01:
Ok thank you for your logs! I will try to reproduce the issue on my end.

jasonpcarroll, 2022-04-01:
I should have asked this before - but do you know which specific choice on the OCW you downloaded? There are a number of different combinations. I am assuming you downloaded Connect to AWS IoT - ESP32-DevKitC? See attached screenshot.
[image]

larryxia-ddl, 2022-04-01:
Thanks. I followed the exact steps from this blog. However, the blogger can build but I got the error. 

https://www.souichi.club/en/aws/amazon-freertos/
https://www.souichi.club/en/aws/amazon-freertos-ota/

larryxia-ddl, 2022-04-01:
The only different is I tried to build the latest release that provided by the AWS freertos wizard, he use an older release in 2020.

larryxia-ddl, 2022-04-01:
I tried to build on Mac, it is better, but still can not build. 

800/1634] Building C object CMakeFiles/afr_ble_hal.dir/vendors/espressif/esp-idf/components/mbedtls/port/mbedtls_debug.c.obj
FAILED: CMakeFiles/afr_ble_hal.dir/vendors/espressif/esp-idf/components/mbedtls/port/mbedtls_debug.c.obj 
/Users/lxia/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc @CMakeFiles/afr_ble_hal.dir/vendors/espressif/esp-idf/components/mbedtls/port/mbedtls_debug.c.obj.rsp -MD -MT CMakeFiles/afr_ble_hal.dir/vendors/espressif/esp-idf/components/mbedtls/port/mbedtls_debug.c.obj -MF CMakeFiles/afr_ble_hal.dir/vendors/espressif/esp-idf/components/mbedtls/port/mbedtls_debug.c.obj.d -o CMakeFiles/afr_ble_hal.dir/vendors/espressif/esp-idf/components/mbedtls/port/mbedtls_debug.c.obj -c /Users/lxia/iot/FreeRTOS/vendors/espressif/esp-idf/components/mbedtls/port/mbedtls_debug.c
In file included from /Users/lxia/iot/FreeRTOS/libraries/3rdparty/mbedtls/include/mbedtls/platform_time.h:77,
                 from /Users/lxia/iot/FreeRTOS/libraries/3rdparty/mbedtls/include/mbedtls/platform.h:68,
                 from /Users/lxia/iot/FreeRTOS/vendors/espressif/esp-idf/components/mbedtls/port/mbedtls_debug.c:18:
/Users/lxia/esp/esp-idf/components/newlib/platform_include/time.h:28:19: error: unknown type name 'clockid_t'; did you mean '_lock_t'?
 int clock_settime(clockid_t clock_id, const struct timespec *tp);
                   ^~~~~~~~~
                   _lock_t
/Users/lxia/esp/esp-idf/components/newlib/platform_include/time.h:29:19: error: unknown type name 'clockid_t'; did you mean '_lock_t'?
 int clock_gettime(clockid_t clock_id, struct timespec *tp);
                   ^~~~~~~~~
                   _lock_t
/Users/lxia/esp/esp-idf/components/newlib/platform_include/time.h:30:18: error: unknown type name 'clockid_t'; did you mean '_lock_t'?
 int clock_getres(clockid_t clock_id, struct timespec *res);
                  ^~~~~~~~~
                  _lock_t
In file included from /Users/lxia/iot/FreeRTOS/libraries/3rdparty/mbedtls/include/mbedtls/platform.h:68,
                 from /Users/lxia/iot/FreeRTOS/vendors/espressif/esp-idf/components/mbedtls/port/mbedtls_debug.c:18:
/Users/lxia/iot/FreeRTOS/libraries/3rdparty/mbedtls/include/mbedtls/platform_time.h:78:9: error: unknown type name 'time_t'
 typedef time_t mbedtls_time_t;
         ^~~~~~
[806/1634] Building C object CMakeFiles/afr_ble_hal.dir/vendors/espressif/esp-idf/components/mbedtls/port/net_sockets.c.obj
FAILED: CMakeFiles/afr_ble_hal.dir/vendors/espressif/esp-idf/components/mbedtls/port/net_sockets.c.obj 
/Users/lxia/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc @CMakeFiles/afr_ble_hal.dir/vendors/espressif/esp-idf/components/mbedtls/port/net_sockets.c.obj.rsp -MD -MT CMakeFiles/afr_ble_hal.dir/vendors/espressif/esp-idf/components/mbedtls/port/net_sockets.c.obj -MF CMakeFiles/afr_ble_hal.dir/vendors/espressif/esp-idf/components/mbedtls/port/net_sockets.c.obj.d -o CMakeFiles/afr_ble_hal.dir/vendors/espressif/esp-idf/components/mbedtls/port/net_sockets.c.obj -c /Users/lxia/iot/FreeRTOS/vendors/espressif/esp-idf/components/mbedtls/port/net_sockets.c
In file included from /Users/lxia/iot/FreeRTOS/libraries/3rdparty/mbedtls/include/mbedtls/platform_time.h:77,
                 from /Users/lxia/iot/FreeRTOS/libraries/3rdparty/mbedtls/include/mbedtls/platform.h:68,
                 from /Users/lxia/iot/FreeRTOS/vendors/espressif/esp-idf/components/mbedtls/port/net_sockets.c:33:
/Users/lxia/esp/esp-idf/components/newlib/platform_include/time.h:28:19: error: unknown type name 'clockid_t'; did you mean '_lock_t'?
 int clock_settime(clockid_t clock_id, const struct timespec *tp);
                   ^~~~~~~~~
                   _lock_t
/Users/lxia/esp/esp-idf/components/newlib/platform_include/time.h:29:19: error: unknown type name 'clockid_t'; did you mean '_lock_t'?
 int clock_gettime(clockid_t clock_id, struct timespec *tp);
                   ^~~~~~~~~
                   _lock_t
/Users/lxia/esp/esp-idf/components/newlib/platform_include/time.h:30:18: error: unknown type name 'clockid_t'; did you mean '_lock_t'?
 int clock_getres(clockid_t clock_id, struct timespec *res);
                  ^~~~~~~~~
                  _lock_t
In file included from /Users/lxia/iot/FreeRTOS/libraries/3rdparty/mbedtls/include/mbedtls/platform.h:68,
                 from /Users/lxia/iot/FreeRTOS/vendors/espressif/esp-idf/components/mbedtls/port/net_sockets.c:33:
/Users/lxia/iot/FreeRTOS/libraries/3rdparty/mbedtls/include/mbedtls/platform_time.h:78:9: error: unknown type name 'time_t'
 typedef time_t mbedtls_time_t;
         ^~~~~~
/Users/lxia/iot/FreeRTOS/vendors/espressif/esp-idf/components/mbedtls/port/net_sockets.c: In function 'mbedtls_net_connect':
/Users/lxia/iot/FreeRTOS/vendors/espressif/esp-idf/components/mbedtls/port/net_sockets.c:108:9: error: implicit declaration of function 'close'; did you mean 'pclose'? [-Werror=implicit-function-declaration]
         close( fd );
         ^~~~~
         pclose
/Users/lxia/iot/FreeRTOS/vendors/espressif/esp-idf/components/mbedtls/port/net_sockets.c: In function 'net_would_block':
/Users/lxia/iot/FreeRTOS/vendors/espressif/esp-idf/components/mbedtls/port/net_sockets.c:199:17: error: 'errno' undeclared (first use in this function)
     int error = errno;
                 ^~~~~
[... cut ...]
/Users/lxia/iot/FreeRTOS/vendors/espressif/esp-idf/components/mbedtls/port/net_sockets.c:414:41: error: 'ECONNRESET' undeclared (first use in this function)
         if ( errno == EPIPE || errno == ECONNRESET ) {
                                         ^~~~~~~~~~
/Users/lxia/iot/FreeRTOS/vendors/espressif/esp-idf/components/mbedtls/port/net_sockets.c:418:23: error: 'EINTR' undeclared (first use in this function)
         if ( errno == EINTR ) {
                       ^~~~~
cc1: some warnings being treated as

larryxia-ddl, 2022-04-01:
I got the package from AWS freertos console. version and platform are default. I added OTA library and select OTA demo.

jasonpcarroll, 2022-04-01:
Ok so I think whats going on here is that AFR actually uses its own version of ESP-IDF found in vendors/espressif/esp-idf. I was able to build by opening a command prompt, navigating to the FreeRTOS/vendors/espressif/esp-idf, running install.bat then export.bat. Then, in the same command prompt, running idf.py -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 build. You can then run idf.py -p < PORT > flash monitor.

jasonpcarroll, 2022-04-01:
Let me know if this works for you.

jasonpcarroll, 2022-04-01:
You can refer to these instructions: https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html

jasonpcarroll, 2022-04-01:
So if you look at the instructions - it just wants you to run the installer of toolchains - as in the xtensa compilers etc - and then stop at the Get ESP-IDF step and instead use the IDF version provided in AFR. I agree that this is confusing and I believe the original reason for this was because ESP-IDF from espressif comes with a FreeRTOS port, however, AFR uses FreeRTOS Kernel's own ports for Xtensa. The reason for this is so we use our versions of the FreeRTOS-Kernel vs ESP-IDF's modified version. With that said, I am currently using Windows, so I'm wondering why you are still experiencing the issues you've mentioned. Are you trying to run from an ESP-IDF command prompt vs. using a regular windows command prompt and running those commands mentioned above? I used a regular command prompt.

jasonpcarroll, 2022-04-01:
Feel free to leave a build log for Windows and I can take a look.

larryxia-ddl, 2022-04-04:
I got error when run install.bat, it complains that I don't have C++ 14 or later. However, after I follow the instructions and got Microsoft Visual C++ 14 (14, 29, 30139) installed, I still got same error. 

 error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for gevent
Successfully built esp-windows-curses
Failed to build gevent
ERROR: Could not build wheels for gevent, which is required to install pyproject.toml-based projects
Traceback (most recent call last):
  File "C:\iot\FreeRTOS-1\vendors\espressif\esp-idf\tools\idf_tools.py", line 1492, in <module>
    main(sys.argv[1:])
  File "C:\iot\FreeRTOS-1\vendors\espressif\esp-idf\tools\idf_tools.py", line 1488, in main
    action_func(args)
  File "C:\iot\FreeRTOS-1\vendors\espressif\esp-idf\tools\idf_tools.py", line 1215, in action_install_python_env
    subprocess.check_call(run_args, stdout=sys.stdout, stderr=sys.stderr)
  File "C:\usr\python\lib\subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['C:\\usr\\esp\\python_env\\idf4.2_py3.10_env\\Scripts\\python.exe', '-m', 'pip', 'install', '--no-warn-script-location', '-r', 'C:\\iot\\FreeRTOS-1\\vendors\\espressif\\esp-idf\\requirements.txt']' returned non-zero exit status 1.

jasonpcarroll, 2022-04-05:
I am running python 3.8.7 and it seems to work for me.

larryxia-ddl, 2022-04-05:
Thanks. After I downgrade my python3 to 3.8.x, I can complete the
installation and build. It is a bumpy road to get here.

larryxia-ddl, 2022-04-05:
issue resolved.

closed 2022-04-05 by larryxia-ddl

jasonpcarroll, 2022-04-05:
Glad we could resolve your issue. Sorry for the inconveniences - we will work to improve this in the future.
Best,
Jason