aws/amazon-freertos #3477: [General] ST-LINK GDB Server Quit Unexpectedly.

sumanclimote, 2022-04-03:
**Briefly summarize the issue being raised**
**ST-Util GDB Server Quit Unexpectedly. See gdb-server output for more details. **

**Describe the desired outcome**
Able to do step by step debugging 

**System information**
- Hardware board: STM32L496VGTX based custom board
- IDE used: VS code
- Operating System:  Linux on VM
- Code version: 202107.00-58-g5adbe3026  and 202108.00-37-g5da3a389
- Project/Demo: custom application 

I'm want to do a step by step debug of the project "https://github.com/aws/amazon-freertos"/(varient), I have installed VM on windows 10 and using vs code I want to do a step by step debugging. I have installed the GDB and all relevant tools in ubuntu 18. The launch file for the same is -
`{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            //"cwd": "${workspaceFolder}/hotwater-fw/",
            "cwd": "${workspaceFolder}",
            "executable": "/home/suman/sandbox/hotwater-fw/build/aws_demos.elf",
            //"executable": "/hotwater-fw/build/aws_demos.elf",
            "name": "Debug hotwater code",
            "request": "launch",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            "showDevDebugOutput": true, //"none",            
            //"servertype": "openocd",
            "servertype": "stlink",
            //"servertype": "stutil",
            "gdbpath": "/usr/bin/gdb-multiarch"
        }

    ]        
}` 

When start debugging in vscode using F5/Run->Start Debugging I get a error with below text in debug terminal-

 
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /home/suman/sandbox/hotwater-fw/build/aws_demos.elf

Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /home/suman/sandbox/hotwater-fw/build/aws_demos.elf

Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2 /home/suman/sandbox/hotwater-fw/build/aws_demos.elf

  Set "showDevDebugOutput": true in your "launch.json" to see verbose GDB transactions here. Helpful to debug issues or report problems

Launching gdb-server: st-util -p 50000 --no-reset

  Please check TERMINAL tab (gdb-server) for output from st-util

Finished reading symbols from objdump: Time: 304 ms

**ST-Util GDB Server Quit Unexpectedly. See gdb-server output for more details.**

I did change the server to ST-Link but the error remains.

Can you please help me resolve the same.

jasonpcarroll, 2022-04-05:
Hi @sumanclimote,
Sorry for the late reply. Have you looked at your gdb-server output to see why it's quitting? This may be a question for the ST forums.
Best,
Jason

sumanclimote, 2022-04-07:
well true thanks for the same. In any case if anyone is looking they can use the below config file for debug - 
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "cwd": "${workspaceRoot}" ,"/home/suman/sandbox/",
            "executable": "/<path>/build/aws_demos.elf",   
            "name": "Debug ",
            "request": "launch",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            "showDevDebugOutput": true,
            "servertype": "stlink",
            "gdbpath": "/usr/bin/gdb-multiarch",
            "svdFile": "/<path>/STM32L4x1.svd",
            "device": "<device name>",            
        }

    ]        
}

closed 2022-04-07 by sumanclimote