aws/amazon-freertos #3478: [General] Build aws_demo into Espressif IDE for esp32

WilliamFrasson, 2022-04-04:
I'm encountering problems in configure, generate and compile the esp32 aws_demo into Espressif IDE.

here below when i try to compile from aws_demo from the Espressif IDE; i guess the problem is due to some set into CMakeLists.txt

here my steps:

here my steps:
1.	folder with amazon-freertos is : "D:\esp_test4\aws\amazon-freertos"
2.	i have create a new project by import the it into Espressif IDE as "Existing IDF project" importing from "D:\esp_test4\aws\amazon-freertos"
3.	for this project, in edit configuration, I have added the "Additional CMake arguments" : -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32
4.	into the workspace environment i have changed the IDF_PATH to "D:\esp_test4\aws\amazon-freertos\vendors\espressif\esp-idf"
5.	into the workspace environment i have changed the PATH to the same as achieved calling from a shell:
vendors\espressif\esp-idf\export.bat
6. now i build the project using the build button

this is the log:
[build_failed.txt](https://github.com/aws/amazon-freertos/files/8412992/build_failed.txt)

can someone help in fix the cmakefiles?

WilliamFrasson, 2022-04-04:
i see that files likes

bugnum.c
includes  "#include <sys/lock.h>" 

static _lock_t mpi_lock;

is included from Espressif framework in 
C:\Espressif\frameworks\esp-idf\components\...
instrad of 
vendors/espressif/esp-idf/components/ ...

can someone help me?

jasonpcarroll, 2022-04-05:
Hi @WilliamFrasson,
Thank you for reaching out - sorry for the late reply. Would you be able to tell me what version of this repo you are using?
Best,
Jason Carroll

jasonpcarroll, 2022-04-05:
Oh ok, just so I can try to recreate your issue - would you mind telling me OS, version of Espressif IDE (eclipse or vscode and version number)? It looks like you are using mainline as well which won't have IDE project files updated necessarily. I would recommend using a release version by checking out 202203.00 tag. I will continue to look into this further.

jasonpcarroll, 2022-04-05:
towards the top of your CMAKE log I see 
```
Environment IDF_PATH will be used for the build:
                              C:/Espressif/frameworks/esp-idf
```
So for some reason the environment path is still using Espressif's ESP-IDF rather than our fork... hmm.

WilliamFrasson, 2022-04-05:
when you wrote the path you turned on a light bulb ... the solution was under my nose, 

folder with amazon-freertos: "D:\esp_test4\aws\amazon-freertos"

1. i changed the IDF_PATH in windows environment variables
from original ESP_IDF location to aws fork
C:\Espressif\frameworks\esp-idf  to D:\esp_test4\aws\amazon-freertos\vendors\espressif\esp-idf
(this step 1.. I GUESS IT IS THE CRITICAL POINT)

2. and then I have open Espressif IDE 

3. from menu "Espressif-> Download or use ESP-IDF"  i have selected "use an existing ESP-IDF directory ..."  i select the aws one: D:\esp_test4\aws\amazon-freertos\vendors\espressif\esp-idf

4. i have create a new project by IMPORT it into Espressif IDE. "File->Import", select "Existing IDF project"  and then select the folder "D:\esp_test4\aws\amazon-freertos"

5. for this project, in edit configuration, I have added the "Additional CMake arguments" : -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32

6. into the workspace environment , from menu Windows->Preferences   i have changed that IDF_PATH to "D:\esp_test4\aws\amazon-freertos\vendors\espressif\esp-idf" and PATH contains the 
	C:\Espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Espressif\tools\cmake\3.16.4\bin;C:\Espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.0\;C:\Espressif\tools\idf-exe\1.0.1\;C:\Espressif\tools\ccache\3.7\;C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Espressif\python_env\idf4.2_py3.10_env\Scripts;D:\esp_test4\aws\amazon-freertos\vendors\espressif\esp-idf\tools
	
	as it was setted running the script "vendors\espressif\esp-idf\export.bat" from a command line shell. I don't know if it is 
        really useful

7. now i build the project using the build button

and all is generated, compiled and build fine!
here is the log:   [build_log3_from_Espressif_IDE_ok.txt](https://github.com/aws/amazon-freertos/files/8421867/build_log3_from_Espressif_IDE_ok.txt)

this is my versions:
    Operating System: windows 10
    Java Runtime Version: 17.0.2+8-LTS-86
    Eclipse Version: 4.22.0.v20211124-1800
    Eclipse CDT Version: 10.5.0.202109270023
    IDF Eclipse Plugin Version: 2.4.2.202203081345
    fatal: not a git repository: ../../../.git/modules/vendors/espressif/esp-idf
    WARNING: Git version unavailable, reading from source
    ESP-IDF v4.2.2
    Python set for IDF_PYTHON_ENV: Python 3.8.7
    Python declared in PATH variable: Python 3.10.2

thanks for focus me ! :) 
i like to share the detailed steps to others 👍

jasonpcarroll, 2022-04-05:
@WilliamFrasson no problem! thank you for sharing your solution as well! Have a good one :)

WilliamFrasson, 2022-04-05:
still problem for flash

Building in: D:\esp_test4\aws\amazon-freertos\build
cmake --build . -- -v
[0/2] C:\Espressif\tools\cmake\3.16.4\bin\cmake.exe -P D:\esp_test4\aws\amazon-freertos\build\CMakeFiles\VerifyGlobs.cmake
[1/3] cmd.exe /C "cd /D D:\esp_test4\aws\amazon-freertos\build\bootloader && C:\Espressif\tools\cmake\3.16.4\bin\cmake.exe --build ."
ninja: no work to do.
Build complete (0 errors, 0 warnings): D:\esp_test4\aws\amazon-freertos\build
C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe D:\esp_test4\aws\amazon-freertos\vendors\espressif\esp-idf\tools\idf_size.py D:/esp_test4/aws/amazon-freertos/build/aws_demos
C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe D:\esp_test4\aws\amazon-freertos\vendors\espressif\esp-idf/components/partition_table/gen_esp32part.py D:/esp_test4/aws/amazon-freertos/build/partition_table/partition-table.bin
Total time taken to build the project: 13,197 ms

-->
Variable references non-existent resource : ${workspace_loc:amazon-freertos}

jasonpcarroll, 2022-04-05:
seems to be another pathing issue with eclipse? my guess is ${workspace_loc:amazon-freertos} is supposed to be ${workspace_loc:D:\esp_test4\aws\amazon-freertos) that way it can find the build folder.

closed 2022-04-06 by WilliamFrasson