aws/amazon-freertos #3480: [General] Nanosecond-level Clock Synchronization

sumanclimote, 2022-04-08:
**Briefly summarize the issue being raised**
How can we archive a micro/Nano second level time synchronization between edge computer and AWS server. 

**Describe the desired outcome**
How Synchronization time to a nano seconds accuracy between the edge device ? I want to detect an even at different edge devices and get the time difference between them at a nano seconds. How can we handle the RTC clock drift between the edge devices and resynch them to a master clock (AWS/NPT)     

System information

Hardware board: STM32L496VGTX based custom board
IDE used: VS code
Operating System: Linux on VM
Code version: 202107.00-58-g5adbe3026 and 202108.00-37-g5da3a389
Project/Demo: custom application

**References**
Just a talk on this topic, but am expecting an AWS solution for this?  
https://youtu.be/Opf9CBwP5R8 

**Code/Steps to reproduce the current status**
N/A

**Additional context**
N/A

archigup, 2022-04-08:
Hi, for time synchronization with the SNTP protocol, we provide the coreSNTP library here: https://github.com/FreeRTOS/coreSNTP

kstribrnAmzn, 2022-04-13:
@sumanclimote [coreSNTP](https://github.com/FreeRTOS/coreSNTP) (and any [Network Time Protocol](https://en.wikipedia.org/wiki/Network_Time_Protocol)) will fail to meet your requirement of nanosecond accuracy. While coreSNTP does output a time value with enough precision to answer in nanoseconds (see [RFC 4330 pg 7](https://datatracker.ietf.org/doc/html/rfc4330#page-7) - which coreSNTP adheres to), ultimately network latencies can and will introduce both some clock jitter and skew which roughly leaves the accuracy valid only to milliseconds. The general consensus seems to be to expect no better than millisecond accuracy from _any_ NTP. 

[PTP](https://en.wikipedia.org/wiki/Precision_Time_Protocol), or the Precision Time Protocol, was created to offer sub-millisecond accuracy. There appear to be a [number of implementations](https://en.wikipedia.org/wiki/List_of_PTP_implementations#Software) - maybe one has been ported to an RTOS.

Now if you need millisecond precision, which sounded like it may work for your post, coreSNTP would be a great place to start; however, the time accuracy is in part determined by the network (latency, reliability, etc) so you may need to do some testing of your own to determine the actual clock accuracy.

sumanclimote, 2022-04-20:
thanks for sharing this info.

closed 2022-04-20 by sumanclimote