aws/amazon-freertos #3485: [General] Cmake Problems with CLion

loncsemi, 2022-04-26:
Hey, i wanted to get started with my first ESP32 AWS-FreeRTOS Project, and i want to use CLion. But when i open the amazon-freertos folder in CLion i get this output:

"C:\Program Files\JetBrains\CLion 2021.2.2\bin\cmake\win\bin\cmake.exe" -DCMAKE_BUILD_TYPE=Debug -DCMAKE_DEPENDS_USE_COMPILER=FALSE -G "CodeBlocks - MinGW Makefiles" C:\Users\loncs\Dokumente\esp32_aws_home\amazon-freertos
-- Submodule update
Skipping submodule '../../../libraries/abstractions/backoff_algorithm/'
Skipping submodule '../../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/coreHTTP/'
Skipping submodule '../../../libraries/coreHTTP/'
Skipping submodule '../../../libraries/coreHTTP/'
Skipping submodule '../../../libraries/coreJSON/'
Skipping submodule '../../../libraries/coreJSON/'
Skipping submodule '../../../libraries/coreJSON/'
Skipping submodule '../../../libraries/coreMQTT/'
Skipping submodule '../../../libraries/coreMQTT/'
Skipping submodule '../../../libraries/coreMQTT/'
Skipping submodule '../../../libraries/coreMQTT-Agent/source/dependency/coreMQTT/'
Skipping submodule '../../../libraries/coreMQTT-Agent/source/dependency/coreMQTT/'
Skipping submodule '../../../libraries/coreMQTT-Agent/source/dependency/coreMQTT/'
Skipping submodule '../../../libraries/device_defender_for_aws/'
Skipping submodule '../../../libraries/device_defender_for_aws/'
Skipping submodule '../../../libraries/device_defender_for_aws/'
Skipping submodule '../../../libraries/device_shadow_for_aws/'
Skipping submodule '../../../libraries/device_shadow_for_aws/'
Skipping submodule '../../../libraries/device_shadow_for_aws/'
Skipping submodule '../../libraries/freertos_plus/standard/freertos_plus_tcp/'
Skipping submodule '../../libraries/freertos_plus/standard/freertos_plus_tcp/'
Skipping submodule '../../libraries/freertos_plus/standard/freertos_plus_tcp/'
Skipping submodule '../../../libraries/jobs_for_aws/'
Skipping submodule '../../../libraries/jobs_for_aws/'
Skipping submodule '../../../libraries/jobs_for_aws/'
Skipping submodule '../../../libraries/ota_for_aws/'
Skipping submodule '../../../libraries/ota_for_aws/'
Skipping submodule '../../../libraries/ota_for_aws/'
Skipping submodule '../../../libraries/ota_for_aws/source/dependency/coreJSON/'
Skipping submodule '../../../libraries/ota_for_aws/source/dependency/coreJSON/'
Skipping submodule '../../../libraries/ota_for_aws/source/dependency/coreJSON/'
CMake Error at CMakeLists.txt:53 (message):
  Board is not supported: vendor.board

-- Configuring incomplete, errors occurred!
See also "C:/Users/loncs/Dokumente/esp32_aws_home/amazon-freertos/cmake-build-debug/CMakeFiles/CMakeOutput.log".

[Finished]

Does anybody know how to fix this?

ravibhagavandas, 2022-05-27:
It seems from the output of cmake command run, its using the default options for building the ESP32 project:
```
"C:\Program Files\JetBrains\CLion 2021.2.2\bin\cmake\win\bin\cmake.exe" -DCMAKE_BUILD_TYPE=Debug -DCMAKE_DEPENDS_USE_COMPILER=FALSE -G "CodeBlocks - MinGW Makefiles"
```
The required cmake options for building the project is :
```
cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B ./YOUR_BUILD_DIRECTORY -DAFR_ENABLE_ALL_MODULES=1 -DAFR_ENABLE_TESTS=0
```
More details can be found under section `Build and Flash FreeRTOS with CMake` in the getting started guide for ESP32:
https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html

aggarg, 2022-06-03:
I am closing this issue. Please feel free to reopen or create a new one if you still face any issue.

closed 2022-06-03 by aggarg