aws/amazon-freertos #3486: [General] Iot_CreateDetachedThread does not have threadName argument

laszlopataki, 2022-05-04:
**Briefly summarize the issue being raised**
In AWS C-SDK there is the `Iot_CreateDetachedThread` function which does not accept a name argument. 
The name parameter is useful to identify a certain task when calling `vTaskList` or using tools like Tracealyzer. 
Is there any particular reason why this argument does not exists and the function creates tasks with a constant "iot_thread" name?

**Describe the desired outcome**
`Iot_CreateDetachedThread` accepts name parameters, so the Platform layer can be used to name Tasks.

ravibhagavandas, 2022-05-28:
Hi,
Could you describe more about your case for `Iot_CreateDetachedThread()` over `xTaskCreate()` API .  As i looked through other repositories such as CSDK the API is currently not being used anywhere. So instead of using the overhead of another abstraction, could you directly use `xTaskCreate` API ?

laszlopataki, 2022-05-31:
Hi, 
My understanding of this abstraction layer is that the application code can be platform-independent. After porting you can run the same code on PC, on different MCUs, and with different OS.

What I did is forked the repo and created an `Iot_CreateNamedDetachedThread` with a name parameter. 

My aim with this query was to give feedback in case these APIs are meant to be used in the future and further develop them. 
If it is only an experiment or some demo-related function, please ignore this post.

closed 2022-06-01 by ravibhagavandas