aws/amazon-freertos #3487: [General] MQTT BLE OTA : No update notification

Steven35700, 2022-05-05:
**Briefly summarize the issue being raised**
I configured the "CONFIG_OTA_MQTT_BLE_TRANSPORT_DEMO_ENABLED" demo on my ESP32 with this [tutorial](https://docs.aws.amazon.com/en_en/freertos/latest/userguide/ota-updates-esp32-ble.html). I manage to connect the ESP32 to the MQTT proxy fine and everything seems to work. Unfortunately, it seems that the ESP32 does not receive a notification that there is a new firmware.

**Describe the desired outcome**
Normally, the ESP32 should receive notification that an update is available and start downloading it, then check the code signing before rebooting with this new version.

**System information**
- Hardware board: ESP32-DevKitC
- IDE used: VSCode
- Operating System: Windows
- Code version: 202107.00-62-g3513d937c
- Demo: CONFIG_OTA_MQTT_BLE_TRANSPORT_DEMO_ENABLED

**References**
ESP32 Logs :
```
5 18042 [iot_thread] [INFO ][DEMO][180420] ---------STARTING DEMO---------

I (180987) BTDM_INIT: BT controller compile version [bfbbe1e]
6 18042 [iot_thread] [INFO ][INIT][180420] SDK successfully initialized.

I (180997) system_api: Base MAC address is not set
I (181007) system_api: read default base MAC address from EFUSE
I (181017) phy_init: phy_version 4660,0162888,Dec 23 2020
W (181017) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=187 adv_itvl_max=375
7 18091 [iot_thread] [INFO ][DEMO][180910] No networks connected for the demo. Waiting for a network connection.

8 56053 [ble] [INFO ][DEMO][560530] BLE Connected to remote device, connId = 0

9 56053 [NetworkManager] [INFO ][DEMO][560530] Invoked callback for network state change event.

10 56054 [iot_thread] [INFO ][DEMO][560540] Successfully initialized the demo. Network type for the demo: 2

11 56054 [iot_thread] [ota_demo_mqtt_ble_transport.c:1922] [INFO] [OTA] OTA over MQTT demo, Application version 0.1.0
12 56207 [NumericComparis] Numeric comparison:844778

13 56208 [NumericComparis] Press 'y' to confirm, 'n' to reject

14 57066 [NumericComparis] Key accepted

15 57233 [ble] [ota_demo_mqtt_ble_transport.c:1222] [INFO] [OTA] BLE Channel is Open.

16 57233 [iot_thread] [ota_demo_mqtt_ble_transport.c:1266] [INFO] [OTA] The channel was initialized successfully

17 57233 [iot_thread] [ota_demo_mqtt_ble_transport.c:1332] [INFO] [OTA] Creating an MQTT connection to a2sx0p4jwqlvgu-ats.iot.eu-west-3.amazonaws.com.

GATT procedure initiated: notify; att_handle=47
18 57341 [iot_thread] [core_mqtt.c:886] [INFO] [MQTT] Packet received. ReceivedBytes=2.

19 57341 [iot_thread] [core_mqtt_serializer.c:970] [INFO] [MQTT] CONNACK session present bit not set.

20 57341 [iot_thread] [core_mqtt_serializer.c:912] [INFO] [MQTT] Connection accepted.

21 57341 [iot_thread] [core_mqtt.c:1563] [INFO] [MQTT] Received MQTT CONNACK successfully from broker.

22 57342 [iot_thread] [core_mqtt.c:1829] [INFO] [MQTT] MQTT connection established with the broker.

23 57342 [iot_thread] [ota_demo_mqtt_ble_transport.c:1890] [INFO] [OTA]  Received: 0   Queued: 0   Processed: 0   Dropped: 0

24 57346 [OTA Agent Task] [ota_pal.c:611] [INFO] [OTA] otaPal_GetPlatformImageState

I (574047) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (574047) esp_ota_ops: [0] aflags/seq:0x2/0x1, pflags/seq:0xffffffff/0x0
25 57348 [OTA Agent Task] [ota.c:2840] [INFO] [OTA] Current State=[RequestingJob], Event=[Start], New state=[RequestingJob]

GATT procedure initiated: notify; att_handle=47
26 57442 [iot_thread] [ota_demo_mqtt_ble_transport.c:1890] [INFO] [OTA]  Received: 0   Queued: 0   Processed: 0   Dropped: 0

27 57453 [MQTT Agent Task] [core_mqtt.c:886] [INFO] [MQTT] Packet received. ReceivedBytes=3.
[... 206 lines cut ...]
2022-05-05 09:18:14.239 25400-25696/software.amazon.freertos.demo D/FRD: There's no ble command in the queue.
2022-05-05 09:18:14.693 25400-25696/software.amazon.freertos.demo D/FRD: ->->-> Characteristic changed for: MQTT_TX with data: a56177036175782524617773
2022-05-05 09:18:14.694 25400-25696/software.amazon.freertos.demo D/FRD: Processing incoming queue. size: 0
2022-05-05 09:18:14.697 25400-25696/software.amazon.freertos.demo I/FRD: Handling Mqtt Message type : 3
2022-05-05 09:18:14.701 25400-25696/software.amazon.freertos.demo I/FRD: Sending mqtt message to IoT on topic: $aws/things/OTA_Device/jobs/$next/get message: {"clientToken":":OTA_Device"} MsgID: 3
2022-05-05 09:18:14.756 25400-26098/software.amazon.freertos.demo I/AWSIotMqttManager: delivery is complete
2022-05-05 09:18:14.757 25400-26098/software.amazon.freertos.demo D/FRD: Publish msg delivery status: Success
2022-05-05 09:18:14.757 25400-26098/software.amazon.freertos.demo I/FRD: Sending PUB ACK back to device. MsgID: 3
2022-05-05 09:18:14.758 25400-26098/software.amazon.freertos.demo D/FRD: Processing BLE command: WRITE_CHARACTERISTIC remaining mqtt queue 0, network queue 0
2022-05-05 09:18:14.760 25400-26098/software.amazon.freertos.demo D/FRD: <-<-<- Writing to characteristic: MQTT_RX  with data: a2616903617704
2022-05-05 09:18:14.861 25400-25696/software.amazon.freertos.demo D/FRD: onCharacteristicWrite for: MQTT_RX; status: Success; value: a2616903617704
2022-05-05 09:18:14.861 25400-25696/software.amazon.freertos.demo D/FRD: There's no ble command in the queue.
2022-05-05 09:19:15.371 25400-25436/software.amazon.freertos.demo D/FRD: ->->-> Characteristic changed for: MQTT_TX with data: a161770c
2022-05-05 09:19:15.372 25400-25436/software.amazon.freertos.demo D/FRD: Processing incoming queue. size: 0
2022-05-05 09:19:15.373 25400-25436/software.amazon.freertos.demo I/FRD: Handling Mqtt Message type : 12
2022-05-05 09:19:15.377 25400-25436/software.amazon.freertos.demo D/FRD: Processing BLE command: WRITE_CHARACTERISTIC remaining mqtt queue 0, network queue 0
2022-05-05 09:19:15.381 25400-25436/software.amazon.freertos.demo D/FRD: <-<-<- Writing to characteristic: MQTT_RX  with data: a161770d
2022-05-05 09:19:15.481 25400-25436/software.amazon.freertos.demo D/FRD: onCharacteristicWrite for: MQTT_RX; status: Success; value: a161770d
2022-05-05 09:19:15.481 25400-25436/software.amazon.freertos.demo D/FRD: There's no ble command in the queue.
```

**Additional context**
This must not come from my job configuration or my AWS configuration because I manage to do an OTA update with the "CONFIG_OTA_MQTT_UPDATE_DEMO_ENABLED" demo.

It doesn't seem to come from my Cognito configuration because I also manage to send MQTT BLE packets with the "CONFIG_MQTT_BLE_TRANSPORT_DEMO_ENABLED" demo. However, it is true that with this demo, I have problems with the reception (subscription) of MQTT messages where the demo crashes (reboot card).

ravibhagavandas, 2022-06-15:
Hi @Steven35700,
I was able to reproduce the issue. It seems the policy ( mentioned as `mqttProxyCognitoPolicy` in the tutorial) assigned to Cognito identity role (mentioned as `Cognito_mqtt_proxy_identity_poolAuth_Role` in the tutorial) does not have permissions to receive packets (`iot:Receive`)  from broker.
Could you go to: IAM Console > Choose Roles > Search for the Cognito identity role you created > Edit Inline policy for IoT.
The updated policy should have following permissions to the resource:
```
{
    "Version": "2012-10-17",
    "Statement": [
        {
            "Sid": "VisualEditor0",
            "Effect": "Allow",
            "Action": [
                "iot:Receive",
                "iot:Subscribe",
                "iot:AttachPolicy",
                "iot:AttachPrincipalPolicy",
                "iot:Connect",
                "iot:GetThingShadow",
                "iot:ListAttachedPolicies",
                "iot:DeleteThingShadow",
                "iot:UpdateThingShadow",
                "iot:CreateCertificateFromCsr",
                "iot:Publish"
            ],
            "Resource": "*"
        }
    ]
}
```
After updating policy please reconnect the mobile app to the BLE device.

Steven35700, 2022-06-16:
Hi,

I made the change, everything works perfectly now!
So there is an error in the tutorial. Thank you very much for your time and help!

Steven

ravibhagavandas, 2022-06-16:
Glad to know it works!
I will close the issue.

closed 2022-06-16 by ravibhagavandas