aws/amazon-freertos #3488: [BUG] c_cpp_properties.json misconfigured for VS Code includes in ESP32 AWS Demos

NicoPowers, 2022-05-07:
**Describe the bug**
Please provide a clear and concise description explaining the bug.

**System information**
- Hardware board: [ ESP32 ]
- IDE used: [ VSCode ]
- Operating System: [ Windows ]
- Code version: [202203.00]
- Project/Demo: [ aws_demos ]

The includes do not exist in the repo:
[image]

I believe whoever was last doing development on this code accidentally pushed their .vscode configuration, but for some reason the code they were including was not pushed up (like the common/includes that are supposed to exists 2 directories back).

ravibhagavandas, 2022-06-01:
Hello
This workspace seems to have been pushed to repo before the folder structure was reorganized as part of this [commit](https://github.com/aws/amazon-freertos/commit/46f809d7660cc87303aeee360d62007c860e5d89) .
That vscode workspace is no longer maintained and I would recommend to instead create a workspace of your own.
For latest information about getting started on ESP32 projects you can refer the documentation here:
https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html

aggarg, 2022-06-06:
This PR deletes these vscode files, which as mentioned by @ravibhagavandas, are no longer maintained - https://github.com/aws/amazon-freertos/pull/3500