aws/amazon-freertos #3491: Automatic WiFi Reconnect

rajasekarl, 2022-05-14:
**Briefly summarize the issue being raised**
How to handling automatic WiFi reconnect for the situation where the network drops?
**Describe the desired outcome**
Identify the intended result of further discussion and eventual resolution of this issue.

**System information**
- Hardware board: [ ESP32 ]
- IDE used: [ Visual code]
- Operating System: [ Windows  ]
- Project/Demo: [ mqtt_demo_mutual_auth ]

ravibhagavandas, 2022-05-28:
Hello @rajasekarl 

You can refer to one of our demo [WiFiConnectTask](https://github.com/aws/amazon-freertos/blob/main/demos/wifi_provisioning/aws_wifi_connect_task.c#L162) which connects to a provisioned WiFi networks. The demo task uses network manager to listen for a WiFi network state change from driver and calls WiFi API to connect to network. You can easily adapt the demo to  re-connect to a WiFi network upon disconnection

aggarg, 2022-06-03:
I am closing this issue. Feel free to reopen or create a new one if you still face any issue.

closed 2022-06-03 by aggarg