aws/amazon-freertos #3494: [Feature Request] esp32s3 support

thomasklaush, 2022-05-23:
Is esp32s3 support planned?

jasonpcarroll, 2022-05-25:
Hi @thomasklaush,
Sorry for the late reply. I cannot answer whether or not esp32s3 support is planned for this repository, but we do have a repository that currently works with the C3, that I believe could easily be made to work with an s3. Here is the repo: https://github.com/FreeRTOS/iot-reference-esp32c3. At the top of this [file](https://github.com/FreeRTOS/iot-reference-esp32c3/blob/main/sdkconfig.defaults), you should be able to just remove the 3 lines from the top: CONFIG_IDF_TARGET_ARCH_RISCV=y
CONFIG_IDF_TARGET="esp32c3"
CONFIG_IDF_TARGET_ESP32C3=y
Then run:
idf.py set-target esp32s3, then follow the getting started guide. 
I have not actually tested this because I have not been able to get my hands on an s3, but I think this should work. Alternatively, we recently partnered with espressif to update their [esp-aws-iot](https://github.com/espressif/esp-aws-iot) repo. Every one of the folders inside the libraries folder of this repo can be dragged into an ESP-IDF project as a component to add any of our AWS libraries - they are also all configurable from ESP-IDF's menuconfig. This repo also offers examples that should work with the s3 in the examples folder. 

Best,

Jason Carroll

thomasklaush, 2022-05-30:
Hi @jasonpcarroll,

thank you for your response.
I get some eval boards soon, I will try.

Best regards,
Thomas

ravibhagavandas, 2022-06-01:
@thomasklaush 
Hope this answers your question. I will close the issue. Feel free to open issues if you have further questions.

closed 2022-06-01 by ravibhagavandas