aws/amazon-freertos #3496: [BUG] Wrong header when using c_sdk https client

Froopas, 2022-05-31:
Hello,

I am trying to connect to a http server using the c_sdk http client using IP.
The request goes through but I get the response "400 bad request".
Requests by other means such as postman works.

I have used wireshark in order to see the requests and notable difference between the requests that work and the ones that doesn't is that the working one has the header "Host:<ip>:<port>" whilst the ones that doesn't work (when using c_sdk client) have the header "Host:<ip>".

If I try to change the address in the request it fails to send at all.

Is there a way to add the port to the header that I have missed?

Kind regards
Froopas

archigup, 2022-06-01:
Hi, which source and version are you using for the http client?
Additionally, that library has been replaced by coreHTTP; perhaps coreHTTP might better meet your needs.

Froopas, 2022-06-02:
Hello,
The version is v1.1.2.
I am taking over a project which is already connected to an AWS instance using mqtt and I am now supposed to extend the capabilities of the embedded system to include http/https requests.
The original project is a copy of the freertos repo: https://github.com/Infineon/pred-main-xmc4700-kit

I was looking to move over to coreHTTP but it did not look like it was compatible with the use of the IoT network manager used in the current project.

Froopas, 2022-06-03:
I have realized I looked at the wrong repository when looking at coreHTTP implementation.
After using the implementation from this repository everything works.

closed 2022-06-03 by Froopas