aws/amazon-freertos #3498: How we can debug the amazon-freertos??

Raghav3107, 2022-06-04:
Hello developers.

I want to know how we can debug the amazon-freertos on VS Code IDE. 
Is there any guide for to get started with?

I checked this [link](https://www.freertos.org/2021/01/using-visual-studio-code-for-freertos-development.html), but did not get much. I m using the esp32 and I know how to debug the esp-idf codes.
But I don't get anything how we can debug the amazon-freertos codes.

Any comments on this will help.

**System information**
- Hardware board: [ esp32 ]
- IDE used: [ VS Code & ESP-IDF]
- Operating System: [ Linux  ]
- Code version: 202107

Thanks in advance for your input.

aggarg, 2022-06-06:
As mentioned in the [getting started guide](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html), you can use command line tools to debug amazon-freertos. If you want to use VS Code, you can try this repo which contains example applications using our libraries - https://github.com/espressif/esp-aws-iot

Raghav3107, 2022-06-09:
@aggarg  Thanks for the reply, I will check this out.

closed 2022-06-09 by Raghav3107