aws/amazon-freertos #3512: [General] FreeRTOS Build Integration Meaning on readme?

t-tbrown, 2022-07-20:
On [the main readme.md](https://github.com/aws/amazon-freertos/blob/main/README.md ), what does the section "In addition, AWS supports the following boards with FreeRTOS Build Integration and maintains them with each release:" mean? What does _FreeRTOS Build Integration_ include, does that mean OTA firmware upddates will work out of the box with these MCUs?

johnrhen, 2022-07-25:
The FreeRTOS Qualified Boards section on the README lists boards/MCUs which are actively supported in Amazon FreeRTOS. After completing the setup steps (see: the ["Getting Started" guides](https://github.com/aws/amazon-freertos/blob/main/README.md#freertos-qualified-boards) under each board) and the required configuration (see: [ESP32 OTA demo setup guide](https://docs.aws.amazon.com/freertos/latest/userguide/download-ota-esp.html)), they will support OTA updates.

johnrhen, 2022-08-04:
I'm going to close this issue - feel free to reopen it if you need anything else from us.

closed 2022-08-04 by johnrhen