aws/amazon-freertos #3513: how to connect local mqtt blocker server on macbook?

misupopo, 2022-07-21:
I set up a local mqtt server on my macbook.
I built it with typescript and docker.

I want to connect to the URL "mqtt://localhost:1883" to receive topic messages.

Is there any sample code or demo code instead of coreMqtt documentation?
I would appreciate it if someone teach me a more concrete sample of the connection, rather than a library.

johnrhen, 2022-07-22:
Hello misupopo,

You should be able to use the coreMQTT mutual authentication demo to test if your mqtt broker is working. Follow the demo setup guides, but when configuring the endpoint, use your URL.

https://docs.aws.amazon.com/freertos/latest/userguide/mqtt-demo-ma.html
https://docs.aws.amazon.com/freertos/latest/userguide/freertos-getting-started-demo.html
https://github.com/aws/amazon-freertos/blob/main/demos/coreMQTT/mqtt_demo_mutual_auth.c

johnrhen, 2022-07-22:
Do you already have a device you intend to use for connection?

misupopo, 2022-07-25:
Thank you for your reply.
I am using ESP32-devkitc.

misupopo, 2022-07-25:
The connection sample is as follows.

```
import * as mqtt from 'mqtt';

(async () => {
  try {
    const host = 'localhost';
    const portNumber = 1883;
    const username = 'admin';
    const password = 'admin';

    const connectUrl = `mqtt://${host}:${portNumber}/`

    const client = await mqtt.connect(connectUrl,{
      clean: true,
      connectTimeout: 4000,
      username,
      password,
      reconnectPeriod: 1000,
    });
```

johnrhen, 2022-07-25:
In that case, there's a Getting Started guide for the ESP32-DevKitC you can read through to get set up prior to running the demos.
https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html

johnrhen, 2022-08-04:
I'm going to close this issue - feel free to reopen it if you need anything else from us.

closed 2022-08-04 by johnrhen