aws/amazon-freertos #3515: [General] CMake file properties

BogdanTheGeek, 2022-07-25:
I am using amazon-freertos as a submodule in my project. I am not able to set the source file properties for files in my own source. I have added the following to my top level CMakeLists.txt:

```
...
set_source_files_properties(
   ${PROJECT_SOURCE_DIR}/src/my_app.c 
   PROPERTIES COMPILE_FLAGS
   "-Wno-unused-function")
...
```
```
├── CMakeLists.txt  <-- top level cmake
├── components
├── freertos        <-- submodule
├── freertos-configs
├── inc
└── src
    └── my_app.c
```

The reason why I am raising this issue is that even though I am adding my `src` files to the `IDF_EXECUTABLE_SRCS`, Ninja output shows my files being built by the freertos CMakeLists.txt:
```
[2/4] Building C object freertos/CMakeFiles/my_app.dir/__/src/my_app.c.obj
```

**System information**
- Hardware board: [ ESP32 ]
- IDE used: [ none ]
- Operating System: [ MacOS ]
- Code version: 202203.00-1-g92c5712f0
- Project/Demo: [ custom ]

**Additional context**
CMakeLists.txt based on https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#getting_started_espressif_cmake_project

dachalco, 2022-07-28:
Hi @BogdanTheGeek 

The CMAKE in that example doesn't have a separate module build for your source, but rather the source you define with `IDF_EXECUTABLE_SRCS` is used and built in the `amazon-freertos` build process. Hence, why you see your source built with it.

Are you certain the specified files are not being built with corresponding flags?

BogdanTheGeek, 2022-08-04:
Hi @dachalco, sorry for the late reply.
Yes, I cannot add any build flags to any sources, mine or AFR.
There are quite a few warnings in the AFR module that I would like to ignore so I can see my own warnings(Eg. unused functions in iot_atomic.h).
Do you have any recommendations to what I should do?

ravibhagavandas, 2022-08-06:
@shubhamkulkarni97 Could you provide more insights into what could be the issue here?
I think root of problem is the target and its associated properties are still defined in the CMakeLists.txt within the AFR subfolder here:
https://github.com/aws/amazon-freertos/blob/main/vendors/espressif/boards/esp32/CMakeLists.txt
Could you try adding your changes in this CMakeLists.txt file ?

shubhamkulkarni97, 2022-08-08:
Tagging @mahavirj @laukik-hase for providing more insights

BogdanTheGeek, 2022-08-08:
> @shubhamkulkarni97 Could you provide more insights into what could be the issue here? I think root of problem is the target and its associated properties are still defined in the CMakeLists.txt [...]

Yes, doing the changes in `freertos/CMakeLists.txt` works.
I would prefer it however if I could change the properties of my source files in my CMakeLists.txt and if AFR wouldn't produce any warnings.
I have a patching mechanism Im place, but I like to keep its use to a minimum.

BogdanTheGeek, 2022-08-08:
Can anyone else replicate the issue?

ravibhagavandas, 2022-08-09:
Could you post the specific warnings you see in AFR code so that we can try to fix them ?

BogdanTheGeek, 2022-08-09:
```
[1564/1616] Building C object freertos/CMakeFiles/hub_app.dir/__/src/libGSM.c.obj
In file included from ../src/libGSM.c:18:
../freertos/vendors/espressif/esp-idf/components/tcpip_adapter/include/tcpip_adapter.h:15:2: warning: #warning "This header is deprecated, please use new network related API in esp_netif.h" [-Wcpp]
 #warning "This header is deprecated, please use new network related API in esp_netif.h"
  ^~~~~~~
[1579/1616] Building C object freertos/CMakeFiles/hub_app.dir/__/src/network/ota/src/aws_esp_ota_ops.c.obj
In file included from ../src/network/ota/src/aws_esp_ota_ops.c:35:
../freertos/vendors/espressif/esp-idf/components/esp32/include/rom/queue.h:1:2: warning: #warning rom/queue.h is deprecated, please use sys/queue.h instead [-Wcpp]
 #warning rom/queue.h is deprecated, please use sys/queue.h instead
  ^~~~~~~
In file included from ../src/network/ota/src/aws_esp_ota_ops.c:36:
../freertos/vendors/espressif/esp-idf/components/esp32/include/rom/crc.h:1:2: warning: #warning rom/crc.h is deprecated, please use esp32/rom/crc.h instead [-Wcpp]
 #warning rom/crc.h is deprecated, please use esp32/rom/crc.h instead
  ^~~~~~~
In file included from ../src/network/ota/src/aws_esp_ota_ops.c:39:
../freertos/vendors/espressif/esp-idf/components/bootloader_support/include/esp_flash_data_types.h:1:2: warning: #warning esp_flash_data_types.h has been merged into esp_flash_partitions.h, please include esp_flash_partitions.h instead [-Wcpp]
 #warning esp_flash_data_types.h has been merged into esp_flash_partitions.h, please include esp_flash_partitions.h instead
  ^~~~~~~
[1581/1616] Building C object freertos/CMakeFiles/hub_app.dir/__/src/network/mqtt/src/iot_mqtt_serialize.c.obj
In file included from ../freertos/libraries/c_sdk/standard/common/include/iot_atomic.h:37,
                 from ../src/network/mqtt/src/iot_mqtt_serialize.c:47:
../freertos/freertos_kernel/include/atomic.h:396:34: warning: 'Atomic_XOR_u32' defined but not used [-Wunused-function]
 static portFORCE_INLINE uint32_t Atomic_XOR_u32( uint32_t volatile * pulDestination,
                                  ^~~~~~~~~~~~~~
../freertos/freertos_kernel/include/atomic.h:369:34: warning: 'Atomic_NAND_u32' defined but not used [-Wunused-function]
 static portFORCE_INLINE uint32_t Atomic_NAND_u32( uint32_t volatile * pulDestination,
                                  ^~~~~~~~~~~~~~~
../freertos/freertos_kernel/include/atomic.h:342:34: warning: 'Atomic_AND_u32' defined but not used [-Wunused-function]
 static portFORCE_INLINE uint32_t Atomic_AND_u32( uint32_t volatile * pulDestination,
                                  ^~~~~~~~~~~~~~
../freertos/freertos_kernel/include/atomic.h:315:34: warning: 'Atomic_OR_u32' defined but not used [-Wunused-function]
 static portFORCE_INLINE uint32_t Atomic_OR_u32( uint32_t volatile * pulDestination,
                                  ^~~~~~~~~~~~~
../freertos/freertos_kernel/include/atomic.h:288:34: warning: 'Atomic_Decrement_u32' defined but not used [-Wunused-function]
 static portFORCE_INLINE uint32_t Atomic_Decrement_u32( uint32_t volatile * pulAddend )
                                  ^~~~~~~~~~~~~~~~~~~~
../freertos/freertos_kernel/include/atomic.h:263:34: warning: 'Atomic_Increment_u32' defined but not used [-Wunused-function]
 static portFORCE_INLINE uint32_t Atomic_Increment_u32( uint32_t volatile * pulAddend )
                                  ^~~~~~~~~~~~~~~~~~~~
../freertos/freertos_kernel/include/atomic.h:237:34: warning: 'Atomic_Subtract_u32' defined but not used [-Wunused-function]
 static portFORCE_INLINE uint32_t Atomic_Subtract_u32( uint32_t volatile * pulAddend,
                                  ^~~~~~~~~~~~~~~~~~~
../freertos/freertos_kernel/include/atomic.h:176:34: warning: 'Atomic_CompareAndSwapPointers_p32' defined but not used [-Wunused-function]
 static portFORCE_INLINE uint32_t Atomic_CompareAndSwapPointers_p32( void * volatile * ppvDestination,
                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../freertos/freertos_kernel/include/atomic.h:144:32: warning: 'Atomic_SwapPointers_p32' defined but not used [-Wunused-function]
 static portFORCE_INLINE void * Atomic_SwapPointers_p32( void * volatile * ppvDestination,
                                ^~~~~~~~~~~~~~~~~~~~~~~
../freertos/freertos_kernel/include/atomic.h:108:34: warning: 'Atomic_CompareAndSwap_u32' defined but not used [-Wunused-function]
 static portFORCE_INLINE uint32_t Atomic_CompareAndSwap_u32( uint32_t volatile * pulDestination,
                                  ^~~~~~~~~~~~~~~~~~~~~~~~~
```

ravibhagavandas, 2022-08-10:
It seems the warnings are from a file `iot_mqtt_serialize.c` which is no longer part of the newer releases of this repository. The file specifically is part of a old beta version of our MQTT library and has been replaced by more modular [coreMQTT](https://github.com/FreeRTOS/coreMQTT) in our new releases. So it possible to move to latest version of the repository?
Also for using FreeRTOS and AWS C libraries within an ESP IDF project I would recommend using ESP32 C3  reference integration at  https://github.com/FreeRTOS/iot-reference-esp32c3

BogdanTheGeek, 2022-08-10:
@ravibhagavandas Yes, our project was based on that beta sadly and we had to port it forward to while we are switching to coreMQTT. I only now realise that these errors were from our code, so the original problem still stands. How can we set compile flag properties for our code?

BogdanTheGeek, 2022-08-16:
Any more ideas about how we could add compilation flags for our source code?

n9wxu, 2022-08-17:
The best choice is probably to modify freertos/CMakeLists.txt for now and fix up the problems when you are able to migrate to newer versions of the IDF.

BogdanTheGeek, 2022-08-17:
So, the best solution for using AFR is to have an IDF project and add it as a component? How would that work with the different version of freertos? I know that the C SDK exists, but all of our products already run AFR, which means that all the certificates, network provisioning and network management settings (nvs) would need to be switched over to a new system with an OTA. I am happy to close this issue, but its a really weird shortcoming of AFR that you can't actually use it as a submodule.

closed 2022-08-17 by BogdanTheGeek