aws/amazon-freertos #3518: [BUG] Core Dump fails compiling

EtienneMdv, 2022-07-28:
**Describe the bug**

I get the following error when building my project trying to use coredump:
`
../amazon-freertos/vendors/espressif/esp-idf/components/espcoredump/src/core_dump_port.c: In function 'esp_core_dump_get_tasks_snapshot':
../amazon-freertos/vendors/espressif/esp-idf/components/espcoredump/src/core_dump_port.c:281:12: error: unknown type name 'TaskSnapshot_t'
     static TaskSnapshot_t s_tasks_snapshots[CONFIG_ESP32_CORE_DUMP_MAX_TASKS_NUM];
            ^~~~~~~~~~~~~~
../amazon-freertos/vendors/espressif/esp-idf/components/espcoredump/src/core_dump_port.c:285:27: error: 'TaskSnapshot_t' undeclared (first use in this function); did you mean 'TaskStatus_t'?
     _Static_assert(sizeof(TaskSnapshot_t) >= sizeof(core_dump_task_header_t), "FreeRTOS task snapshot binary compatibility issue!");
                           ^~~~~~~~~~~~~~
                           TaskStatus_t
../amazon-freertos/vendors/espressif/esp-idf/components/espcoredump/src/core_dump_port.c:285:27: note: each undeclared identifier is reported only once for each function it appears in
../amazon-freertos/vendors/espressif/esp-idf/components/espcoredump/src/core_dump_port.c:285:20: error: expression in static assertion is not an integer
     _Static_assert(sizeof(TaskSnapshot_t) >= sizeof(core_dump_task_header_t), "FreeRTOS task snapshot binary compatibility issue!");
                    ^~~~~~
../amazon-freertos/vendors/espressif/esp-idf/components/espcoredump/src/core_dump_port.c:287:35: error: implicit declaration of function 'uxTaskGetSnapshotAll'; did you mean 'uxTaskGetSystemState'? [-Werror=implicit-function-declaration]
     uint32_t task_num = (uint32_t)uxTaskGetSnapshotAll(s_tasks_snapshots,
                                   ^~~~~~~~~~~~~~~~~~~~
                                   uxTaskGetSystemState
cc1: some warnings being treated as errors
`

It looks like `uxTaskGetSnapshotAll` and `TaskSnapshot_t` are no longer defined in FreeRTOS. This issue echoes #2820. Is there any solution for this? I am using AFR latest version.

**System information**
- Hardware board: ESP32-WROOM-32D
- Operating System: Linux 
- Project/Demo: custom application

**Steps to reproduce bug**
Configure project to use coredump.
[image]
Build project using `idf.py build`.

dachalco, 2022-07-29:
Hi @EtienneMdv 

Reviewed the last ticket. To clarify, AFAIK `uxTaskGetSnapshotAll` was never a 1st party feature provided by FreeRTOS, but rather an extension added to ESP-IDF's `components/freertos` -- a revision specific for ESP devices.

Amazon-freertos uses the 1st party FreeRTOS-Kernel esp32 port, which differs from ESP-IDF's `components/freertos` in several ways, such as the former port only supporting single-core, while the latter supports multicore. 

So the issue is rather that FreeRTOS-Kernel 1st party port for esp32 does not contain these IDF added members e.g. `uxTaskGetSnapshotAll`. There's quite a bit more additions in IDF's `components/freertos`, besides multicore support, that may make importing `uxTaskGetSnapshotAll` to the 1st party port complicated. For example there's other extended RTOS features that are used throughout `components/freertos` that don't exist in the 1st party port.

@shubhamkulkarni97 Are you able to suggest a simplest route here?

As suggested in the linked ticket, if it's possible, I would recommend using a JTAG like [this one](https://ftdichip.com/product-category/products/cables/usb-mpsse-spi-i2c-jtag-master-cable-series/) which can be hooked up to your board for full stop-mode debugging.

EtienneMdv, 2022-07-29:
Thank you for your help @dachalco. It all becomes clearer.

However, the difference between the linked ticket and my project is that I am trying to use [Core Dump](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/core_dump.html?highlight=core%20dump) and not the debugger. I don't really see any work-around...

dachalco, 2022-07-29:
@EtienneMdv 

What I mean to ask is, what are your high level intentions? Are you looking to debug the board in _any_ way, or are you looking to specifically have this work with IDF's `core dump`?

If you're looking to generally debug, the simplest route, and my best suggestion, would be to get a JTAG like the one I linked earlier, or a similar tool. In practice, such debugging capability is almost necessary for developing on MCU's.

If you're looking to use `core dump` with amazon-freertos as a means for post-mortem crash analysis, I can take a look at what can be done to the 1st party port. Though, even with 1st party support for this feature, I suspect you wouldn't be able to use IDF's `menuconfig` for its configuration, and would need to use a FreeRTOS config `*.h` instead.

EtienneMdv, 2022-07-29:
I am sorry my intentions weren't really clear. I am indeed trying to use core dump to perform post-mortem crash analysis with AFR. It isn't a problem if I can't use menuconfig to configure core dump. 

I would be very grateful if you could take a look at it.