aws/amazon-freertos #3519: How to include fatfs component

taherrera, 2022-07-29:
Hello,

I am trying to;
```
#include <esp_vfs_fat.h>
```

However, the compiler complains with:

esp_vfs_fat.h: No such file or directory

I have been trying for some time to include this into the CMakeLists.txt file but to no avail. How do I add this component to my project ?

Thanks

dachalco, 2022-08-03:
Hi @taherrera 

The following should work
```
diff --git a/vendors/espressif/boards/esp32/CMakeLists.txt b/vendors/espressif/boards/esp32/CMakeLists.txt
index 11d77e4ad..d5e9b1ea1 100644
--- a/vendors/espressif/boards/esp32/CMakeLists.txt
+++ b/vendors/espressif/boards/esp32/CMakeLists.txt
@@ -440,6 +440,7 @@ target_include_directories(
     ${IDF_PROJECT_EXECUTABLE}
     PUBLIC
         "${esp_idf_dir}/components/esp_ringbuf/include"
+        "${esp_idf_dir}/components/fatfs"
         $<TARGET_PROPERTY:AFR::kernel,INTERFACE_INCLUDE_DIRECTORIES>
         $<TARGET_PROPERTY:AFR::ble_hal::mcu_port,INTERFACE_INCLUDE_DIRECTORIES>
         $<TARGET_PROPERTY:AFR::wifi::mcu_port,INTERFACE_INCLUDE_DIRECTORIES>
@@ -452,6 +453,8 @@ target_link_libraries(
         AFR::utils
         AFR::ble
         AFR::common_io
+    PUBLIC
+        idf::fatfs
 )
 
 if(AFR_METADATA_MODE)
diff --git a/vendors/espressif/boards/esp32/aws_demos/application_code/main.c b/vendors/espressif/boards/esp32/aws_demos/application_code/main.c
index 617a99e61..325bbc230 100644
--- a/vendors/espressif/boards/esp32/aws_demos/application_code/main.c
+++ b/vendors/espressif/boards/esp32/aws_demos/application_code/main.c
@@ -74,6 +74,8 @@
     #include "iot_ble_numericComparison.h"
 #endif
 
+#include "vfs/esp_vfs_fat.h"
+
 /* Logging Task Defines. */
 #define mainLOGGING_MESSAGE_QUEUE_LENGTH    ( 32 )
 #define mainLOGGING_TASK_STACK_SIZE         ( configMINIMAL_STACK_SIZE * 4 )
@@ -136,6 +138,7 @@ int app_main( void )
     /* Perform any hardware initialization that does not require the RTOS to be
      * running.  */
 
+    esp_vfs_fat_rawflash_unmount("test", "test");
     prvMiscInitialization();
 
     if( SYSTEM_Init() == pdPASS )

```

taherrera, 2022-08-03:
Thanks, now it isn't complaining about not finding the directory.

Is this procedure something I should do on all native IDF components that seem not to be found ? Has this component been ever used on amazon-freertos? I would like to know in case I get some strange errors while using the component..

Is changing vendors/espressif/boards/esp32/aws_demos/application_code/main.c mandatory for this to work on a custom application or is it just to prove its working?

Thanks

dachalco, 2022-08-05:
Hi @taherrera 

The change to `main.c` was to verify that it linked correctly, and that the change works. You can omit these changes to `main.c` and use the `esp_vfs_fat.h` as you please. 

If you want to add more components, you can follow the same process, including that component's includes, and linking in a similar fashion with `idf::<component>` syntax.

closed 2022-08-09 by taherrera