aws/amazon-freertos #3520: [General] Root CA rotation for server authentication

EtienneMdv, 2022-08-03:
Hi,

I have been working on creating secure sockets between my device (ESP32) and the AWS IoT broker. I am using server authentication by adding a ATS root certificate on-board. I have the following questions:
Is there anywhere online where I can get updates on the ATS certificates status? I know these certificates expire but I am wondering if they could be revoked before that date (for any reason). Furthermore, is there any guideline on how to update this certificate? I noticed that it is hard coded in the MQTT demo. This would imply initiating an OTA update to update the certificate. Is there any feature that would allow me to place this certificate at the same location as for where the device credentials are (default partition used by PKCS11)? This way I could update the certificate by sending it in the payload of a MQTT message.

I would be very interested to get your input on this. I hope I am not totally wrong in my thought process. It started being concerned after reading the following in [AWS docs:](https://docs.aws.amazon.com/iot/latest/developerguide/server-authentication.html)

> 
> Note
> 
> CA certificates have an expiration date after which they cannot be used to validate a server's certificate. CA certificates might have to be replaced before their expiration date. Make sure that you can update the root CA certificates on all of your devices or clients to help ensure ongoing connectivity and to keep up to date with security best practices.

Thank you.

ravibhagavandas, 2022-08-04:
Hi,
You can read more about all ATS certificates and check the status of the certificates using Test URLS provided here:
https://www.amazontrust.com/repository/
The page also contains "Certification Practice Statement" which describes process for revocation/status checking of the certificates. The status checking can be done through OCSP protocol when HTTPS requests are made.

If using a private CA registered with AWS IoT you can also do device defender audits to alert when a certificate is expiring:
https://docs.aws.amazon.com/iot/latest/developerguide/audit-chk-ca-cert-approaching-expiration.html

A secure element (PKCS11) slot is not required to store a root CA, as the contents are not sensitive. You can store it in non volatile memory and have the contents transferred over as MQTT payload. However most secure elements/PKCS11 provides extra slots to store certificates so you could use the slots if there is no other backing store. Here is a blog post for more details on certificate rotation:
https://aws.amazon.com/blogs/iot/how-to-manage-iot-device-certificate-rotation-using-aws-iot/

EtienneMdv, 2022-08-04:
Thank you very much @ravibhagavandas. That answered my questions.

However, I do have one more question. This topic is quite new for me... 

I am also looking at using code signing certificates. The following statement can be found in [AWS docs](https://docs.aws.amazon.com/freertos/latest/userguide/ota-code-sign-cert.html):

> To digitally sign firmware images, you need a code-signing certificate and private key. For testing purposes, you can create a self‐signed certificate and private key. For production environments, purchase a certificate through a well‐known certificate authority (CA).

Why is using a self-signed certificate for code signing in production a security concern? Is the ESP32 doing any verification on the CA that signed the certificate? I believe I am missing something to get the big picture.

ravibhagavandas, 2022-08-04:
Hi @EtienneMdv 

Having a Certificate authority(CA) allows you to validate that the code signing certificate (public key) is coming from a trusted source. The party who choose to sign the code generates a public private key pair and submits the public key to CA along with the request to generate a certificate. The certificate generated by CA can be validated at the device side to verify the identity.

The demos provides example with a self signed certificate hence the CA is not validated, however for production use cases its recommended to use a CA for generating the code signing certificates.

EtienneMdv, 2022-08-08:
Thank you @ravibhagavandas. That answers all my questions.

closed 2022-08-08 by EtienneMdv