aws/amazon-freertos #3521: [BUG] `CORE_MQTT_MUTUAL_AUTH` Demo: Failed to establish new connection

laukik-hase, 2022-08-08:
**Description**

- The `CORE_MQTT_MUTUAL_AUTH` demo fails in the handshake stage with the connection indicating an EOF.
- On using `git bisect`, the commit 747f07402 is found to be the bad commit. The demo works fine at a commit before this - 8e69ca72.

**System information**
- Hardware board: `ESP32-WROVER-KIT V4.1`
- IDE used: NA
- Operating System: Linux
- Code version: `202107.00-78-gb3fde239b` | `main` branch commit -> b3fde239b3861be9a
- Project/Demo: `mqtt_demo_mutual_auth`

**Screenshots or console output**
<details>
  <summary> Actual Output </summary>

```
I (29) boot: ESP-IDF v4.2.2-276-g4e4142ef8 2nd stage bootloader
I (29) boot: compile time 16:56:55
I (29) boot: chip revision: 3
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00010000 00006000
I (78) boot:  1 otadata          OTA data         01 00 00016000 00002000
I (85) boot:  2 phy_init         RF data          01 01 00018000 00001000
I (93) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (100) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (108) boot:  5 storage          WiFi data        01 02 00317000 00010000
I (115) boot: End of partition table
I (120) boot: No factory image, trying OTA 0
I (124) boot_comm: chip revision: 3, min. application chip revision: 0
I (132) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x29090 (168080) map
I (205) esp_image: segment 1: paddr=0x000490b8 vaddr=0x3ffbdb60 size=0x03994 ( 14740) load
I (211) esp_image: segment 2: paddr=0x0004ca54 vaddr=0x40080000 size=0x035c4 ( 13764) load
0x40080000: _WindowOverflow4 at /home/laukik/work/amazon-freertos/build/../freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1821

I (218) esp_image: segment 3: paddr=0x00050020 vaddr=0x400d0020 size=0xa8f94 (692116) map
0x400d0020: _stext at ??:?

I (484) esp_image: segment 4: paddr=0x000f8fbc vaddr=0x400835c4 size=0x19454 (103508) load
0x400835c4: esp_flash_init at /home/laukik/work/amazon-freertos/build/../vendors/espressif/esp-idf/components/spi_flash/esp_flash_api.c:116 (discriminator 3)

I (545) boot: Loaded app from partition at offset 0x20000
I (621) boot: Set actual ota_seq=1 in otadata[0]
I (621) boot: Disabling RNG early entropy source...
I (622) cpu_start: Pro cpu up.
I (625) cpu_start: Application information:
I (630) cpu_start: Project name:     amazon-freertos
I (635) cpu_start: App version:      1
I (640) cpu_start: Compile time:     Aug  8 2022 16:56:47
I (646) cpu_start: ELF file SHA256:  2e9b24952efbfaa1...
I (652) cpu_start: ESP-IDF:          v4.2.2-276-g4e4142ef8
I (658) cpu_start: Single core mode
I (662) heap_init: Initializing. RAM available for dynamic allocation:
I (669) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
I (675) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (682) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (688) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (694) heap_init: At 3FFCA388 len 00015C78 (87 KiB): DRAM
I (700) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (706) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (713) heap_init: At 4009CA18 len 000035E8 (13 KiB): IRAM
I (719) cpu_start: Pro cpu start user code
I (736) spi_flash: detected chip: generic
I (737) spi_flash: flash i�0 0 [main] Initializing lwIP TCP stack
[... 913 lines cut ...]
251 9317 [iot_thread] [INFO ][DEMO][93170] memory_metrics::freertos_heap::after::bytes::123652

252 9317 [iot_thread] [INFO ][DEMO][93170] memory_metrics::demo_task_stack::before::bytes::5848

253 9317 [iot_thread] [INFO ][DEMO][93170] memory_metrics::demo_task_stack::after::bytes::2408

I (94913) wifi:state: run -> init (0)
I (94913) wifi:pm stop, total sleep time: 18329154 us / 90769674 us

I (94913) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (94913) WIFI: WIFI_EVENT_STA_DISCONNECTED: 63
254 9417 [iot_thread] [INFO ][DEMO][94170] Demo completed successfully.

E (94923) wifi_init: Wi-Fi not stop
I (94933) WIFI: WIFI_EVENT_STA_STOP
I (94933) wifi:flush txq
I (94933) wifi:stop sw txq
I (94943) wifi:lmac stop hw txq
I (94943) wifi:Deinit lldesc rx mblock:10
255 9421 [iot_thread] [INFO ][INIT][94210] SDK cleanup done.

256 9421 [iot_thread] [INFO ][DEMO][94210] -------DEMO FINISHED-------

```
</details>

ravibhagavandas, 2022-08-08:
Hi @laukik-hase 

I tried with the commit `b3fde23`  on main branch, but could not reproduce the issue. Please find attached logs for MQTT Mutual Auth demo on ESP32 WROOM board.

[mqtt_success_logs.txt](https://github.com/aws/amazon-freertos/files/9285491/mqtt_success_logs.txt)

> 7 601 [iot_thread] ERROR: TLS handshake failed trying to connect. SSL - The connection indicated an EOF : <No-Low-Level-Code> 
8 601 [iot_thread] TLS_Connect fail (0x7280, a10y2vqnezxqbn-ats.iot.us-east-1.amazonaws.com)

The error 0x7280 indicates a terminated connection by the server. Can you confirm if the certificate used is valid and registered with AWS IoT ? Also could you check if the policy attached to the certificate has sufficient permissions?

laukik-hase, 2022-08-09:
Hello, @ravibhagavandas!

I have checked the AWS IoT configuration (thing, certificate and policy attached) and it is correct.
I am confused why a configuration would work at https://github.com/aws/amazon-freertos/commit/8e69ca72d015e6285f49c90a202651a73acfc692 but fail at a commit just after it. 

(_**Note**: The demo also works on the `202203.00` release with the same config_)

Apart from setting the appropriate parameters in `aws_clientcredential.h` and `aws_clientcredential_keys.h`, is anything else required for the `mqtt_demo_mutual_auth` for the latest `main` branch?

ravibhagavandas, 2022-08-10:
> Apart from setting the appropriate parameters in aws_clientcredential.h and aws_clientcredential_keys.h, is anything else required for the mqtt_demo_mutual_auth on the latest main branch?

Only these configuration files needs to be modified for the demo.

Could you enable mbedtls debug logs and share the logs for TLS handshake failure?

laukik-hase, 2022-08-10:
PFA the Mbed TLS debug logs: [mqtt_fail.log](https://github.com/aws/amazon-freertos/files/9297674/mqtt_fail.log)

On another note, increasing the `mqttexampleTRANSPORT_SEND_RECV_TIMEOUT_MS` to 1000 in `mqtt_demo_mutual_auth.c` makes the example as expected.

n9wxu, 2022-08-12:
We have seen changes to the timeout making a big difference in different regions.

closed 2022-08-12 by n9wxu