aws/amazon-freertos #3525: [BUG] Array bound warning observed in iot_test_tcp.c

RashidAhmed-Silabs, 2022-08-18:
Description:
Observing the array bound warning in iot_test_tcp.c (libraries/abstractions/secure_sockets/test)
Please find the attached screenshot for the warning observed.
[image]

The warning is being observed in the below line of code from iot_test_tcp.c
Line number -2509   cFakeAddress[ sizeof( clientcredentialMQTT_BROKER_ENDPOINT ) - 3 ]++;

aggarg, 2022-08-22:
What is the value of `clientcredentialMQTT_BROKER_ENDPOINT`? Assuming that you are compiling a test project, you are supposed to define it here -  https://github.com/aws/amazon-freertos/blob/main/tests/include/aws_clientcredential.h#L37

alfred2g, 2022-09-17:
Closing for no activity, please reopen if anything new comes up

closed 2022-09-17 by alfred2g