aws/amazon-freertos #3527: [Feature Request] Allow to define custom 'help' command in freertos-cli

EdwinFairchild, 2022-08-27:
**Is your feature request related to an issue you've encountered? If so, please elaborate.**
Please provide a clear and concise description of the problem. For example, "I'm always frustrated when [ ... ]".
- I would like define my own format to the output of the `help` command in freertos-cli. The current printed output leaves something to be desired and you have to  basically do all the formatting in the help message string to get something looking the way you want it.

**Describe the new feature request**
Provide a clear and concise description of the feature you are proposing, as well as applicable use cases.
- Add a switch to freeRTOSConfig.h to use custom `help` command so that the one define in freertos-cli.c is not registered.

**Describe other alternatives you've considered**
Provide a clear and concise description of any alternatives you've considered.
- I could define another custom command but `help` is just so convenient and universal

**Additional context**
Add any additional information supporting the request here (screenshots, logs, references, etc.).

jasonpcarroll, 2022-09-01:
Hi @EdwinFairchild.
I will bring it up with the team.
Best
Jason Carroll

kstribrnAmzn, 2022-11-11:
@EdwinFairchild, posting this to keep you informed as to what my team (aka the team) has decided and explain our direction. Thank you for bringing this up, your request led to an interesting conversation within the team.

AmazonFreeRTOS is essentially a monolith. This package has a lot of libraries and functionality integrated out of the box which our simpler FreeRTOS/Kernel does not. Ideally this overridable 'help' feature could be easily used across both the standalone FreeRTOS/Kernel and AmazonFreeRTOS. With this in mind, we plan on implementing this feature into a separate module which can be added to a project (through a submodule for example). By doing this we hope to ensure that users like yourself can easily have this functionality while users who don't require it don't need to configure a thing.

We've created an issue in our own backlog - which is unfortunately hidden from you - as well as on the module repo we plan on implementing this in as an issue. This repo is currently private but when it's set for release you'll be able to view the status of the feature request [here](https://github.com/FreeRTOS/FreeRTOS-Plus-CLI/issues/1).

Now this doesn't mean you (or someone else invested) can't make the change. If you would like to make a PR addressing this issue, we'd be happy to review and pull the change into our code base.

Anyway, I hope this sheds light on why I'll be closing this Feature Request here. Your request isn't ignored, we intend on taking this work up, simply elsewhere.

closed 2022-11-11 by kstribrnAmzn